MiczFlor/RPi-Jukebox-RFID #1238: 🚀 | Rework SPOTIFY_INTEGRATION.md

s-martin, 2020-12-27:
[SPOTIFY-INTEGRATION.md](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/develop/docs/SPOTIFY-INTEGRATION.md) is still the only document describing how to use the Spotify edition.

however, most of the content is outdated, because it describes how to manually extend the Classic edition to Spotify.

should be reworked to the current state of implementation.

s-martin, 2021-01-01:
See also #1252

Groovylein, 2021-01-04:
Since there is only one script for both installation paths, I would merge it with the Spotify-FAQ and delete this readme

Groovylein, 2021-01-04:
Looks like the information after the section **MUST READ** is not necessary. Therefore I would just erase [SPOTIFY-INTEGRATION.md](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/develop/docs/SPOTIFY-INTEGRATION.md)

pabera, 2021-02-15:
I might have addressed this issue a little with [this change](https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/1297). But I agree, it could use a little more care.

Groovylein, 2021-04-07:
Now this can be closed @s-martin :)

closed 2021-04-07 by s-martin