MiczFlor/RPi-Jukebox-RFID #1243: Stoping playing mp3 and webstreams after starting certain mp3s on Raspberry Pi 4

alwashe, 2020-12-30:
After starting (but not playing) certain mp3 files the phoniebox stops to play anything.
After a reboot Webstreams and the working mp3s are playing again. Until I try again to start this certain mp3s.

I have tried several installs. It does not matter if I start via RFID or webinterface.
I tried two diffrent Raspberry Pi 4. No soundcards.

This does not happen on Raspberry  Pi 3 at all. Using only Raspberry Pi 3 This is for now my workaround.

s-martin, 2020-12-30:
What’s the output of `tail /var/log/syslog`?

alwashe, 2021-01-01:
After restart:
pi@phoniebox2a:~ $ tail /var/log/syslog
Jan  1 13:24:17 phoniebox2a mopidy[5411]: INFO     [MpdSession-40] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:42022
Jan  1 13:24:17 phoniebox2a mopidy[5411]: INFO     [MpdSession-41] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:42026
Jan  1 13:24:17 phoniebox2a mopidy[5411]: INFO     [MpdSession-42] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:42030
Jan  1 13:24:18 phoniebox2a mpd: exception: bind to '127.0.0.1:6600' failed (continuing anyway, because binding to '[::1]:6600' succeeded): Failed to bind socket: Address already in use
Jan  1 13:24:18 phoniebox2a mpd: exception: Failed to set group 0: Operation not permitted
Jan  1 13:24:18 phoniebox2a systemd[7218]: mpd.service: Main process exited, code=exited, status=1/FAILURE
Jan  1 13:24:18 phoniebox2a systemd[7218]: mpd.service: Failed with result 'exit-code'.
Jan  1 13:24:18 phoniebox2a systemd[7218]: Failed to start Music Player Daemon.
Jan  1 13:24:18 phoniebox2a systemd[7218]: Reached target Default.
Jan  1 13:24:18 phoniebox2a systemd[7218]: Startup finished in 2.676s.

after starting a working mp3
pi@phoniebox2a:~ $ tail /var/log/syslog
Jan  1 13:24:52 phoniebox2a mopidy[5411]: INFO     [MpdSession-113] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:42322
Jan  1 13:24:56 phoniebox2a systemd[1]: phoniebox-idle-watchdog.service: Succeeded.
Jan  1 13:24:58 phoniebox2a mopidy[5411]: INFO     [MpdSession-114] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:42326
Jan  1 13:24:58 phoniebox2a mopidy[5411]: INFO     [MpdSession-115] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:42330
Jan  1 13:24:58 phoniebox2a mopidy[5411]: INFO     [MpdSession-116] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:42334
Jan  1 13:24:58 phoniebox2a mopidy[5411]: INFO     [MpdSession-117] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:42338
Jan  1 13:24:58 phoniebox2a mopidy[5411]: INFO     [MpdSession-118] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:42342
Jan  1 13:24:58 phoniebox2a mopidy[5411]: INFO     [MpdSession-119] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:42346
Jan  1 13:24:58 phoniebox2a mopidy[5411]: INFO     [MpdSession-120] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:42350
Jan  1 13:24:58 phoniebox2a mopidy[5411]: INFO     [MpdSession-121] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:42354

after starting an not working mp3
pi@phoniebox2a:~ $ tail /var/log/syslog
Jan  1 13:25:27 phoniebox2a mopidy[5411]: INFO     [MpdSession-173] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:42562
Jan  1 13:25:27 phoniebox2a mopidy[5411]: INFO     [MpdSession-174] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:42566
Jan  1 13:25:27 phoniebox2a mopidy[5411]: INFO     [MpdSession-175] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:42570
Jan  1 13:25:27 phoniebox2a mopidy[5411]: INFO     [MpdSession-176] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:42574
Jan  1 13:25:27 phoniebox2a mopidy[5411]: INFO     [MpdSession-177] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:42578
Jan  1 13:25:27 phoniebox2a mopidy[5411]: INFO     [MpdSession-178] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:42582
Jan  1 13:25:27 phoniebox2a mopidy[5411]: INFO     [MpdSession-179] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:42586
Jan  1 13:25:27 phoniebox2a mopidy[5411]: INFO     [MpdSession-180] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:42590
Jan  1 13:25:27 phoniebox2a mopidy[5411]: INFO     [MpdSession-181] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:42594
Jan  1 13:25:27 phoniebox2a mopidy[5411]: INFO     [MpdSession-182] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:42598

s-martin, 2021-01-01:
Ok, please post the output of `journalctl -u mopidy`

alwashe, 2021-01-01:
```
pi@phoniebox2a:~ $ journalctl -u mopidy
-- Logs begin at Fri 2021-01-01 13:30:37 GMT, end at Fri 2021-01-01 18:34:51 GMT
Jan 01 13:30:53 phoniebox2a systemd[1]: Starting Mopidy music server...
Jan 01 13:30:53 phoniebox2a systemd[1]: Started Mopidy music server.
Jan 01 13:30:56 phoniebox2a mopidy[6045]: INFO     [MainThread] mopidy.__main__
Jan 01 13:30:56 phoniebox2a mopidy[6045]: INFO     [MainThread] mopidy.config Lo
Jan 01 13:30:56 phoniebox2a mopidy[6045]: INFO     [MainThread] mopidy.config Lo
Jan 01 13:30:56 phoniebox2a mopidy[6045]: INFO     [MainThread] mopidy.config Lo
Jan 01 13:30:56 phoniebox2a mopidy[6045]: INFO     [MainThread] mopidy.config Lo
Jan 01 13:30:56 phoniebox2a mopidy[6045]: INFO     [MainThread] mopidy.config Lo
Jan 01 13:30:56 phoniebox2a mopidy[6045]: ERROR    [MainThread] mopidy.internal.
Jan 01 13:30:56 phoniebox2a mopidy[6045]: INFO     [MainThread] mopidy.__main__
Jan 01 13:30:56 phoniebox2a mopidy[6045]: INFO     [MainThread] mopidy.__main__
Jan 01 13:30:56 phoniebox2a mopidy[6045]: WARNING  [MainThread] mopidy.__main__
Jan 01 13:30:56 phoniebox2a mopidy[6045]: WARNING  [MainThread] mopidy.__main__
Jan 01 13:30:56 phoniebox2a mopidy[6045]: WARNING  [MainThread] mopidy.__main__
Jan 01 13:30:56 phoniebox2a mopidy[6045]: WARNING  [MainThread] mopidy.__main__
Jan 01 13:30:56 phoniebox2a mopidy[6045]: WARNING  [MainThread] mopidy.__main__
Jan 01 13:30:56 phoniebox2a mopidy[6045]: WARNING  [MainThread] mopidy.__main__
Jan 01 13:30:57 phoniebox2a mopidy[6045]: INFO     [MainThread] mopidy.commands
Jan 01 13:30:57 phoniebox2a mopidy[6045]: INFO     [MainThread] mopidy.commands
Jan 01 13:30:57 phoniebox2a mopidy[6045]: INFO     [MainThread] mopidy.commands
Jan 01 13:30:57 phoniebox2a mopidy[6045]: INFO     [MainThread] mopidy.commands
Jan 01 13:30:57 phoniebox2a mopidy[6045]: INFO     [MainThread] mopidy.commands
lines 1-23...skipping...
-- Logs begin at Fri 2021-01-01 13:30:37 GMT, end at Fri 2021-01-01 18:34:51 GMT. --
Jan 01 13:30:53 phoniebox2a systemd[1]: Starting Mopidy music server...
Jan 01 13:30:53 phoniebox2a systemd[1]: Started Mopidy music server.
Jan 01 13:30:56 phoniebox2a mopidy[6045]: INFO     [MainThread] mopidy.__main__ Starting Mopidy 3.0.2
Jan 01 13:30:56 phoniebox2a mopidy[6045]: INFO     [MainThread] mopidy.config Loading config from builtin defaults
Jan 01 13:30:56 phoniebox2a mopidy[6045]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy-local.conf
Jan 01 13:30:56 phoniebox2a mopidy[6045]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy.conf
Jan 01 13:30:56 phoniebox2a mopidy[6045]: INFO     [MainThread] mopidy.config Loading config from file:///etc/mopidy/mopidy.conf
Jan 01 13:30:56 phoniebox2a mopidy[6045]: INFO     [MainThread] mopidy.config Loading config from command line options
Jan 01 13:30:56 phoniebox2a mopidy[6045]: ERROR    [MainThread] mopidy.internal.log Loading logging config '/etc/mopidy/logging.conf' failed. 'formatters'
Jan 01 13:30:56 phoniebox2a mopidy[6045]: INFO     [MainThread] mopidy.__main__ Enabled extensions: m3u, file, http, stream, mpd, local, iris, softwaremixer
Jan 01 13:30:56 phoniebox2a mopidy[6045]: INFO     [MainThread] mopidy.__main__ Disabled extensions: spotify
Jan 01 13:30:56 phoniebox2a mopidy[6045]: WARNING  [MainThread] mopidy.__main__ Found spotify configuration errors. The extension has been automatically disabled:
Jan 01 13:30:56 phoniebox2a mopidy[6045]: WARNING  [MainThread] mopidy.__main__   spotify/username must be set.
Jan 01 13:30:56 phoniebox2a mopidy[6045]: WARNING  [MainThread] mopidy.__main__   spotify/password must be set.
Jan 01 13:30:56 phoniebox2a mopidy[6045]: WARNING  [MainThread] mopidy.__main__   spotify/client_id must be set.
Jan 01 13:30:56 phoniebox2a mopidy[6045]: WARNING  [MainThread] mopidy.__main__   spotify/client_secret must be set.
Jan 01 13:30:56 phoniebox2a mopidy[6045]: WARNING  [MainThread] mopidy.__main__ Please fix the extension configuration errors or disable the extensions to silence these messages.
Jan 01 13:30:57 phoniebox2a mopidy[6045]: INFO     [MainThread] mopidy.commands Starting Mopidy mixer: SoftwareMixer
Jan 01 13:30:57 phoniebox2a mopidy[6045]: INFO     [MainThread] mopidy.commands Mixer volume set to 30
Jan 01 13:30:57 phoniebox2a mopidy[6045]: INFO     [MainThread] mopidy.commands Starting Mopidy audio
Jan 01 13:30:57 phoniebox2a mopidy[6045]: INFO     [MainThread] mopidy.commands Starting Mopidy backends: FileBackend, M3UBackend, StreamBackend, LocalBackend
Jan 01 13:30:57 phoniebox2a mopidy[6045]: INFO     [MainThread] mopidy.commands Starting Mopidy core
Jan 01 13:30:57 phoniebox2a mopidy[6045]: INFO     [Audio-2] mopidy.audio.actor Audio output set to "alsasink"
Jan 01 13:30:57 phoniebox2a mopidy[6045]: INFO     [MainThread] mopidy.commands Starting Mopidy frontends: HttpFrontend, IrisFrontend, MpdFrontend
Jan 01 13:30:57 phoniebox2a mopidy[6045]: INFO     [HttpFrontend-9] mopidy.http.actor HTTP server running at [::ffff:0.0.0.0]:6680
Jan 01 13:30:57 phoniebox2a mopidy[6045]: INFO     [IrisFrontend-10] mopidy_iris.core Starting Iris 3.45.0
Jan 01 13:30:57 phoniebox2a mopidy[6045]: INFO     [MainThread] mopidy_mpd.actor MPD server running at [::ffff:0.0.0.0]:6600
Jan 01 13:30:57 phoniebox2a mopidy[6045]: INFO     [MainThread] mopidy.commands Starting GLib mainloop
[... cut ...]
Jan 01 18:36:24 phoniebox2a mopidy[6568]: WARNING  [FileBackend-3] mopidy.file.library Failed looking up file:///home/pi/RPi-Jukebox-RFID/shared/audiofolders/Ghostsitter/Tommy%20Krappweis%20-%20Ghostsitter%202.%20Staffel/Ghostsitter%20Die%20komplette%202.%20Staffel%20-%2012.mp3: Timeout after 1ms
Jan 01 18:36:24 phoniebox2a mopidy[6568]: WARNING  [FileBackend-3] mopidy.file.library Failed looking up file:///home/pi/RPi-Jukebox-RFID/shared/audiofolders/Ghostsitter/Tommy%20Krappweis%20-%20Ghostsitter%202.%20Staffel/Ghostsitter%20Die%20komplette%202.%20Staffel%20-%2013.mp3: Timeout after 1ms
```

s-martin, 2021-01-01:
You don't have set your Spotify credentials

And for the second error, please try:
`sudo pip3 install --upgrade --force-reinstall Mopidy-Iris` and reboot

alwashe, 2021-01-01:
hi, i have no Spotify, i use the classic version.

I found the problem, when the Cover-Picture is named folder.jpg it is part of the playlist, when it is named cover.jpg everything is ok.

s-martin, 2021-01-03:
Ok, please close it, if it's done for you.

closed 2021-01-07 by alwashe