MiczFlor/RPi-Jukebox-RFID #1254: Played file stopps randomly / mpg123 direct play working

rendgeor, 2021-01-01:
## Bug

### What I did

Play mp3 file via Web API AND via token. Same result.

### What happened

The played file stops at a random position.

### I expected this to happen

File plays till the end.
Playing the same file via mpg123 works as expected.

### Further information that might help

Analytics_AfterInstallScript.sh 
--------------------------------------
<pre>************************************
*** PHONIEBOX INFO
*** version: 2.2 - 305325d - master
*** edition: classic
*** Audio_iFace_Name: Headphone
*** Audio_Folders_Path: /home/pi/RPi-Jukebox-RFID/shared/audiofolders
*** Audio_Volume_Change_Step: 3
*** Max_Volume_Limit: 100
*** Idle_Time_Before_Shutdown: 0
*** Second_Swipe: RESTART
*** Playlists_Folders_Path: /home/pi/RPi-Jukebox-RFID/playlists
*** ShowCover: ON
************************************
*** CONF FILES DEFAULT
 
*** /etc/samba/smb.conf
-rw-r--r-- 1 root root 8.2K Dec 30 22:27 /etc/samba/smb.conf
   path=/home/pi/RPi-Jukebox-RFID/shared
   path=/home/pi/RPi-Jukebox-RFID/shared/audiofolders
*** /etc/lighttpd/lighttpd.conf
-rw-r--r-- 1 root root 2.1K Dec 30 22:27 /etc/lighttpd/lighttpd.conf
*** /etc/lighttpd/conf-available/15-fastcgi-php.conf
-rw-r--r-- 1 root root 575 Dec 30 22:27 /etc/lighttpd/conf-available/15-fastcgi-php.conf
*** /etc/php/7.0/fpm/php.ini
ls: cannot access &apos;/etc/php/7.0/fpm/php.ini&apos;: No such file or directory
*** /etc/sudoers
-r--r----- 1 root root 784 Dec 30 22:27 /etc/sudoers
*** /etc/systemd/system/phoniebox*
-rw-r--r-- 1 root root 295 Dec 30 22:27 /etc/systemd/system/phoniebox-rfid-reader.service
ls: cannot access &apos;/etc/systemd/system/phoniebox-startup-sound.service&apos;: No such file or directory
-rw-r--r-- 1 root root 322 Dec 30 22:51 /etc/systemd/system/phoniebox-gpio-buttons.service
-rw-r--r-- 1 root root 291 Dec 30 22:27 /etc/systemd/system/phoniebox-idle-watchdog.service
*** /etc/mpd.conf
-rw-r----- 1 mpd audio 14K Dec 30 23:39 /etc/mpd.conf
music_directory		&quot;/home/pi/RPi-Jukebox-RFID/shared/audiofolders&quot;
# music_directory are changed.
# symbolic links outside of the configured music_directory.
# symbolic links inside of the configured music_directory.
	mixer_control	&quot;Headphone&quot;		# optional
#	mixer_control	&quot;PCM&quot;		# optional
*** /etc/dhcpcd.conf
-rw-rw-r-- 1 root netdev 1.8K Nov 13  2019 /etc/dhcpcd.conf
#static ip_address=192.168.0.10/24
#static ip_address=192.168.1.23/24
#static routers=192.168.0.1
#static routers=192.168.1.1
option domain_name_servers, domain_name, domain_search, host_name
#static domain_name_servers=192.168.0.1 8.8.8.8 fd51:42f8:caae:d92e::1
#static domain_name_servers=192.168.1.1
*** /etc/wpa_supplicant/wpa_supplicant.conf
-rw------- 1 root root 186 Dec 30 18:54 /etc/wpa_supplicant/wpa_supplicant.conf
country=DE
************************************
*** +Spotify Edition
 
*** /etc/locale.gen
-rw-r--r-- 1 root root 9.2K Dec  2 12:38 /etc/locale.gen
*** /etc/locale.gen
-rw-r--r-- 1 root root 9.2K Dec  2 12:38 /etc/locale.gen
*** /etc/mopidy/mopidy.conf
ls: cannot access &apos;/etc/mopidy/mopidy.conf&apos;: No such file or directory
cat: /etc/mopidy/mopidy.conf: No such file or directory
*** ~/.config/mopidy/mopidy.conf
ls: cannot access &apos;/home/pi/.config/mopidy/mopidy.conf&apos;: No such file or directory
cat: /home/pi/.config/mopidy/mopidy.conf: No such file or directory
</pre>

## Software

### Base image and version

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian

### Branch / Release

On branch master
Your branch is up to date with 'origin/master'.
### Installscript

## Hardware

### RaspberryPi version

Raspberry Pi 4 Model B Rev 1.1
### RFID Reader

Neuftech

### Soundcard

pifi dac+

### Other notable hardware

buttons not used so far

s-martin, 2021-01-01:
What's the output of `tail -n 500 /var/log/syslog`?

rendgeor, 2021-01-01:
aktuell läuft es... warte auf den Abbruch;-)

MPD service thrown an exception...?!

Jan  1 15:49:20 pipi01 systemd[1]: Started User Manager for UID 1000.
Jan  1 15:49:20 pipi01 systemd[1162]: Starting Music Player Daemon...
Jan  1 15:49:20 pipi01 systemd[1]: Started Session 1 of user pi.
Jan  1 15:49:21 pipi01 mpd: exception: Failed to bind to '[::1]:6600'
Jan  1 15:49:21 pipi01 mpd: exception: nested: Failed to bind socket: Address already in use
Jan  1 15:49:21 pipi01 systemd[1162]: mpd.service: Main process exited, code=exited, status=1/FAILURE
Jan  1 15:49:21 pipi01 systemd[1162]: mpd.service: Failed with result 'exit-code'.
Jan  1 15:49:21 pipi01 systemd[1162]: Failed to start Music Player Daemon.
Jan  1 15:49:21 pipi01 systemd[1162]: Reached target Default.
Jan  1 15:49:21 pipi01 systemd[1162]: Startup finished in 741ms.
Jan  1 15:49:21 pipi01 dhcpcd[581]: wlan0: no IPv6 Routers available
Jan  1 15:49:30 pipi01 systemd[1]: systemd-fsckd.service: Succeeded.
Jan  1 15:58:34 pipi01 systemd-timesyncd[320]: Synchronized to time server for the first time 129.70.132.32:123 (2.debian.pool.ntp.org).
Jan  1 15:58:36 pipi01 daemon_rfid_reader.py[599]: 2021-01-01 15:58:36,079 - root - INFO - Trigger Play Cardid=0003081534
Jan  1 15:58:36 pipi01 daemon_rfid_reader.py[599]: volume: 95%   repeat: off   random: off   single: off   consume: off
Jan  1 15:58:36 pipi01 daemon_rfid_reader.py[599]: volume: 95%   repeat: off   random: off   single: off   consume: off
Jan  1 15:58:36 pipi01 daemon_rfid_reader.py[599]: loading: Janosch – Löwenzahn und Seidenpfote
Jan  1 15:58:36 pipi01 daemon_rfid_reader.py[599]: volume: 95%   repeat: off   random: off   single: off   consume: off
Jan  1 15:58:36 pipi01 daemon_rfid_reader.py[599]: volume: 95%   repeat: off   random: off   single: off   consume: off
Jan  1 15:58:36 pipi01 daemon_rfid_reader.py[599]: OK MPD 0.21.4
Jan  1 15:58:36 pipi01 daemon_rfid_reader.py[599]: OK

rendgeor, 2021-01-01:
ok, now stopped...

<pre>Jan  1 15:58:36 pipi01 kernel: [   39.940136] pcm512x 1-004d: No SCLK, using BCLK: -2
Jan  1 15:58:42 pipi01 systemd[1]: systemd-hostnamed.service: Succeeded.
Jan  1 15:59:17 pipi01 systemd[1]: phoniebox-idle-watchdog.service: Succeeded.
Jan  1 15:59:36 pipi01 systemd[1]: getty@tty1.service: Succeeded.
Jan  1 15:59:36 pipi01 systemd[1]: getty@tty1.service: Service has no hold-off time (RestartSec=0), scheduling restart.
Jan  1 15:59:36 pipi01 systemd[1]: getty@tty1.service: Scheduled restart job, restart counter is at 1.
Jan  1 15:59:36 pipi01 systemd[1]: Stopped Getty on tty1.
Jan  1 15:59:36 pipi01 systemd[1]: Started Getty on tty1.
Jan  1 16:01:11 pipi01 daemon_rfid_reader.py[599]: 2021-01-01 16:01:11,493 - root - INFO - Trigger Play Cardid=0003081534
Jan  1 16:01:12 pipi01 mpd: player: played &quot;Janosch – Löwenzahn und Seidenpfote/kleiner-eisbaer--lars-bring-uns-nach-hause.mp3&quot;
Jan  1 16:01:12 pipi01 daemon_rfid_reader.py[599]: volume: 95%   repeat: off   random: off   single: off   consume: off
Jan  1 16:01:12 pipi01 daemon_rfid_reader.py[599]: volume: 95%   repeat: off   random: off   single: off   consume: off
Jan  1 16:01:12 pipi01 daemon_rfid_reader.py[599]: loading: Janosch – Löwenzahn und Seidenpfote
Jan  1 16:01:12 pipi01 daemon_rfid_reader.py[599]: volume: 95%   repeat: off   random: off   single: off   consume: off
Jan  1 16:01:12 pipi01 daemon_rfid_reader.py[599]: volume: 95%   repeat: off   random: off   single: off   consume: off
Jan  1 16:01:12 pipi01 daemon_rfid_reader.py[599]: OK MPD 0.21.4
Jan  1 16:01:12 pipi01 daemon_rfid_reader.py[599]: OK
Jan  1 16:01:12 pipi01 kernel: [  195.378757] pcm512x 1-004d: No SCLK, using BCLK: -2
Jan  1 16:01:13 pipi01 daemon_rfid_reader.py[599]: OK MPD 0.21.4
Jan  1 16:01:13 pipi01 daemon_rfid_reader.py[599]: OK
Jan  1 16:02:11 pipi01 systemd[1]: getty@tty1.service: Succeeded.
Jan  1 16:02:11 pipi01 systemd[1]: getty@tty1.service: Service has no hold-off time (RestartSec=0), scheduling restart.
Jan  1 16:02:11 pipi01 systemd[1]: getty@tty1.service: Scheduled restart job, restart counter is at 2.
Jan  1 16:02:11 pipi01 systemd[1]: Stopped Getty on tty1.
Jan  1 16:02:11 pipi01 systemd[1]: Started Getty on tty1.
Jan  1 16:06:33 pipi01 mpd: player: played &quot;Janosch – Löwenzahn und Seidenpfote/kleiner-eisbaer--lars-bring-uns-nach-hause.mp3&quot;
Jan  1 16:06:33 pipi01 python3[590]: volume: 95%   repeat: off   random: off   single: off   consume: off</pre>

s-martin, 2021-01-01:
Looks pretty normal. Do the files work in another player?

rendgeor, 2021-01-01:
Just had this first file for testing and then the problem showed up:

U can get it there.
https://www.gratis-hoerspiele.de/go/seidenpfote

I will get a second mp3 and test it with that...

rendgeor, 2021-01-01:
Happens also with other file.
BTW Don have any hard buttons connected so far, just card reader, audio card, battery hat and speaker...

Jan  1 17:05:53 pipi01 daemon_rfid_reader.py[599]: /home/pi/RPi-Jukebox-RFID/scripts/resume_play.sh: line 76: [: ==: unary operator expected
Jan  1 17:05:53 pipi01 daemon_rfid_reader.py[599]: /home/pi/RPi-Jukebox-RFID/scripts/resume_play.sh: line 76: [: ==: unary operator expected
Jan  1 17:05:53 pipi01 daemon_rfid_reader.py[599]: volume: 45%   repeat: off   random: off   single: off   consume: off
Jan  1 17:05:53 pipi01 daemon_rfid_reader.py[599]: volume: 45%   repeat: off   random: off   single: off   consume: off
Jan  1 17:05:53 pipi01 daemon_rfid_reader.py[599]: loading: Sandelf
Jan  1 17:05:53 pipi01 daemon_rfid_reader.py[599]: volume: 45%   repeat: off   random: off   single: off   consume: off
Jan  1 17:05:53 pipi01 daemon_rfid_reader.py[599]: volume: 45%   repeat: off   random: off   single: off   consume: off
Jan  1 17:05:53 pipi01 daemon_rfid_reader.py[599]: OK MPD 0.21.4
Jan  1 17:05:53 pipi01 daemon_rfid_reader.py[599]: OK
Jan  1 17:05:53 pipi01 kernel: [ 4077.008802] pcm512x 1-004d: No SCLK, using BCLK: -2
Jan  1 17:06:30 pipi01 mpd: player: played "Sandelf/kinderhoerspiel_der_sandelf_drk_20200823_0805_0c074cf7.mp3"
Jan  1 17:06:30 pipi01 python3[590]: volume: 75%   repeat: off   random: off   single: off   consume: off
Jan  1 17:06:53 pipi01 systemd[1]: getty@tty1.service: Succeeded.
Jan  1 17:06:53 pipi01 systemd[1]: getty@tty1.service: Service has no hold-off time (RestartSec=0), scheduling restart.
Jan  1 17:06:53 pipi01 systemd[1]: getty@tty1.service: Scheduled restart job, restart counter is at 8.
Jan  1 17:06:53 pipi01 systemd[1]: Stopped Getty on tty1.
Jan  1 17:06:53 pipi01 systemd[1]: Started Getty on tty1.
Jan  1 17:07:00 pipi01 mpd: exception: Not playing
Jan  1 17:07:00 pipi01 python3[590]: mpd error: Not playing
Jan  1 17:07:00 pipi01 python3[590]:   File "/usr/lib/python3.7/subprocess.py", line 347, in check_call
Jan  1 17:07:00 pipi01 python3[590]:     raise CalledProcessError(retcode, cmd)
Jan  1 17:07:00 pipi01 python3[590]: subprocess.CalledProcessError: Command '/home/pi/RPi-Jukebox-RFID/scripts/playout_controls.sh -c=playernext' returned non-zero exit status 1.
Jan  1 17:07:00 pipi01 python3[590]: Traceback (most recent call last):
Jan  1 17:07:00 pipi01 python3[590]:   File "/usr/lib/python3/dist-packages/gpiozero/pins/rpigpio.py", line 244, in _call_when_changed
Jan  1 17:07:00 pipi01 python3[590]:     super(RPiGPIOPin, self)._call_when_changed()
Jan  1 17:07:00 pipi01 python3[590]:   File "/usr/lib/python3/dist-packages/gpiozero/pins/local.py", line 143, in _call_when_changed
Jan  1 17:07:00 pipi01 python3[590]:     self.state if state is None else state)
Jan  1 17:07:00 pipi01 python3[590]:   File "/usr/lib/python3/dist-packages/gpiozero/pins/pi.py", line 293, in _call_when_changed
Jan  1 17:07:00 pipi01 python3[590]:     method(ticks, state)
Jan  1 17:07:00 pipi01 python3[590]:   File "/usr/lib/python3/dist-packages/gpiozero/input_devices.py", line 197, in _pin_changed
Jan  1 17:07:00 pipi01 python3[590]:     self._fire_events(ticks, bool(self._state_to_value(state)))
Jan  1 17:07:00 pipi01 python3[590]:   File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 368, in _fire_events
Jan  1 17:07:00 pipi01 python3[590]:     self._fire_activated()
Jan  1 17:07:00 pipi01 python3[590]:   File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 397, in _fire_activated
Jan  1 17:07:00 pipi01 python3[590]:     super(HoldMixin, self)._fire_activated()
Jan  1 17:07:00 pipi01 python3[590]:   File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 344, in _fire_activated
Jan  1 17:07:00 pipi01 python3[590]:     self.when_activated()
Jan  1 17:07:00 pipi01 python3[590]:   File "/home/pi/oled_phoniebox/scripts/gpio-buttons/./gpio-buttons.py", line 42, in def_next
Jan  1 17:07:00 pipi01 python3[590]:     check_call(jukebox4kidsPath+"/scripts/playout_controls.sh -c=playernext", shell=True)
Jan  1 17:07:00 pipi01 python3[590]:   File "/usr/lib/python3.7/subprocess.py", line 347, in check_call
Jan  1 17:07:00 pipi01 python3[590]:     raise CalledProcessError(retcode, cmd)
Jan  1 17:07:00 pipi01 python3[590]: subprocess.CalledProcessError: Command '/home/pi/RPi-Jukebox-RFID/scripts/playout_controls.sh -c=playernext' returned non-zero exit status 1.
Jan  1 17:07:00 pipi01 python3[590]: Traceback (most recent call last):
Jan  1 17:07:00 pipi01 python3[590]:   File "/usr/lib/python3/dist-packages/gpiozero/pins/rpigpio.py", line 244, in _call_when_changed
Jan  1 17:07:00 pipi01 python3[590]:     super(RPiGPIOPin, self)._call_when_changed()
Jan  1 17:07:00 pipi01 python3[590]:   File "/usr/lib/python3/dist-packages/gpiozero/pins/local.py", line 143, in _call_when_changed
Jan  1 17:07:00 pipi01 python3[590]:     self.state if state is None else state)
Jan  1 17:07:00 pipi01 python3[590]:   File "/usr/lib/python3/dist-packages/gpiozero/pins/pi.py", line 293, in _call_when_changed
Jan  1 17:07:00 pipi01 python3[590]:     method(ticks, state)
Jan  1 17:07:00 pipi01 python3[590]:   File "/usr/lib/python3/dist-packages/gpiozero/input_devices.py", line 197, in _pin_changed
Jan  1 17:07:00 pipi01 python3[590]:     self._fire_events(ticks, bool(self._state_to_value(state)))
Jan  1 17:07:00 pipi01 python3[590]:   File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 368, in _fire_events
[... cut ...]
Jan  1 17:07:00 pipi01 python3[590]:   File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 397, in _fire_activated
Jan  1 17:07:00 pipi01 python3[590]:     super(HoldMixin, self)._fire_activated()
Jan  1 17:07:00 pipi01 python3[590]:   File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 344, in _fire_activated
Jan  1 17:07:00 pipi01 python3[590]:     self.when_activated()
Jan  1 17:07:00 pipi01 python3[590]:   File "/home/pi/oled_phoniebox/scripts/gpio-buttons/./gpio-buttons.py", line 42, in def_next

rendgeor, 2021-01-01:
Still not solved...
Jan  1 23:45:02 pipi01 mpd: player: played "Sandelf/kinderhoerspiel_der_sandelf_drk_20200823_0805_0c074cf7.mp3"
Jan  1 23:45:02 pipi01 python3[599]: volume: 45%   repeat: off   random: off   single: off   consume: off
**Jan  1 23:45:30 pipi01 mpd: exception: Not playing
Jan  1 23:45:30 pipi01 python3[599]: mpd error: Not playing**
Jan  1 23:46:51 pipi01 rngd[379]: stats: bits received from HRNG source: 120064
Jan  1 23:46:51 pipi01 rngd[379]: stats: bits sent to kernel pool: 64096
Jan  1 23:46:51 pipi01 rngd[379]: stats: entropy added to kernel pool: 64096
Jan  1 23:46:51 pipi01 rngd[379]: stats: FIPS 140-2 successes: 6
Jan  1 23:46:51 pipi01 rngd[379]: stats: FIPS 140-2 failures: 0
Jan  1 23:46:51 pipi01 rngd[379]: stats: FIPS 140-2(2001-10-10) Monobit: 0
Jan  1 23:46:51 pipi01 rngd[379]: stats: FIPS 140-2(2001-10-10) Poker: 0
Jan  1 23:46:51 pipi01 rngd[379]: stats: FIPS 140-2(2001-10-10) Runs: 0
Jan  1 23:46:51 pipi01 rngd[379]: stats: FIPS 140-2(2001-10-10) Long run: 0
Jan  1 23:46:51 pipi01 rngd[379]: stats: FIPS 140-2(2001-10-10) Continuous run: 0
Jan  1 23:46:51 pipi01 rngd[379]: stats: HRNG source speed: (min=160.104; avg=289.789; max=540.373)Kibits/s
Jan  1 23:46:51 pipi01 rngd[379]: stats: FIPS tests speed: (min=10.028; avg=12.622; max=20.777)Mibits/s
Jan  1 23:46:51 pipi01 rngd[379]: stats: Lowest ready-buffers level: 2
Jan  1 23:46:51 pipi01 rngd[379]: stats: Entropy starvations: 0
Jan  1 23:46:51 pipi01 rngd[379]: stats: Time spent starving for entropy: (min=0; avg=0.000; max=0)us
**Jan  1 23:47:21 pipi01 mpd: exception: Not playing
Jan  1 23:47:21 pipi01 python3[599]: mpd error: Not playing**

rendgeor, 2021-01-02:
I did also stop the OLED as it also uses I2C, as the soundcard... but no success.

Jan  2 00:35:48 pipi01 systemd[1]: Stopped OLED Phoniebox Display Service.
**Jan  2 00:35:55 pipi01 kernel: [10152.133972] pcm512x 1-004d: No SCLK, using BCLK: -2**
Jan  2 00:39:01 pipi01 CRON[1073]: (root) CMD (  [ -x /usr/lib/php/sessionclean ] && if [ ! -d /run/systemd/system ]; then /usr/lib/php/sessionclean; fi)
Jan  2 00:39:14 pipi01 systemd[1]: Starting Clean php session files...
Jan  2 00:39:14 pipi01 systemd[1]: phpsessionclean.service: Succeeded.
Jan  2 00:39:14 pipi01 systemd[1]: Started Clean php session files.
Jan  2 00:46:51 pipi01 rngd[379]: stats: bits received from HRNG source: 140064
Jan  2 00:46:51 pipi01 rngd[379]: stats: bits sent to kernel pool: 94336
Jan  2 00:46:51 pipi01 rngd[379]: stats: entropy added to kernel pool: 94336
Jan  2 00:46:51 pipi01 rngd[379]: stats: FIPS 140-2 successes: 7
Jan  2 00:46:51 pipi01 rngd[379]: stats: FIPS 140-2 failures: 0
Jan  2 00:46:51 pipi01 rngd[379]: stats: FIPS 140-2(2001-10-10) Monobit: 0
Jan  2 00:46:51 pipi01 rngd[379]: stats: FIPS 140-2(2001-10-10) Poker: 0
Jan  2 00:46:51 pipi01 rngd[379]: stats: FIPS 140-2(2001-10-10) Runs: 0
Jan  2 00:46:51 pipi01 rngd[379]: stats: FIPS 140-2(2001-10-10) Long run: 0
Jan  2 00:46:51 pipi01 rngd[379]: stats: FIPS 140-2(2001-10-10) Continuous run: 0
Jan  2 00:46:51 pipi01 rngd[379]: stats: HRNG source speed: (min=160.104; avg=310.664; max=547.140)Kibits/s
Jan  2 00:46:51 pipi01 rngd[379]: stats: FIPS tests speed: (min=10.028; avg=13.177; max=20.777)Mibits/s
Jan  2 00:46:51 pipi01 rngd[379]: stats: Lowest ready-buffers level: 2
Jan  2 00:46:51 pipi01 rngd[379]: stats: Entropy starvations: 0
Jan  2 00:46:51 pipi01 rngd[379]: stats: Time spent starving for entropy: (min=0; avg=0.000; max=0)us
Jan  2 00:50:35 pipi01 mpd: player: played "Sandelf/kinderhoerspiel_der_sandelf_drk_20200823_0805_0c074cf7.mp3"
Jan  2 00:50:35 pipi01 python3[599]: volume: 35%   repeat: off   random: off   single: off   consume: off
**Jan  2 00:50:40 pipi01 mpd: exception: Not playing
Jan  2 00:50:40 pipi01 python3[599]: mpd error: Not playing**
Jan  2 00:51:24 pipi01 kernel: [11080.719660] pcm512x 1-004d: No SCLK, using BCLK: -2
Jan  2 00:51:56 pipi01 mpd: player: played "Sandelf/kinderhoerspiel_der_sandelf_drk_20200823_0805_0c074cf7.mp3"
Jan  2 00:51:57 pipi01 python3[599]: volume: 35%   repeat: off   random: off   single: off   consume: off
Jan  2 00:52:07 pipi01 mpd: exception: Not playing
Jan  2 00:52:07 pipi01 python3[599]: mpd error: Not playing
Jan  2 00:52:41 pipi01 mpd: exception: Not playing
Jan  2 00:52:41 pipi01 python3[599]: mpd error: Not playing

 @s-martin : Warum kommt diese Meldung: kernel: [10152.133972] pcm512x 1-004d: No SCLK, using BCLK: -2¨?
 Was bedeutet Jan  2 00:50:40 pipi01 mpd: exception: Not playing
Jan  2 00:50:40 pipi01 python3[599]: mpd error: Not playing¨??

s-martin, 2021-01-02:
The sclk message is just an info that there sound card doesn't use the raspberry clock, see https://support.hifiberry.com/hc/en-us/community/posts/201847081-No-SCLK-using-BCLK-2

s-martin, 2021-01-02:
Please show the outputs of

`cat /proc/asound/modules`

`cat /etc/mpd.conf`

`cat /etc/asound.conf`

`alsamixer`

`amixer scontrols`

`aplay -l`

rendgeor, 2021-01-02:
@s-martin Thanks for supporting me. Strange thing is that audio always starts AND the duration of playing is randomly...

pi@pipi01:~ $ cat **/etc/mpd.conf**
# Files and directories #######################################################
#
# This setting controls the top directory which MPD will search
music_directory		"/home/pi/RPi-Jukebox-RFID/shared/audiofolders"
#
# This setting sets the MPD internal playlist directory.
playlist_directory		"/home/pi/RPi-Jukebox-RFID/playlists"
#
# This setting sets the location of the MPD database.
db_file			"/var/lib/mpd/tag_cache"
#
# These settings are the locations for the daemon log files for the daemon.
log_file			"syslog"
#
# This setting sets the location of the file which stores the process ID
pid_file			"/run/mpd/pid"
#
# This setting sets the location of the file which contains information about most variables
state_file			"/var/lib/mpd/state"
#
# The location of the sticker database.
sticker_file                   "/var/lib/mpd/sticker.sql"
#
###############################################################################

# General music daemon options ################################################
#
# This setting specifies the user that MPD will run as. MPD should never run as
# root and you may use this setting to make MPD change its user ID after
# initialization. This setting is disabled by default and MPD is run as the
# current user.
#
user				"root"
#
# This setting specifies the group that MPD will run as.
#group                          "nogroup"
#
# This setting sets the address for the daemon to listen on.
# For network
bind_to_address		"localhost"
#
# This setting is the TCP port that is desired for the daemon to get assigned to
port				"6600"
#
# This setting controls the type of information which is logged.
log_level			"default"
# log_level			"verbose"
#
# This setting enables automatic update of MPD's database
auto_update    "yes"
#
# Limit the depth of the directories being watched
auto_update_depth "10"
#
###############################################################################

# Input #######################################################################
#

input {
        plugin "curl"
#       proxy "proxy.isp.com:8080"
#       proxy_user "user"
#       proxy_password "password"
}

# QOBUZ input plugin
input {
        enabled    "no"
        plugin     "qobuz"
#        app_id     "ID"
#        app_secret "SECRET"
#        username   "USERNAME"
#        password   "PASSWORD"
#        format_id  "N"
}

# TIDAL input plugin
input {
        enabled      "no"
        plugin       "tidal"
#        token        "TOKEN"
#        username     "USERNAME"
#        password     "PASSWORD"
#        audioquality "Q"
}

# Decoder #####################################################################
#

decoder {
        plugin                  "hybrid_dsd"
        enabled                 "no"
#       gapless                 "no"
}

#
###############################################################################

# Audio Output ################################################################
#
# MPD supports various audio output types
#
# An example of an ALSA output:
#
audio_output {
	type		"alsa"
	name		"Pipis PIFI DAC"
	device		"hw:0,0"	# optional, HIER nur ein Device, keine Subdevices
	mixer_type      "software"      # optional
#	mixer_device	"default"	# optional
#	mixer_control	"Digital"		# optional
#	mixer_index	"0"		# optional
	enabled         "yes"
}

# Normalization automatic volume adjustments ##################################
#
# This setting enables on-the-fly normalization volume adjustment.
volume_normalization		"yes"
#
###############################################################################

# Character Encoding ##########################################################
filesystem_charset		"UTF-8"
#
###############################################################################

pi@pipi01:~ $ cat **/proc/asound/modules**
 0 snd_soc_hifiberry_dacplus

pi@pipi01:~ $ cat /etc/asound.conf 
cat: /etc/asound.conf: No such file or directory

>  FRÜHER: Raspbian Jessie: sudo nano /etc/asound.conf (https://learn.adafruit.com/usb-audio-cards-with-a-raspberry-pi/updating-alsa-config)
> -Earlier versions of Raspbian store sound settings in different places, with different syntax… For Raspbian Jessie, run sudo nano /etc/asound.conf

pi@pipi01:~ $ **amixer scontrols**
Simple mixer control 'DSP Program',0
Simple mixer control 'Analogue',0
Simple mixer control 'Analogue Playback Boost',0
Simple mixer control 'Auto Mute',0
Simple mixer control 'Auto Mute Mono',0
Simple mixer control 'Auto Mute Time Left',0
Simple mixer control 'Auto Mute Time Right',0
Simple mixer control 'Clock Missing Period',0
Simple mixer control 'Deemphasis',0
Simple mixer control 'Digital',0
Simple mixer control 'Max Overclock DAC',0
Simple mixer control 'Max Overclock DSP',0
Simple mixer control 'Max Overclock PLL',0
Simple mixer control 'Volume Ramp Down Emergency Rate',0
Simple mixer control 'Volume Ramp Down Emergency Step',0
Simple mixer control 'Volume Ramp Down Rate',0
Simple mixer control 'Volume Ramp Down Step',0
Simple mixer control 'Volume Ramp Up Rate',0
Simple mixer control 'Volume Ramp Up Step',0

pi@pipi01:~ $ **aplay -l**
**** List of PLAYBACK Hardware Devices ****
card 0: sndrpihifiberry [snd_rpi_hifiberry_dacplus], device 0: HiFiBerry DAC+ HiFi pcm512x-hifi-0 [HiFiBerry DAC+ HiFi pcm512x-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

s-martin, 2021-01-02:
You could set `log_level			"default"` temporarily to  `log_level			"verbose"` (maybe with a reboot) to get more info on n the logfile.

rendgeor, 2021-01-03:
@s-martin: Why is the command exited with status code 1?
**Command '/home/pi/RPi-Jukebox-RFID/scripts/playout_controls.sh -c=playernext' returned non-zero exit status 1.**
You should know better why the python script logs to syslog....

Jan  3 15:42:06 pipi01 daemon_rfid_reader.py[582]: volume: 60%   repeat: off   random: off   single: off   consume: off
Jan  3 15:42:07 pipi01 daemon_rfid_reader.py[582]: volume: 60%   repeat: off   random: off   single: off   consume: off
**Jan  3 15:42:07 pipi01 daemon_rfid_reader.py[582]: loading: sprechstunde**
Jan  3 15:42:07 pipi01 daemon_rfid_reader.py[582]: volume: 60%   repeat: off   random: off   single: off   consume: off
Jan  3 15:42:07 pipi01 daemon_rfid_reader.py[582]: volume: 60%   repeat: off   random: off   single: off   consume: off
Jan  3 15:42:07 pipi01 daemon_rfid_reader.py[582]: OK MPD 0.21.4
Jan  3 15:42:07 pipi01 daemon_rfid_reader.py[582]: OK
Jan  3 15:42:07 pipi01 kernel: [  319.184185] pcm512x 1-004d: No SCLK, using BCLK: -2
Jan  3 15:43:06 pipi01 systemd[1]: getty@tty1.service: Succeeded.
Jan  3 15:43:06 pipi01 systemd[1]: getty@tty1.service: Service has no hold-off time (RestartSec=0), scheduling restart.
Jan  3 15:43:06 pipi01 systemd[1]: getty@tty1.service: Scheduled restart job, restart counter is at 2.
Jan  3 15:43:06 pipi01 systemd[1]: Stopped Getty on tty1.
Jan  3 15:43:06 pipi01 systemd[1]: Started Getty on tty1.
**Jan  3 15:43:31 pipi01 mpd: player: played "http://swr-swr-raka11.cast.addradio.de/swr/swr/raka11/mp3/128/stream.mp3"**
Jan  3 15:43:32 pipi01 python3[581]: volume: 60%   repeat: off   random: off   single: off   consume: off
**Jan  3 15:46:10 pipi01 mpd: exception: Not playing
Jan  3 15:46:10 pipi01 python3[581]: mpd error: Not playing**
Jan  3 15:46:10 pipi01 python3[581]: Traceback (most recent call last):
Jan  3 15:46:10 pipi01 python3[581]:   File "/usr/lib/python3/dist-packages/gpiozero/pins/rpigpio.py", line 244, in _call_when_changed
Jan  3 15:46:10 pipi01 python3[581]:     super(RPiGPIOPin, self)._call_when_changed()
Jan  3 15:46:10 pipi01 python3[581]:   File "/usr/lib/python3/dist-packages/gpiozero/pins/local.py", line 143, in _call_when_changed
Jan  3 15:46:10 pipi01 python3[581]:     self.state if state is None else state)
Jan  3 15:46:10 pipi01 python3[581]:   File "/usr/lib/python3/dist-packages/gpiozero/pins/pi.py", line 293, in _call_when_changed
Jan  3 15:46:10 pipi01 python3[581]:     method(ticks, state)
Jan  3 15:46:10 pipi01 python3[581]:   File "/usr/lib/python3/dist-packages/gpiozero/input_devices.py", line 197, in _pin_changed
Jan  3 15:46:10 pipi01 python3[581]:     self._fire_events(ticks, bool(self._state_to_value(state)))
Jan  3 15:46:10 pipi01 python3[581]:   File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 368, in _fire_events
Jan  3 15:46:10 pipi01 python3[581]:     self._fire_activated()
Jan  3 15:46:10 pipi01 python3[581]:   File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 397, in _fire_activated
Jan  3 15:46:10 pipi01 python3[581]:     super(HoldMixin, self)._fire_activated()
Jan  3 15:46:10 pipi01 python3[581]:   File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 344, in _fire_activated
Jan  3 15:46:10 pipi01 python3[581]:     self.when_activated()
Jan  3 15:46:10 pipi01 python3[581]:   File "/home/pi/oled_phoniebox/scripts/gpio-buttons/./gpio-buttons.py", line 42, in def_next
Jan  3 15:46:10 pipi01 python3[581]:     check_call(jukebox4kidsPath+"/scripts/playout_controls.sh -c=playernext", shell=True)
Jan  3 15:46:10 pipi01 python3[581]:   File "/usr/lib/python3.7/subprocess.py", line 347, in check_call
Jan  3 15:46:10 pipi01 python3[581]:     raise CalledProcessError(retcode, cmd)
Jan  3 15:46:10 pipi01 python3[581]: subprocess.CalledProcessError: Command '/home/pi/RPi-Jukebox-RFID/scripts/playout_controls.sh -c=playernext' returned non-zero exit status 1.
Jan  3 15:46:10 pipi01 python3[581]: Traceback (most recent call last):
Jan  3 15:46:10 pipi01 python3[581]:   File "/usr/lib/python3/dist-packages/gpiozero/pins/rpigpio.py", line 244, in _call_when_changed
Jan  3 15:46:10 pipi01 python3[581]:     super(RPiGPIOPin, self)._call_when_changed()
Jan  3 15:46:10 pipi01 python3[581]:   File "/usr/lib/python3/dist-packages/gpiozero/pins/local.py", line 143, in _call_when_changed
Jan  3 15:46:10 pipi01 python3[581]:     self.state if state is None else state)
Jan  3 15:46:10 pipi01 python3[581]:   File "/usr/lib/python3/dist-packages/gpiozero/pins/pi.py", line 293, in _call_when_changed
Jan  3 15:46:10 pipi01 python3[581]:     method(ticks, state)
Jan  3 15:46:10 pipi01 python3[581]:   File "/usr/lib/python3/dist-packages/gpiozero/input_devices.py", line 197, in _pin_changed
Jan  3 15:46:10 pipi01 python3[581]:     self._fire_events(ticks, bool(self._state_to_value(state)))
Jan  3 15:46:10 pipi01 python3[581]:   File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 368, in _fire_events
Jan  3 15:46:10 pipi01 python3[581]:     self._fire_activated()
Jan  3 15:46:10 pipi01 python3[581]:   File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 397, in _fire_activated
Jan  3 15:46:10 pipi01 python3[581]:     super(HoldMixin, self)._fire_activated()
Jan  3 15:46:10 pipi01 python3[581]:   File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 344, in _fire_activated
Jan  3 15:46:10 pipi01 python3[581]:     self.when_activated()
[... cut ...]
Jan  3 15:46:10 pipi01 python3[581]:   File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 344, in _fire_activated
Jan  3 15:46:10 pipi01 python3[581]:     self.when_activated()
Jan  3 15:46:10 pipi01 python3[581]:   File "/home/pi/oled_phoniebox/scripts/gpio-buttons/./gpio-buttons.py", line 42, in def_next
Jan  3 15:46:10 pipi01 python3[581]:     check_call(jukebox4kidsPath+"/scripts/playout_controls.sh -c=playernext", shell=True)
Jan  3 15:46:10 pipi01 python3[581]:   File "/usr/lib/python3.7/subprocess.py", line 347, in check_call

s-martin, 2021-01-03:
What I don't understand in the log in the first post is stated you use the current Phoniebox version 2.2, but in syslog there's an error coming from `gpio_buttons.py`, which isn't used anymore in 2.2.

Did you upgrade from a version <2.1?

rendgeor, 2021-01-03:
No, just did use the one line installer:
cd; rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/master/scripts/installscripts/buster-install-default.sh; chmod +x buster-install-default.sh; ./buster-install-default.sh

Hm, maybe the installer still installs some old stuff??

s-martin, 2021-01-03:
Why do you have a directory `/home/pi/oled_phoniebox/`, we install in `/home/pi/RPi-Jukebox-RFID`

Please post the following outputs:

- `cat /home/pi/RPi-Jukebox-RFID/settings/version-number`
- `cat /home/pi/RPi-Jukebox-RFID/settings/gpio_settings.ini`
- `cat /home/pi/RPi-Jukebox-RFID/scripts/gpio-buttons/gpio-buttons.py`
- `sudo systemctl status phoniebox-gio-control`
- `sudo systemctl status phoniebox-gio-buttons`

rendgeor, 2021-01-03:
Was using the part in the manual: **Oled Display, One line installer**

cd; rm o4p_installer.sh; wget https://raw.githubusercontent.com/splitti/oled_phoniebox/master/scripts/install/o4p_installer.sh; chmod +x o4p_installer.sh; ./o4p_installer.sh

BTW: existing dirs:
<pre>-rw-r--r--  1 pi   pi     297 Dec 30 22:16 PhonieboxInstall.conf
drwxr-xr-x 15 pi   pi    4096 Dec 30 22:27 <font color="#729FCF"><b>RPi-Jukebox-RFID</b></font>
-rw-r--r--  1 pi   pi    5512 Jan  2 10:26 bla
-rwxr-xr-x  1 pi   pi   48510 Dec 30 22:07 <font color="#8AE234"><b>buster-install-default.sh</b></font>
-rwxr-xr-x  1 pi   pi   23429 Dec 30 22:50 <font color="#8AE234"><b>o4p_installer.sh</b></font>
drwxr-xr-x  8 pi   pi    4096 Dec 30 22:51 <font color="#729FCF"><b>oled_phoniebox</b></font>
drwxr-xr-x  2 pi   pi    4096 Dec 30 22:32 <font color="#729FCF"><b>phoniebox_logs</b></font>
</pre>

    **cat /home/pi/RPi-Jukebox-RFID/settings/version-number** --> 2.2

    **cat /home/pi/RPi-Jukebox-RFID/settings/gpio_settings.ini**
[DEFAULT]
enabled: True
[VolumeControl]
enabled: True
Type: TwoButtonControl ;or RotaryEncoder
PinUp: 5
PinDown: 6
pull_up: True
hold_time: 0.3
hold_repeat: True
timeBase: 0.1 ;only for RotaryEncoder
functionCallDown: functionCallVolD
functionCallUp: functionCallVolU
functionCallTwoButtons: functionCallVol0 ;only for TwoButtonControl

[PrevNextControl]
enabled: True
Type: TwoButtonControl
Pin1: 00
Pin2: 01
functionCall1: functionCallPlayerPrev
functionCall2: functionCallPlayerNext
functionCallTwoButtons: None
pull_up: True
hold_time: 0.3
hold_repeat: False

[PlayPause]
enabled: True
Type: Button
Pin: 12
pull_up: True
hold_time: 0.3
functionCall: functionCallPlayerPause

[Shutdown]
enabled: True
Type:  Button
Pin: 13
pull_up: True
hold_time: 2
functionCall: functionCallShutdown

[Volume0]
enabled: False
Type:  Button
Pin: 17
pull_up: True
hold_time: 0.3
functionCall: functionCallVol0

[VolumeUp]
enabled: False
Type:  Button
Pin: 16
pull_up: True
hold_time: 0.3
hold_repeat: True
functionCall: functionCallVolU

[VolumeDown]
enabled: False
Type:  Button
Pin: 19
pull_up: True
hold_time: 0.3
hold_repeat: True
functionCall: functionCallVolD

[NextSong]
enabled: False
Type:  Button
Pin: 26
pull_up: True
hold_time: 0.3
functionCall: functionCallPlayerNext

[PrevSong]
enabled: False
Type:  Button
Pin: 20
pull_up: True
hold_time: 0.3
functionCall: functionCallPlayerPrev

[Halt]
enabled: False
Type:  Button
Pin: 21
pull_up: True
hold_time: 0.3
functionCall: functionCallPlayerPauseForce

    **cat /home/pi/RPi-Jukebox-RFID/scripts/gpio-buttons/gpio-buttons.py** --> No such file or directory

    **sudo systemctl status phoniebox-gio-control** --> Unit phoniebox-gio-control.service could not be found

    **sudo systemctl status phoniebox-gio-buttons** --> Unit phoniebox-gio-buttons.service could not be found

rendgeor, 2021-01-03:
**cat /home/pi/RPi-Jukebox-RFID/scripts/gpio-buttons/gpio-buttons.py** --> No such file or directory

**BUT**
**cat oled_phoniebox/scripts/gpio-buttons/gpio-buttons.py**
#!/usr/bin/python3
from gpiozero import Button
from signal import pause
from subprocess import check_call
from time import sleep

# This script will block any I2S DAC e.g. from Hifiberry, Justboom, ES9023, PCM5102A
# due to the assignment of GPIO 19 and 21 to a buttons

# 2018-10-31
# Added the function on holding volume + - buttons to change the volume in 0.3s interval
#
# 2018-10-15
# this script has the `pull_up=True` for all pins. See the following link for additional info:
# https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/259#issuecomment-430007446
#
# 2017-12-12
# This script was copied from the following RPi forum post:
# https://forum-raspberrypi.de/forum/thread/13144-projekt-jukebox4kids-jukebox-fuer-kinder/?postID=312257#post312257
# I have not yet had the time to test is, so I placed it in the misc folder.
# If anybody has ideas or tests or experience regarding this solution, please create pull requests or contact me.

jukebox4kidsPath = "/home/pi/RPi-Jukebox-RFID"

def def_shutdown():
    check_call(jukebox4kidsPath+"/scripts/playout_controls.sh -c=shutdown", shell=True)

def def_volU():
    check_call(jukebox4kidsPath+"/scripts/playout_controls.sh -c=volumeup", shell=True)

def def_volD():
    check_call(jukebox4kidsPath+"/scripts/playout_controls.sh -c=volumedown", shell=True)

def def_vol0():
    check_call(jukebox4kidsPath+"/scripts/playout_controls.sh -c=mute", shell=True)

def def_next():
    for x in range(0, 19):
        if btn_next.is_pressed == True :
            sleep(0.1)
        else:
            check_call(jukebox4kidsPath+"/scripts/playout_controls.sh -c=playernext", shell=True)
            break

def def_contrastup():
    if btn_prev.is_pressed == True :
        check_call("/usr/bin/touch /tmp/o4p_overview.temp", shell=True)
    else:
        check_call("/usr/bin/python3 /home/pi/oled_phoniebox/scripts/contrast/contrast_up.py", shell=True)

def def_contrastdown():
    if btn_next.is_pressed == True :
        check_call("/usr/bin/touch /tmp/o4p_overview.temp", shell=True)
    else:
        check_call("/usr/bin/python3 /home/pi/oled_phoniebox/scripts/contrast/contrast_down.py", shell=True)

def def_prev():
    for x in range(0, 19):
        if btn_prev.is_pressed == True :
            sleep(0.1)
        else:
            check_call(jukebox4kidsPath+"/scripts/playout_controls.sh -c=playerprev", shell=True)
            break

def def_halt():
    check_call(jukebox4kidsPath+"/scripts/playout_controls.sh -c=playerpause", shell=True)
	
def toggle_display():
    check_call("/home/pi/oled_phoniebox/scripts/toggle_display/toggle_display.sh", shell=True)

#btn_shut = Button(3, hold_time=2)
#btn_vol0 = Button(21,pull_up=True)
btn_volup = Button(16,pull_up=True,hold_time=0.3,hold_repeat=True)
btn_voldown = Button(13,pull_up=True,hold_time=0.3,hold_repeat=True)
btn_next = Button(26,pull_up=True,hold_time=2.0,hold_repeat=False)
btn_prev = Button(27,pull_up=True,hold_time=2.0,hold_repeat=False)
btn_halt = Button(12,pull_up=True,hold_time=2.0,hold_repeat=False)

#btn_shut.when_held = def_shutdown
#btn_vol0.when_pressed = def_vol0
btn_volup.when_pressed = def_volU
#When the Volume Up button was held for more than 0.3 seconds every 0.3 seconds he will call a ra$
btn_volup.when_held = def_volU
btn_voldown.when_pressed = def_volD
#When the Volume Down button was held for more than 0.3 seconds every 0.3 seconds he will lower t$
btn_voldown.when_held = def_volD
btn_next.when_pressed = def_next
btn_next.when_held = def_contrastup
btn_prev.when_pressed = def_prev
btn_prev.when_held = def_contrastdown
btn_halt.when_pressed = def_halt
btn_halt.when_held = toggle_display

pause()

s-martin, 2021-01-03:
> **sudo systemctl status phoniebox-gio-control** --> Unit phoniebox-gio-control.service could not be found
> 
> **sudo systemctl status phoniebox-gio-buttons** --> Unit phoniebox-gio-buttons.service could not be found

Sorry, there was a typo, please use:
`sudo systemctl status phoniebox-gpio-control`

`sudo systemctl status phoniebox-gpio-buttons`

rendgeor, 2021-01-03:
pi@pipi01:~ $ sudo systemctl status phoniebox-gpio-control
● phoniebox-gpio-control.service - Phoniebox GPIO Control Service
   Loaded: loaded (/etc/systemd/system/phoniebox-gpio-control.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2021-01-03 15:37:06 GMT; 6h ago
 Main PID: 596 (python3)
    Tasks: 2 (limit: 3861)
   CGroup: /system.slice/phoniebox-gpio-control.service
           └─596 python3 /home/pi/RPi-Jukebox-RFID/components/gpio_control/gpio_control.py

Jan 03 15:37:07 pipi01 PhonieboxGPIOControl[596]: INFO:GPIODevices.simple_button:add new action
Jan 03 15:37:07 pipi01 PhonieboxGPIOControl[596]: INFO:root:adding GPIO-Device, PlayPause
Jan 03 15:37:07 pipi01 PhonieboxGPIOControl[596]: INFO:root:adding GPIO-Device, Shutdown
Jan 03 15:37:07 pipi01 PhonieboxGPIOControl[596]: INFO:root:Device Volume0 not enabled
Jan 03 15:37:07 pipi01 PhonieboxGPIOControl[596]: INFO:root:Device VolumeUp not enabled
Jan 03 15:37:07 pipi01 PhonieboxGPIOControl[596]: INFO:root:Device VolumeDown not enabled
Jan 03 15:37:07 pipi01 PhonieboxGPIOControl[596]: INFO:root:Device NextSong not enabled
Jan 03 15:37:07 pipi01 PhonieboxGPIOControl[596]: INFO:root:Device PrevSong not enabled
Jan 03 15:37:07 pipi01 PhonieboxGPIOControl[596]: INFO:root:Device Halt not enabled
Jan 03 15:37:07 pipi01 PhonieboxGPIOControl[596]: INFO:root:Ready for taking actions

pi@pipi01:~ $ sudo systemctl status phoniebox-gpio-buttons
● phoniebox-gpio-buttons.service - Phoniebox GPIO Buttons Service
   Loaded: loaded (/etc/systemd/system/phoniebox-gpio-buttons.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2021-01-03 15:37:06 GMT; 6h ago
 Main PID: 581 (python3)
    Tasks: 7 (limit: 3861)
   CGroup: /system.slice/phoniebox-gpio-buttons.service
           └─581 /usr/bin/python3 /home/pi/oled_phoniebox/scripts/gpio-buttons/./gpio-buttons.py

Jan 03 22:01:42 pipi01 python3[581]:     self._fire_activated()
Jan 03 22:01:42 pipi01 python3[581]:   File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 397, in _fire_activated
Jan 03 22:01:42 pipi01 python3[581]:     super(HoldMixin, self)._fire_activated()
Jan 03 22:01:42 pipi01 python3[581]:   File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 344, in _fire_activated
Jan 03 22:01:42 pipi01 python3[581]:     self.when_activated()
Jan 03 22:01:42 pipi01 python3[581]:   File "/home/pi/oled_phoniebox/scripts/gpio-buttons/./gpio-buttons.py", line 42, in def_next
Jan 03 22:03:03 pipi01 python3[581]:     check_call(jukebox4kidsPath+"/scripts/playout_controls.sh -c=playernext", shell=True)mpd error: Not playing
Jan 03 22:03:16 pipi01 python3[581]: mpd error: Not playing
Jan 03 22:06:50 pipi01 python3[581]: mpd error: Not playing
Jan 03 22:08:55 pipi01 python3[581]: mpd error: Not playing

s-martin, 2021-01-03:
I'm not sure what oled_phoniebox uses of original Phoniebox and if they should be installed in parallel. Maybe @splitti can help here.

However Phoniebox 2.2 uses a different implementation for GPIO now and oled_phoniebox uses the old implementation, I'm not sure, if this works.

s-martin, 2021-01-03:
As you said you don't use any buttons you could disable one of the services, e.g.
`sudo systemctl stop phoniebox-gpio-control`
`sudo systemctl disable phoniebox-gpio-control`

rendgeor, 2021-01-03:
ok, I was just following splittis Howto;) Yes, but next step is to connect real buttons...

I was modifying the OLED gpio-buttons.py to use the same keys as gpio_settings.ini. Seems this at least solves the problem, that the phoniebox-gpio-buttons.service now runs without warnings.

In General: Why can the two setting files be merged??

closed 2021-01-03 by rendgeor