MiczFlor/RPi-Jukebox-RFID #1259: 🚀 | Phniebox running as non root user

rendgeor, 2021-01-03:
## Feature Description

### What functionality would you like to see in your phoniebox?

Run phoniebox as normal user. I want to get rid of phoniebox commands show up in auth.log
Is that possible?

### How do you envision the feature to work from a users perspective?
pi@pipi01:/var/log $ sudo tail -n 10 **auth.log**
Jan  3 23:22:12 pipi01 sudo: pam_unix(sudo:session): session opened for user root by (uid=0)
Jan  3 23:22:12 pipi01 sudo: pam_unix(sudo:session): session closed for user root
Jan  3 23:22:12 pipi01 sudo: www-data : TTY=unknown ; PWD=/home/pi/RPi-Jukebox-RFID/htdocs/api ; USER=root ; COMMAND=/home/pi/RPi-Jukebox-RFID/scripts/playout_controls.sh -c=getvolume
Jan  3 23:22:12 pipi01 sudo: pam_unix(sudo:session): session opened for user root by (uid=0)
Jan  3 23:22:12 pipi01 sudo: pam_unix(sudo:session): session closed for user root
Jan  3 23:22:13 pipi01 sudo: www-data : TTY=unknown ; PWD=/home/pi/RPi-Jukebox-RFID/htdocs/api ; USER=root ; COMMAND=/home/pi/RPi-Jukebox-RFID/scripts/playout_controls.sh -c=getchapters
Jan  3 23:22:13 pipi01 sudo: pam_unix(sudo:session): session opened for user root by (uid=0)
Jan  3 23:22:13 pipi01 sudo: pam_unix(sudo:session): session closed for user root
Jan  3 23:22:16 pipi01 sudo:       pi : TTY=pts/0 ; PWD=/var/log ; USER=root ; COMMAND=/usr/bin/tail -n 10 auth.log
Jan  3 23:22:16 pipi01 sudo: pam_unix(sudo:session): session opened for user root by pi(uid=0)

s-martin, 2022-05-29:
Related to #1785

s-martin, 2023-12-05:
Was fixed for future3 in #2132

closed 2023-12-05 by s-martin