MiczFlor/RPi-Jukebox-RFID #1266: 🐛 | Can someone verify the latest Raspberry OS version not working with RPi-Jukebox + Spotify?

Piehti, 2021-01-08:
## Bug

### What I did

Installed the latest Raspberry OS + Updates + Upgrades
After I installed the jukebox (WITH SPOTIFY) via the one line install command and also tryed the dev version after several fails.
In all cases Mopidy was setup and authorized properbly.

### What happened

The installation stumbled upon a lot of strabge errors - no matter what...

I.e. :

`Generating locales (this might take a while)...
  en_GB.ISO-8859-1... done
`

while:

`
perl: warning: Please check that your locale settings:
        LANGUAGE = (unset),
        LC_ALL = (unset),
        LANG = "de_DE.UTF-8"
    are supported and installed on your system.
`

AFAIK my locale id DE, DE. I'm aware of that stupid initial "change timezone"-bug while doing the initial setup of the pie but highly doubt that I did stumble into it on any try.

Also I went through:

`
Could not install packages due to an EnvironmentError: HTTPSConnectionPool(host='files.pythonhosted.org', port=443): Max retries exceeded with url: /packages/88/0f/5cdb31cdbcaca0af24958ecbddde00717c219ba775edbc1cbfec30a6f834/youtube_dl-2021.1.8-py2.py3-none-any.whl (Caused by NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x756a71d0>: Failed to establish a new connection: [Errno -3] Temporärer Fehler bei der Namensauflösung'))
`

or

`Please select the RFID reader you want to use
1 USB-Reader (e.g. Neuftech)  3) PN532                       5) Multiple RFID reader
2 RC522                       4) Manual configuration
#? 1
Traceback (most recent call last):
  File "RegisterDevice.py", line 4, in <module>
    from Reader import get_devices
  File "/home/pi/RPi-Jukebox-RFID/scripts/Reader.py", line 23, in <module>
    from evdev import InputDevice, ecodes, list_devices
ModuleNotFoundError: No module named 'evdev'
chown: cannot access '/home/pi/RPi-Jukebox-RFID/scripts/deviceName.txt': No such file or directory
chmod: cannot access '/home/pi/RPi-Jukebox-RFID/scripts/deviceName.txt': No such file or directory`

### I expected this to happen

Flawless easy install as usual. 

### Further information that might help

## Software

### Base image and version

2020-12-02-raspios-buster-armhf-lite

### Branch / Release

On branch master
Your branch is up to date with 'origin/master'.

### Installscript

cd; rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/master/scripts/installscripts/buster-install-default.sh; chmod +x buster-install-default.sh; ./buster-install-default.sh

## Hardware

### RaspberryPi version

Raspberry Pi 3 Model B Rev 1.2

### RFID Reader

The suggested Neuftech HXGCoLtd

### Soundcard

3,5-Jack and / or HDMI neighter works.

### Other notable observations:

~/RPi-Jukebox-RFID $ sudo journalctl -u mopidy
-- Logs begin at Fri 2021-01-08 21:52:34 CET, end at Fri 2021-01-08 22:32:29 CET. --
-- No entries --

~/RPi-Jukebox-RFID $ sudo tail -n 500 /var/log/syslog

Lot of errors are like:

`PIDFile= references path below legacy directory /var/run/, updating /var/run/samba/smbd.pid → /run/samba/smbd.pid; please update the unit file accordingly.`

Issue #1: Script fails to fill my Spotify password properbly in mopidy.conf
Reason: (guess) Special Characters
Solution: Added the password manually to mopidy.conf

Issue #1: Script fails to fill my WiFi password properbly in wpa_supllicant file
Reason: (guess) Special Characters
Solution: Denyed the script to configure WiFi or resupplying the right data into the wpa_supplicant BEFORE rebooting

Issue #2: Raspberry OS release seems to have f*cked up the sound device interface names AGAIN (pun me if I'm wrong). At least the alsa output for the HDMI interface seems to be called "b1" instead of HDMI now, while they left Headphones as it was.
Reason: Dunno. Ask the Raspberry OS maintainers. Could have to do with the Raspi 4 dual HDMI compability or my lesser linux skills.
Solution: Got none. Using the 3,5-Jack for now.

### Workarounds:

Propably going to use an older Raspian image from backups or an working Buster/Jukebox4kids image without spotify.

s-martin, 2021-01-08:
`Could not install packages due to an EnvironmentError: HTTPSConnectionPool(hos`
This error looks like the server to download the package could not be reached. I would assume this is temporarily and you should try at a later time.

This could lead to subsequent errors

s-martin, 2021-01-08:
In 
`/home/pi/phoniebox_logs`
the installer script logs everything.

Could you post one completely?

s-martin, 2021-02-28:
Is this still an issue?

sebrep, 2021-07-16:
I just installed a new phoniebox with a raspberry pi zero with version 2.2 - 305325d - master on raspbian buster and had no problems at all. (other than the IRQ cable with rfc522 so I made a pull request on the readme file ;)

s-martin, 2021-07-17:
Closing as the question was answered.
Feel free to reopen, if necessary.

closed 2021-07-17 by s-martin