MiczFlor/RPi-Jukebox-RFID #1267: Seek Forward and Backward

Kyokuhin-b, 2021-01-09:
Hi everybody,

i´m new on this Forum and I'm a beginner with Phoniebox.

I tried to implement a new button to seek Forward and Backward.
But on "RPi-Jukebox-RFID/settings/gpio_settings.ini" I Just can use "VolumenControl", "PrevNextContro"l and "PlayPause" on the standart PINs. The other Pins doesn't works.

So I think I have to activate the diferents PINs on "RPi-Jukebox-RFID/scripts/gpio-buttons.py", but I can't find the "gpio-buttons.py".

My next step is to replace two Buttons, one for NextTrack and one for SeekFwd with one which do both things. NextTrack with a simple press and SeekFwd holding the Butoons.

Can somobody help me or maybe it is implemented yet but i don't know about this feature.

s-martin, 2021-01-09:
Please have a look at the examples folder:
https://github.com/MiczFlor/RPi-Jukebox-RFID/tree/develop/components/gpio_control/example_configs

The example file should help you to find configurations for your need. Basically you can use any GPIO pin.

Kyokuhin-b, 2021-01-09:
Hi and thank you!

I know the Example Folder but the pins also don’t working for me

s-martin, 2021-01-09:
I guess you have already seen https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Using-GPIO-hardware-buttons

One important thing to note: for `Pin` you need to enter the **GPIO number**!

Kyokuhin-b, 2021-01-10:
Hi,

I found my problem. The buttons wasn't enabled...

But I still habe problems. I can't change the hold-time from 0.3 to another value like 2

s-martin, 2021-01-12:
What are you trying to do exactly and what doesn't work?

Kyokuhin-b, 2021-01-13:
Übersetzungstypen
Textübersetzung
Ausgangstext
1003 / 5000
Übersetzungsergebnisse
So, I'll use the translator now, I hope that you will understand me better that way and that I am not sitting on a message for what feels like 3 hours.

I have integrated a button each for "PlayerNext" and "PlayerPrev".

Now I would like to realize that if you press it briefly, you fast-forward ("seekFwd") and if you hold it down, it jumps to the next track ("PlayerNext"). Unfortunately, I can't get the PIN assigned twice. I guess I have to program it, unfortunately I don't really know which file is in which it has to be done.

Unfortunately I haven't found this project anywhere.

The other thing I meant was that I tried in the file:
"RPi-Jukebox-RFID / settings / gpio_settings.ini" to extend the duration of the key press until the function is executed.
So I increased the "hold_time" value to "1". Nevertheless, he always switched after a few milliseconds.

Can you please help me with this?

splitti, 2021-02-25:
I am also interested in a solution for different function-calls for short and long pressed buttons. This was possible in the old version...