MiczFlor/RPi-Jukebox-RFID #1268: launch playlist with gpio

tibex44, 2021-01-10:
Hello,

After several tests I finally managed to get my Pi3 to work with Phoniebox 2.2.
the pause, next and previous buttons function but I cannot launch a playlist with a button.

is it possible to modify the gpio_settings.ini file to read a directory?

s-martin, 2021-01-10:
You would need to modify or add a function call here
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/develop/components/gpio_control/function_calls.py

and use that function call in your settings file.

s-martin, 2021-01-10:
Please see also #1221

tibex44, 2021-01-10:
Thank you for your responsiveness but I have a lot of trouble with the coding ...

**1- Define a function call in the gpio_settings.ini** 

[PlayFolder]
enabled: True
Type: Button
Pin: 32
pull_up: True
hold_time: 0.3
functionCall: functionCallPlayerFolder

**-2 added function call in function_calls.py**

def functionCallPlayerFolder(*args):
    function_call("{command} -c=playerfolder".format(command=playout_control), shell=True)

how can i give it the path to the folder  "/home/pi/RPi-Jukebox-RFID/shared/audiofolders/comptine" ?

s-martin, 2021-01-10:
Did you have a look at https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Hack-Playlist-triggered-by-GPIO-Button

tibex44, 2021-01-10:
yes but I did not find how to assign an RFID number to a file. (I don't have an RFID reader and I don't want one)
thank you for your help. I will investigate this lead

closed 2024-03-01 by AlvinSchiller