MiczFlor/RPi-Jukebox-RFID #1270: 🐛 | bash: id: command not found

Andreas-Roth-27, 2021-01-11:
When I swipe the RFID chip over the reader its id is shown in the RPi console (terminal) but there is  an output 
bash: id: command not found 
(correct id of the RFID chip)
The RFID is not used in web interface of the phoniebox. What is wrong?

Andreas-Roth-27, 2021-01-20:
Wer könnte denn bitte hier mal hineinschauen?

MiczFlor, 2021-01-20:
I don't understand the problem. The RFID chip is registered via the web app to do something or play something?
If you swipe the card and look at the RPi terminal, what happens is that you see the ID and then there is a "virtual" keystroke: enter. So the RPi thinks you typed a command. But you typed the rfid ID, a few numbers.
So RPi replies: command not found
because a string of numbers like 3456273845
is not a command.
Is that what you mean?

Andreas-Roth-27, 2021-01-20:
Yes. 
I swipe the card and then I can see the id 3456273845 in the terminal window.
What I see is exactly "bash: 3456273845: command not found". That's all what is happening.
The id does not make it to the web app.

Andreas-Roth-27, 2021-01-20:
So my little son is already admiring his wooden radio as he calls it. But I think the RDID function is the hit and is unfortunately missing...
Thank you for your help in advance!!

Andreas-Roth-27, 2021-01-24:
Any idea?

MiczFlor, 2021-01-25:
Can you check in the web app, if the RFID reader is switched on? 
It's under settings.
Here you can see the section:
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/MANUAL#external-devices--interfaces

Andreas-Roth-27, 2021-01-25:
I already checked it a few weeks ago. It seemed ok. 
But now looking a bit more focussed I found a weird behavior. It was switched off when I checked it in the first place. Then I switch it on (clearly indicated by green and red squares). Then I swipe the RFID chip. It beeps. And when I refresh the settings page in the web app the RFID reader is disabled. As if the swipe of the RFID chip switches off the reader. And by doing so has a totally different function. I can repeat this over and over again.

Andreas-Roth-27, 2021-01-26:
Apparently the web app reacts on the swipe but not in the way anticipated

Andreas-Roth-27, 2021-02-01:
Web App says RFID reader is switched off
I switch it ON
I swipe the card over the reader (beep)
Web App says (after a refresh of the page) RFID reader is OFF

That sounds odd. Is there at least an educated guess what that could be? Does sound like a bug.
Any idea, please?!

MiczFlor, 2021-02-01:
Hi @Andreas-Roth-27 
it's very difficult to really understand. It sounds to me that the reader you are using is not the one that you have told the Phoniebox to look for. Then the service stops, because it throws an error.
How did you install your Phoniebox?
You can manually register a USB device manually like this:
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/CONFIGURE-stretch#register-your-usb-device-for-the-phoniebox
What RFID reader do you use?

Andreas-Roth-27, 2021-02-01:
I installed the phoniebox with the 1-line-installer (
cd; rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/master/scripts/installscripts/buster-install-default.sh; chmod +x buster-install-default.sh; ./buster-install-default.sh
)
The RFID reader is this one "Neuftech USB RFID Reader ID Kartenlesegerät Kartenleser Kontaktlos Card Reader für EM4100".

Andreas-Roth-27, 2021-02-01:
Hey hey hurra. That was it. Now the RFID does not go OFF when I swipe a chip. But instead changes the media. Great! 
Thank you!! 
I manually installed the device with the hint referred to with your above mentioned link

closed 2021-02-01 by Andreas-Roth-27

jukeben, 2021-05-19:
Ich hab das selbe Problem! Der RFID reader wird angezeigt aber in der web app die ID nicht. Seltsam ist dass ich die Box schon am laufen hatte dann aber neu aufgesetzt habe und jetzt die webb app mir keine ID liefert wenn ich eine Karte auf den Reader lege.

jukeben, 2021-05-21:
> Ich hab das selbe Problem! Der RFID-Leser wird in der Web-App die ID nicht erstellt. Seltsam ist, dass ich die Box schon am laufen habe, dann aber neu auf habe habe und jetzt die webb App mir keine ID gegeben wenn ich eine Karte auf den Reader lege.

Erledigt!