MiczFlor/RPi-Jukebox-RFID #1272: Web Interface funktioniert nur an der Phoniebox selber

OlafStyle, 2021-01-11:
Hallo da draußen,

ich habe ein kleines aber stark nerviges Problem mit ner neuen Phoniebox und weiß leider nicht wie ich des beheben soll.
Das Problem ist nämlich, dass das Web-Interface nur an der Phoniebox selber funktioniert, nicht aber an anderen PC`s.
Und selbst wenn ich versuche eine VNC verbindung aufzubauen klappt des au ned.
Zusätzlich ist immer wenn ich die Phoniebox starte wieder das Power-Management an, gibt es evtl eine Lösung mit der ich das für immer ausschalten kann?

Mit freundlichen Füßen
Leon

MiczFlor, 2021-01-20:
Is the Phoniebox in the WiFi network?
What is the IP address of the Phoniebox?
Did you type the IP address into the browser of the PC (which is in the same Wifi Network)?

s-martin, 2023-05-01:
Closing this issue for now. Feel free to reopen, if the problem persists

closed 2023-05-01 by s-martin