MiczFlor/RPi-Jukebox-RFID #1275: 🚀 | Firewall or VPN for public wifi

topas-rec, 2021-01-12:
## Feature Description

### What functionality would you like to see in your phoniebox?

Do we need a firewall or VPN?

### How do you envision the feature to work from a users perspective?

Firewall and/or VPN is an option during install, or an option in the web interface.

### Further information that might help

When using public wifi one should use a trusted VPN provider or at least a firewall beside other things (encryption and up to date software,...).
When I was on holiday last year I used the Phoniebox in a public wifi (freifunk) and enabled a firewall for the box. I didn't used VPN.
Setting up the firewall was pretty easy (used firewalld, used standard services for http and SSH).
I always wanted to ask the Phoniebox community about that addition.

What do you think?

s-martin, 2021-01-12:
If anyone's willing to implement it, I think we could add it.

But frankly I don't see a big threat scenario using a Phoniebox in a public WiFi. What's the worst an attacker could get? Your music taste 😉

topas-rec, 2021-01-18:
I won't have time to do it. Sorry. I just wanted to start this discussion. I also think it is not something really necessary, but maybe someone alse comes by and want to do it. I think there is currently no interest in this and we'll close it so we don't see it in the list of issues anymore.

s-martin, 2021-01-18:
You could leave it open, so someone interested could pick it up.
There are lots of issues tagged with enhancement open, so people can read new ideas and implement them.