MiczFlor/RPi-Jukebox-RFID #1276: No Shutdown with OnOffShim

thiyyas, 2021-01-12:
I don't have a Phonie-Box-Bug, it is more a question to experienced Users/developers hoping to find a solution for my problem. 
If this is not the right place for this kind of questions, please let me know.

## Problem:
By Pressing the (internal) button on OnOffShim the Raspberry Pi is turning on correctly. Pressing the button for Shutdown has no effect. 

When I installed everything for the first time some months ago, Shutdown and start of the Phoniebox did work well over OnOffShim. I rember that there was also some kind of LED-Signal (blinking) on OnOff-Shim, when the button was pressed long enough to shut down. 

Now I get with each pressing of the small Button on the OnOffShim a red LED for the duration of the press. Same for an external Button connected to the OnOffShim. 

The OnOffShim is soldered on the header which is put directly on the GPIOs (no double use). I also tried it wih an extension board (with that I only use 3,3V and Ground doubled). I checked the soldering between OnOffShime and header, but I m not sure how to measure while the RaspBerryPi is running. 

### What I did
I installed Phoniebox newly a few days ago (with OneLine-Installer) after installing Buster on Raspberry Pi from scratch. 
OnOff Shim was installed with the OneLine Installer: curl https://get.pimoroni.com/onoffshim | bash
This are the settings: 

Cleanshutd.conf: 
```
daemon_active=1
trigger_pin=17
led_pin=17
poweroff_pin=4
hold_time=1
shutdown_delay=0
polling_rate=1
```

For Led_Pin I normally want to use GPIO 13 with an external LED, which is working well, because of the issue here I removed all other connections (LED/Buttons/RotaryEncoder) from the GPIOS to be sure not to have any interferences. 

gpio_settings.ini: 
```
[DEFAULT]
enabled: True
[VolumeControl]
enabled: True
Type: RotaryEncoder ;or  TwoButtonControl
PinUp: 5
PinDown: 6
pull_up: True
hold_time: 0.3
hold_repeat: True
timeBase: 0.1 ;only for RotaryEncoder
functionCallDown: functionCallVolD
functionCallUp: functionCallVolU
;functionCallTwoButtons: functionCallVol0 ;only for TwoButtonControl

[PrevNextControl]
enabled: False
Type: TwoButtonControl
Pin1: ;22
Pin2: ;23
functionCall1: functionCallPlayerPrev
functionCall2: functionCallPlayerNext
functionCallTwoButtons: None
pull_up: True
hold_time: 0.3
hold_repeat: False

[PlayPause]
enabled: True 
Type: Button
Pin: 27
pull_up: True
hold_time: 0.3
functionCall: functionCallPlayerPause

[Shutdown]
enabled: False
Type:  Button
Pin: ;3
pull_up: True
hold_time: 2
functionCall: functionCallShutdown

[Volume0]
enabled: False
Type:  Button
Pin: ;17
pull_up: True
hold_time: 0.3
functionCall: functionCallVol0

[VolumeUp]
enabled: False
Type:  Button
Pin: ;16
pull_up: True
hold_time: 0.3
hold_repeat: True
functionCall: functionCallVolU

[VolumeDown]
enabled: False
Type:  Button
Pin: ;19
pull_up: True
hold_time: 0.3
hold_repeat: True
functionCall: functionCallVolD

[NextSong]
enabled: True
Type:  Button
Pin: 26
pull_up: True
hold_time: 0.3
functionCall: functionCallPlayerNext

[PrevSong]
enabled: True
Type:  Button
Pin: 20
pull_up: True
hold_time: 0.3
functionCall: functionCallPlayerPrev

[Halt]
enabled: False
Type:  Button
Pin: ;21
pull_up: True
hold_time: 0.3
functionCall: functionCallPlayerPauseForce
```

I disabled [shutdown] because I use the OnOffShim, but I also tried setting this value to GPIO 17, as it is the trigger pin if I got it correctly. 

I also tried different durations of holding the button (<1 sec, 1 sec, 2 sec, 5 sec and 10 sec). But no different behaviour. 

Following three ways to shut down are working correctly and immediatly: 
```
pi@raspyberrypi ~/RPi-Jukebox-RFID $ ./scripts/playout_controls.sh -c=shutdown
sudo: lsof: command not found
sudo: lsof: command not found
sudo: lsof: command not found
volume: 30%   repeat: off   random: off   single: off   consume: off
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
        version 1.25.10; written and copyright by Michael Hipp and others
        free software (LGPL) without any warranty but with best wishes

Directory: /home/pi/RPi-Jukebox-RFID/scripts/../shared/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: shutdownsound.mp3 ...

MPEG 1.0 L III cbr64 44100 mono

[0:02] Decoding of shutdownsound.mp3 finished.
```
```
pi@raspberrypi:~/RPi-Jukebox-RFID $ sudo shutdown -h 0
Shutdown scheduled for Tue 2021-01-12 19:36:59 GMT, use 'shutdown -c' to cancel.
```
```
pi@raspberrypi:~/RPi-Jukebox-RFID $ sudo /lib/systemd/system-shutdown/gpio-poweroff poweroff
Reading config file /etc/cleanshutd.conf
Using power off pin 4
Using LED pin 17
```

But with all three ways I cannot restart the Phoniebox with the OnOffShim-Button afterwards. I first need to disconnect the OnOffShim shortly from the Powerbank. 

When Using "Herunterfahren" in the Phoniebox-UI I can restart it afterwards by using the Button on the OnOffShim. 
In all three cases the LEDs on the RaspBerryPi are off. 

The Status of Cleanshutd is: 
```
pi@raspberrypi:~/RPi-Jukebox-RFID $ service cleanshutd status
● cleanshutd.service - LSB: Monitoring GPIO shutdown trigger
   Loaded: loaded (/etc/init.d/cleanshutd; generated)
   Active: active (running) since Tue 2021-01-12 19:41:17 GMT; 18min ago
     Docs: man:systemd-sysv-generator(8)
  Process: 371 ExecStart=/etc/init.d/cleanshutd start (code=exited, status=0/SUCCESS)
    Tasks: 2 (limit: 2063)
   CGroup: /system.slice/cleanshutd.service
           ├─ 450 /bin/bash /usr/bin/cleanshutd
           └─9560 sleep 1

Jan 12 19:41:16 musikbox systemd[1]: Starting LSB: Monitoring GPIO shutdown trigger...
Jan 12 19:41:16 musikbox cleanshutd[371]: Starting...
Jan 12 19:41:17 musikbox systemd[1]: Started LSB: Monitoring GPIO shutdown trigger.
```
It seems that also the right Pin is monitored (when pressing the button while this is running, nothing happens, but I m not sure, if something should happen here):
```
pi@musikbox:~/RPi-Jukebox-RFID $ /usr/bin/./cleanshutd
monitoring BCM 17
```

I tried: 
- reinstalling OnOffShim
- reinstalling Phoniebox
- new OnOffShim in case of hardware defect, but the new one has the same behaviour
- with and without ExtensionBoard
- with and without all Buttons/LEDs/RotaryEncoder

My last Idea is oly to buy a new RasberryPi to be sure, that there is no Hardware-Defect on this specific GPIOs. But befor I get an new RaspberryPi, I hope someone has an idea, how to solve this problem. Maybe I m just blind. 
If I missed any needed information, please let me know. 

Thanks in advance for your help and sorry, if this is not the correct place for this kind of questions.

s-martin, 2021-01-12:
Maybe you have a cold solder joint, which works in the beginning, but can make problems over time.
https://learn.adafruit.com/adafruit-guide-excellent-soldering/common-problems

thiyyas, 2021-01-13:
I did all soldering on the OnOffShim newly and for me it looks ok, not pretty but ok. 
I also checked the link already previously, it seems to be ok, but i m not an expert in soldering. 

This was also a reason, why i bought a second onoffshim. I wanted to be sure, it was not the Hardware and not the soldering. 

I also tried to measure the Connections. From the Header to the solderpoints it is ok, but i have no idea how to check if the connection from the solderpoint to the platine itself. 

Thanks for your effort.

thiyyas, 2021-01-13:
I just found the problem after several different trys. 

If someone has some similar problem - maybe this helps. 
I wanted to buy a new RaspberryPi but then I found a easy way to check my GPIOS (all GPIOs need to be unconnected): 
http://abyz.me.uk/rpi/pigpio/faq.html#Have_I_fried_my_GPIO: (no idea, if this steps are totally corerct, for me it worked)
- Downloading the file "abyz.me.uk/rpi/pigpio/code/gpiotest.zip", to RaspberryPi
- unzip and save 
- "sudo apt install pigpio" 
- "sudo pigpiod"
- "sudo ./gpiotest" in the folder, where the file is saved

This showed that my GPIO 14 is not working, but GPIO 14 is not used by the OnOffShim. 
But I lost a bit of hesitation knowing that already a GPIO is not working anymore. 
I connected the OnOffShim with cables instead of pluging it directly on the RaspberryPi and tried to use different Pins. 
At the end, it seems that my GPIO 17 is not working (even if gpiotest did not recognize any issues there).
Connecting the Pin 11 of OnOffShim to GPIO18 solved my problem (after adjusting "cleanshutd.conf"-file (replaceing 17 by 18).

Now I just need to find a way of properly soldering OnOffShim to my Extensionboard, because the header cannot be used anymore. 

Thanks for checking on my issue.

closed 2021-01-13 by thiyyas