MiczFlor/RPi-Jukebox-RFID #1278: TTS Feature: Read track titles aloud for non-display phonieboxes

Caliandroid, 2021-01-16:
One idea i've had a while ago:
Track title will be read aloud before the track starts playing.

Use-case: 

- One folder with a lot of children audiobooks linked to one single rfid card
     -  example: Benjamin Blümchen, 15 stories. My kids have to wait a while to figure out which story is currently playing to find a particular track without display (especially if they are still too young to read).
     -  the annoncement of the track title would help them to know this in a few seconds
     - this prevents the usage of 15 rfid cards for 15 tracks

MiczFlor, 2021-01-20:
It's possible, there are text to speech encoders for open source / linux. They are a bit CPU heavy. And language sensitive. So you need to decide english or german. I have not worked with such software for a decade :) but let's just float the ticket here, a nice idea.

Caliandroid, 2021-02-20:
I did a first test yesterday with pyttsx3.
It seems to be the best approach in terms of voice quality and avoiding proprietary code like google tts.

As i did not work on this project since years, could anybody help me to find the right place to call my text2voice function if a new track starts playing?

s-martin, 2024-04-19:
Implemented in V3 ([see here](https://github.com/MiczFlor/RPi-Jukebox-RFID/tree/future3/develop/src/jukebox/jukebox/speaking_text)).

pabera, 2024-04-20:
We have Text-to-Speech implemented, but it does not read the song out loud. This should be implemented in V3 as an option.