MiczFlor/RPi-Jukebox-RFID #1280: 🚀 | Spotify integration - automatic restart of the mopidy service after boot

Huemmel, 2021-01-19:
Hey, this will be my first input to this project. So I hope it is correct in here and will help other.

In my setup the mopidy service won't wait for the network. The solution in this topic (https://github.com/mopidy/mopidy/pull/1946) is already configured

So I've implemented a `sudo systemctl restart mopidy` in the `startup-scripts.sh` File. I've inserted it in line 44 so it will fire when the startupsound is played.
This solution is working fine for now.

Groovylein, 2021-04-09:
@Huemmel ,
How does the restart of the mopidy improve your experience? What was the problem before implementing this?

Huemmel, 2021-04-09:
In my setup the mopidy service startet before the WiFi Connection was fully establisehed. So the service didn't get a connection and won't work. There is a way to restart the service within the web gui, but the jukebox should work without that gui for usual.

So I've implemented the restart command. For now it worked fine. I had may be 2 or 3 cases in which the jukebox didn't play some musik after boot. But in the most cases (95% +) everything works in the way it should do :-)

Groovylein, 2021-04-09:
Ok. Luckily I did not experience such behavior and I'm asking myself if we should add this code in the `startup-scripts.sh`

s-martin, 2021-04-09:
> In my setup the mopidy service startet before the WiFi Connection was fully establisehed. So the service didn't get a connection and won't work. There is a way to restart the service within the web [...]

For that problem the following workaround should help:
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ#spotify-wont-play-but-after-scan-of-library-it-works

If this is your scenario, I think that workaround is the better way to go.

Huemmel, 2021-04-09:
Well, it adresses the issue #893  .
The mopidy service seems not to wait for the network as described in [https://github.com/mopidy/mopidy/pull/1946](url) and [https://raspberrypi.stackexchange.com/questions/45769/how-to-wait-for-networking-on-login-after-reboot](url)
I've checked the file for the "wait for network" and it was already configured. But the mopidy service seems not start up successfully.

@s-martin that's what i've tested already, but it won't work fine.

SchreibAl, 2021-05-12:
Hello,
it is strange... neither the adjustment of the startup-scrits.sh nor the "wait for the network" brought me the solution. I still have to scan the music library or run the command "sudo systemctl restart mopidy" via terminal after every start or reboot.

Or did I miss something else?

Greetings

Huemmel, 2021-05-18:
> Hello,
> it is strange... neither the adjustment of the startup-scripts.sh nor the "wait for the network" brought me the solution. I still have to scan the music library or run the command "sudo systemctl restart mopidy" via terminal after every start or reboot.
> 
> Or did I miss something else?
> 
> Greetings

Hey,

may be you can have a deeper watch into the code. So you can insert the restart command before playing a playlist.
This idea i got first, but my workaround was enough. So I don't have to search deeper.
If you got a solution, let me know :-)

SchreibAl, 2021-05-21:
> Hey,
> 
> may be you can have a deeper watch into the code. So you can insert the restart command before playing a playlist.
> This idea i got first, but my workaround was enough. So I don't have to search deeper.
> If you got a solution, let me know :-)

Hi, 
I've wrote the restart command "sudo systemctl restart mopidy" into the rc.local file. Now it works...

For those who need more Info:

- Open the terminal
- type "sudo nano /etc/rc.local"
- add "sudo systemctl restart mopidy"
- save with CONTROL + O -> ENTER
- exit with CONTROL + X -> ENTER
- restart with sudo reboot

[image]

Hope this helps...

s-martin, 2021-05-21:
@SchreibAl, do you think it’s worth to add this to the standard installation?

SchreibAl, 2021-05-22:
> @SchreibAl, do you think it’s worth to add this to the standard installation?

Hi,
@s-martin 
good question... I did invest some time to find this "error". It's just strange that it doesn't appear for all users. 
I'm not a developer, so I can't say if this should be included in the code.  But if it doesn't hurt, why not. I think a restart of mopidy might cost a few seconds, but might save some people a lot of time.

BTW: The project here is great. Thanks to all who participate and make this possible!

jonaskor, 2024-04-03:
I'm sorry to comment here again, but I kind of hopeless at the moment. 
I recently installed phoniebox on an Raspberry Pi 4 and I am unable to use Spotify without restarting Mopidy via the command `sudo systemctl restart mopidy` or via rescanning library on the "folder and files" tab. Playing offline files (not playlists) works without this, but that's not what I build the box for. 

Neither editing the `startup-scripts.sh` nor `rc.local` nor using `sudo raspi-config` to make the Pi wait for network on boot worked (setting that option just seems to be ignored).

Any help appreciated!

AlvinSchiller, 2024-04-03:
Hi,
which version did you install? And did you just a plain installation?
Like described [here](https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Spotify-FAQ#spotify-support-for-phoniebox) the spotify integration is broken and needs some manual adjustement to work.

But there is a [PR](https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/2315) to bring these changes in the next release.
Its still a draft, but we are happy for early adopters and feedback.

jonaskor, 2024-04-04:
> Hi, which version did you install? And did you just a plain installation? Like described [here](https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Spotify-FAQ#spotify-support-for-phoniebox) the [...]

I did the manual adjustements, so Spotify works fine after restarting mopidy. But it doesnt work without this trigger, therefore the phoniebox is only usable with "supervision" and my child can't use it without me (what is kind of the idea, right?).

I dont feel comfortable enough in linux to give feedback as an early adopter, I'm just hoping for some rather stable version to use 👍

AlvinSchiller, 2024-04-05:
> I did the manual adjustements, so Spotify works fine after restarting mopidy. But it doesnt work without this trigger, therefore the phoniebox is only usable with "supervision" and my child can't [...]

Alright :)
Nevertheless if you want to try out, you could just install the PR branch on a seperate SDCard.
Just use this command for the installation (no other changes needed):
`cd; rm install-jukebox.sh; wget https://raw.githubusercontent.com/AlvinSchiller/RPi-Jukebox-RFID/feature/mopidy-spotify-fixes/scripts/installscripts/install-jukebox.sh; chmod +x install-jukebox.sh; GIT_URL=https://github.com/AlvinSchiller/RPi-Jukebox-RFID.git GIT_BRANCH=feature/mopidy-spotify-fixes ./install-jukebox.sh`

jonaskor, 2024-04-10:
Unfortunately this didn't fix anything for me (as also mentioned [here](https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/2315#issuecomment-2048393318) ) :/

> > > Hi, which version did you install? And did you just a plain installation? Like described [here](https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Spotify-FAQ#spotify-support-for-phoniebox) the spotify integration is broken and needs some manual adjustement to work.
> > > But there is a [PR](https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/2315) to bring these changes in the next release. Its still a draft, but we are happy for early adopters and feedback.
> > 
> > 
> > I did the manual adjustements, so Spotify works fine after restarting mopidy. But it doesnt work without this trigger, therefore the phoniebox is only usable with "supervision" and my child can't use it without me (what is kind of the idea, right?).
> > I dont feel comfortable enough in linux to give feedback as an early adopter, I'm just hoping for some rather stable version to use 👍
> 
> Alright :) As far is i could test it, theres no problem with the playback after the startup anymore.
> 
> Nevertheless if you want to try out, you could just install the PR branch on a seperate SDCard. Just use this command for the installation (no other changes needed):
> 
> ```
> cd; rm install-jukebox.sh; wget https://raw.githubusercontent.com/AlvinSchiller/RPi-Jukebox-RFID/feature/mopidy-spotify-fixes/scripts/installscripts/install-jukebox.sh; chmod +x install-jukebox.sh; GIT_URL=https://github.com/AlvinSchiller/RPi-Jukebox-RFID.git GIT_BRANCH=feature/mopidy-spotify-fixes ./install-jukebox.sh
> ```

closed 2024-04-14 by AlvinSchiller