MiczFlor/RPi-Jukebox-RFID #1281: Use shairport-sync AirPlay with RPi-Jukebox-RFID

JoernUhl, 2021-01-23:
Hi,

I hope IT is correct that I am posting my question here as I am pretty sure it is actually not an issue with RPi-Jukebox-RFID but rather my own inability to make this work. Please let me know if this question should be asked elsewhere :-)

I installed [shairport-sync](https://github.com/mikebrady/shairport-sync) along with [RPi-Jukebox-RFID](https://github.com/MiczFlor/RPi-Jukebox-RFID/). I installed both on a Raspberry Pi 3, with an [Adafruit Speaker Bonnet](https://www.adafruit.com/product/3346). If I use the functionality provided by RPi-Jukebox-RFID, I get sound and music plays as it should. However, if I try to use airplay from any of my iDevices, no sound is heard.

After running the command `systemctl status shairport-sync` I get the following output a couple of times: 

```
 shairport-sync.service - Shairport Sync - AirPlay Audio Receiver
   Loaded: loaded (/lib/systemd/system/shairport-sync.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2021-01-22 17:39:53 GMT; 30min ago
 Main PID: 480 (shairport-sync)
    Tasks: 11 (limit: 2063)
   CGroup: /system.slice/shairport-sync.service
           └─480 /usr/local/bin/shairport-sync
```
as well as:

```
warning: alsa: error -16 ("Device or resource busy") opening alsa device "hw:0".
```
Running `aplay --list-devices`, I get:

```
**** List of PLAYBACK Hardware Devices ****
card 0: sndrpihifiberry [snd_rpi_hifiberry_dac], device 0: HifiBerry DAC HiFi pcm5102a-hifi-0 [HifiBerry DAC HiFi pcm5102a-hifi-0]
  Subdevices: 0/1
  Subdevice #0: subdevice #0
```

In  `/etc/shairport-sync.conf` I added within the `alsa` section:
```
output_device = "hw:0";
```
and
```
mixer_control_name = "Master";
```

I guess the RPi-Jukebox-RFID installation occupies the Adafruit Speaker Bonnet and is not letting shairport-sync play any music? I would appreciate any tipps or pointers into the right direction :-)

And again, sorry if I am posting this question in the wrong forum.

Thanks a lot for your help,

jrn

s-martin, 2021-02-28:
@ChisSoc can you help here?

ChisSoc, 2021-03-02:
I had a similar issue in another context. And indeed, ALSA resources are exclusive to a single process using them.  With mpc, it takes hold of the Alsa ressource when playing music or when pausing. When you do a proper stop with `mpc stop`, the Alsa ressource is released and another process can use them. Check with `mpc status` which state mpc is in.

For me that was good enough. I did no investigate further how to configure Alsa to mix together multiple streams. It should be possible with the dmix plug. Hit your favorite search enginge with `alsa mixer two streams` or similar. Good luck!

jrnhl, 2021-03-02:
Hi @ChisSoc,

Thank you very much for your reply.

Once I play some music using RPi-Jukebox-RFID and run this command `mpc status`, I receive this output:

```
Some Artist - Some Song
[playing] #1/1   0:00/0:00 (0%)
volume: 15%   repeat: off   random: off   single: off   consume: off
```

When I run `mpc stop` followed by `mpc status` I receive this output:
`volume: 15%   repeat: off   random: off   single: off   consume: off`

However, at all times I am receiving this error running `systemctl status shairport-sync`:
`warning: alsa: error -16 ("Device or resource busy") opening alsa device "hw:0".`

I also tried to add the following to my `.asoundrc` file to allow for multiple streams:

```
pcm.!default {
              type plug
              slave.pcm "dmixer"
}
pcm.dmixer  {
            type dmix
        ipc_key 1024
        slave {
          pcm "hw:1,0"
          period_time 0
              period_size 1024
          buffer_size 4096
          rate 44100
        }
        bindings {
          0 0
          1 1
        }
}
ctl.dmixer {
       type hw
       card 0
}
```

Unfortunately with the same error messages as stated above.

Do you have any other idea what I could try? I appreciate all your help! Thank you very much :-)

ChisSoc, 2021-03-03:
Looking at your /etc/shairport-sync.conf, I'd it is likely this is the culprit. You are trying to use the hardware card **directly** instead of any of your Alsa devices. You will need to setup both, shariport and mpd, to use the dmixer as device.

First try playing sound with `aplay -D dmixer /usr/share/sounds/alsa/Front_Center.wav` to ensure your asound is correctly set up. Then try playing multiple files at once from different consoles. 

Then check your mpd.conf. You might find the issue helpful #1198 for the mpd.conf options.