MiczFlor/RPi-Jukebox-RFID #1284: 🚀 | Control your smarthome, a simple solution via "wget"

berndbarker, 2021-01-28:
I would love to turn my lights on or of via swiping an rfid-card.

I have a simple solution in mind, it should be easy to get the phoniebox to send a wget http request. 
There is already a way to put in links to podcasts, radio stations, etc. 
A little modification of the code might be enough to check if a link starts with "wget" instead of http, then put it directly through without trying to "play" some content.

It would be awesome to have a "light"-card: 
Sitting in front of the phoniebox and listening to music, then just swipe the card if the sun went down and the lights go on.

I could’nt find the part where the handling of the links is done, so I could‘nt try to modify the code myself. 
Maybe you can help with a quick hack?

And eventually in a future release there could be a hole new category for „web requests“ :-)

Thanks a lot!

Bernd

koeberlue, 2021-03-24:
As a temporary solution you could use the [MQTT-Integration](https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Smart-Home-remote-control-with-MQTT), which periodically sends all attributes, including `last_card`. This is not a perfect solution but would work without modifications of the code.

Would it help you, if `last_card` would be published directly over MQTT every time a card is swiped? Maybe that would be a good solution to fully integrate the RFID cards into smart homes.

berndbarker, 2021-03-30:
Hey koeberlue,
thanks for that idea. Yes, it would help if last_card would be published directly over MQTT every time a card is swiped.
I could set up an MQTT server who then sends the http post command.
But it still would be the easiest way, if Phoniebox itself could send an http post request. 
Either way, thanks a lot for your help!