MiczFlor/RPi-Jukebox-RFID #1288: 🚀 | Commands to disable and enable the functionality of hardware buttons

LorenzKahl, 2021-02-02:
## Feature Description

We have two kids. A boy of 3,5 years (the main user of the box that would love being able to enjoy an uninterrupted episode of Bob the Builder) and a little girl of 11 months (that just loooooves pushing buttons). I need a quick way of being able to disable and re-enable the pushbuttons for playback and track-control.

### What functionality would you like to see in your phoniebox?

Please provide a two according commands that I can assign to cards.

### How do you envision the feature to work from a users perspective?

As a user I'd like to go into the RFID section of the settings and assign one of the new commands ("Disable hardware buttons" / "Enable hardware buttons") to a card. 

### Further information that might help

It would be extra nice if I could also configure which buttons should be affected using the [`gpio_settings.ini`](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/dfeed95eac288489409267ee7ea4cdf374ab0936/components/gpio_control/example_configs/gpio_settings.ini#L26) - for example by adding a property like
`canBeDisabledByCommand: [True|False]`

MiczFlor, 2021-02-03:
Hi @LorenzKahl 
if you are using the web app, you can do that easily already. If you go to settings, there is the option to switch GPIO on and off. 
All the best, micz

s-martin, 2024-04-21:
This new feature should be implemented in V3. [RPC Commands](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/future3/develop/documentation/builders/rpc-commands.md) are the way to implement this.