MiczFlor/RPi-Jukebox-RFID #1290: 🐛 | Spotify connection via mobile hotspot doesn't work

mathias-wi, 2021-02-06:
Hello everybody,

I have a working phoniebox based on the default one-line install script for buster (version 2.2). Everything works fine and today I wanted to add the "mobil hotspot" functionality as @gizmo21 explains in #225.
I can see on the mobile hotspot settings on my phone, that the phoniebox is connected, but when I swipe a card, no music is playing. Maybe someone can help me with my questions:
- How can I access the Phoniebox cmd-line when it is connected via mobile hotspot?
- Do I need to make some changes in mopidy/iris?
- When connected to my normal WiFi at home I gave a static IP-Adress to the Phoniebox - could it be possible, that I need to make some changes here?

It would be great if someone could give me a hint.
Best, Mathias

---------------------------

_Here are my settings:
/etc/dhcpcd.conf

#Heimnetzwerk
ssid Name1
interface wlan0
static ip_address=192.168.178.201
static routers=192.168.178.1
static domain_name_servers=192.168.178.1

#Mobiler Hotspot Handy
ssid Name2

-----------------------

_/etc/wpa_supplicant/wpa_supplicant.conf 

ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=DE
network={
        ssid="Name1"
        psk=PW1
        priority=1
}
network={
        ssid="Namea2"
        psk=PW2
        priority=0
}_

s-martin, 2021-02-28:
I’m not sure what your use case is.

The mentioned hotspot mode in #225 means that the Phoniebox provides their own WiFi you can connect to to control the Phoniebox e.g. from your smartphone even if there’s no other WiFi around.

But in that case the Phoniebox won’t have an internet connection and then Spotify won’t work.

The use case is mainly, if you bring the Phoniebox somewhere where no WiFi is around both the box and your smartphone are connected to.

what do you want to achieve?

mathias-wi, 2021-03-24:
Hello @s-martin,
I would like to achieve the following: If there is no WLAN available for the phoniebox (e.g. car), then I would like to set up a mobile hotspot with my phone. The phoniebox then should connect to this mobile hotspot and work as it would at home in the regular WLAN. Is this possible at all?

s-martin, 2021-03-26:
@mathias-wi 
This could help:
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ#i-want-to-add-another-wi-ssid-eg-at-a-friend