MiczFlor/RPi-Jukebox-RFID #1292: Install Phoniebox after the new Linux Update

mimi-ac, 2021-02-08:
Hello everybody,

a few month ago I installed a phoniebox on a raspberrry pi 3b+ by following steps:

- install the old Raspian Lite
- install of all upgrades and updates
- install phoniebox (rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/master/scripts/installscripts/buster-install-default.sh; chmod +x buster-install-default.sh; ./buster-install-default.sh)

Now I am not sure if I can install a new phoniebox the same way, because of

* the new "pulse audio"
* the new "Raspberry Pi OS Lite"

Should I use an older image of raspbian and do without "upgrade" and "update"?
Can I use alsa instead of pulseaudio although I use the new Rasberry Pi OS Lite?

Thankx

smu, 2021-02-11:
Recently, I installed a new phonibox using Raspberry Pi OS Lite version 2021-01-11 and it works like a charm.

Caliandroid, 2021-02-15:
Hello @mimi-ac ,

i did the same like @smu  with Pi OS Lite and RaspberryPi 3b+  one week ago because of a microSD card corrupt.

I transferred also the shared audiofolder afterwards from the broken sd card to restore all rfid tags and audio files without problems.

In fact Pi OS Lite can be recommended as preferred OS version for the Phoniebox to save space on the sd card and unnecessary data transfer.

closed 2021-05-14 by s-martin