MiczFlor/RPi-Jukebox-RFID #1295: 🚀 | Battery SOC (State of Charge) Monitor in the Web Interfcae

arne123, 2021-02-10:
Battery SOC (State of Charge) Monitor in the Web Interface

I'd like to know the state of charge of my battery powered phoniebox.
I'd like to be aware of the Battery charge every time I open the Web Interface, regardless of the menu

So my proposal is to place a Battery Symbol right side of the shutdown button or menu selector on mobile,
e.g. a symbol like this one (there are full empty etc. available): 
https://raw.githubusercontent.com/xfce-mirror/xfce4-power-manager/master/data/icons/scalable/battery-low.svg

Unfortunately I am a unteachable rookie when it come to visual Design and css.
Also I have no real clue about AJAX.
I would be very pleased if someone could help me to place a Battery symbol right next to here:

From that point on I can easily provide a php script, providing the correct image (empty, full etc.)

Btw. I used the Web UI many times form the Webbrowser on the PC, but so far never from the Smartphone.
So I was ,lets say, not surprised about the UI, it was ok and usable.
The Last days I used it first time from the smartphone and was very impressed of the Web UI on a smartphone.
I think you have don't a good job here!

br
  Arne

ZyanKLee, 2021-03-05:
Hey Arne,
The biggest part of this problem would be to find the source for this data. What kind of battery so you use?
I merely put a powerbank in the box, but that does not transmit it's charging state to the raspberry pi. So in my system this won't work.

arne123, 2021-03-05:
I am using a single cell LiIon, and intending to use an I²C ADC to monitor the Voltage (ADS1015 Modules are widely used).

ZyanKLee, 2021-03-05:
Sounds like a little daemon that polls the adc might be a viable small side project. Then the phoniebox webinterface can query that battery daemon to get the state and display it.

s-martin, 2023-05-07:
Battery monitoring is implemented in **future3** although this is not yet part of the official documentation, see here for some info https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/future3/develop/src/jukebox/components/battery_monitor/batt_mon_i2c_ads1015/__init__.py

I assume this will not be implemented in V2, so I’m closing this issue for now.

closed 2023-05-07 by s-martin