MiczFlor/RPi-Jukebox-RFID #1309: Spotify Playlist Titles not Showing in App Player

TimiRose, 2021-02-24:
Hi,
enjoying the Phoniebox very much. I use the Spotify Version, and have come a cross a new issue recently.

Somehow when I add new spotifyplaylists, they work allright, but in the app player they aren´t shown by their name anymore. It didn´t used to be like this. 
This means I have a long list of playlists that are all unnamed. They play, but this way I don´t know which playlist I can add shuffle or whatever options to. 
If I ssh into the Phoniebox they have the right names in the playlist folder and and also in the app they appear under dropdown menu "Link Card to an Existing Playlist" with the names I gave them.
I notice that I can´t seem to upgrade to to Iris 3.56.1 but appart from this everything works just fine.

Thanks for any help!
Tim

s-martin, 2021-02-28:
What’s the output of `journalctl -u mopidy`?

TimiRose, 2021-03-01:
-- Logs begin at Thu 2019-02-14 11:11:59 CET, end at Mon 2021-03-01 22:25:15 CET. --
Feb 28 11:51:25 raspberrypi systemd[1]: Starting Mopidy music server...
Feb 28 11:51:25 raspberrypi systemd[1]: Started Mopidy music server.
Feb 28 11:51:40 raspberrypi mopidy[524]: INFO     [MainThread] mopidy.__main__ Starting Mopidy 3.0.2
Feb 28 11:51:40 raspberrypi mopidy[524]: INFO     [MainThread] mopidy.config Loading config from builtin defaults
Feb 28 11:51:40 raspberrypi mopidy[524]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy-local.conf
Feb 28 11:51:40 raspberrypi mopidy[524]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy.conf
Feb 28 11:51:40 raspberrypi mopidy[524]: INFO     [MainThread] mopidy.config Loading config from file:///etc/mopidy/mopidy.conf
Feb 28 11:51:40 raspberrypi mopidy[524]: INFO     [MainThread] mopidy.config Loading config from command line options
Feb 28 11:51:40 raspberrypi mopidy[524]: ERROR    [MainThread] mopidy.internal.log Loading logging config '/etc/mopidy/logging.conf' failed. 'formatters'
Feb 28 11:51:41 raspberrypi mopidy[524]: INFO     [MainThread] mopidy.__main__ Enabled extensions: local, softwaremixer, iris, file, stream, mpd, http, m3u, 
Feb 28 11:51:41 raspberrypi mopidy[524]: INFO     [MainThread] mopidy.__main__ Disabled extensions: none
Feb 28 11:51:43 raspberrypi mopidy[524]: INFO     [MainThread] mopidy.commands Starting Mopidy mixer: SoftwareMixer
Feb 28 11:51:43 raspberrypi mopidy[524]: INFO     [MainThread] mopidy.commands Mixer volume set to 30
Feb 28 11:51:43 raspberrypi mopidy[524]: INFO     [MainThread] mopidy.commands Starting Mopidy audio
Feb 28 11:51:43 raspberrypi mopidy[524]: INFO     [MainThread] mopidy.commands Starting Mopidy backends: FileBackend, M3UBackend, StreamBackend, SpotifyBacke
Feb 28 11:51:43 raspberrypi mopidy[524]: INFO     [Audio-2] mopidy.audio.actor Audio output set to "alsasink"
Feb 28 11:51:44 raspberrypi mopidy[524]: INFO     [SpotifyBackend-6] mopidy_spotify.web Logged into Spotify Web API as 19dt9ll1m5195tvpy76ndss7q
Feb 28 11:51:44 raspberrypi mopidy[524]: INFO     [MainThread] mopidy.commands Starting Mopidy core
Feb 28 11:51:44 raspberrypi mopidy[524]: INFO     [MainThread] mopidy.commands Starting Mopidy frontends: HttpFrontend, IrisFrontend, MpdFrontend
Feb 28 11:51:44 raspberrypi mopidy[524]: INFO     [HttpFrontend-12] mopidy.http.actor HTTP server running at [::ffff:0.0.0.0]:6680
Feb 28 11:51:44 raspberrypi mopidy[524]: INFO     [IrisFrontend-13] mopidy_iris.core Starting Iris 3.55.2
Feb 28 11:51:44 raspberrypi mopidy[524]: INFO     [MainThread] mopidy_mpd.actor MPD server running at [::ffff:0.0.0.0]:6600
Feb 28 11:51:44 raspberrypi mopidy[524]: INFO     [MainThread] mopidy.commands Starting GLib mainloop
Feb 28 11:51:44 raspberrypi mopidy[524]: INFO     [MpdSession-15] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:51000
Feb 28 11:51:44 raspberrypi mopidy[524]: INFO     [MpdSession-16] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:51004
Feb 28 11:51:44 raspberrypi mopidy[524]: INFO     [MpdSession-17] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:51008
Feb 28 11:51:44 raspberrypi mopidy[524]: INFO     [MpdSession-18] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:51012
Feb 28 11:51:44 raspberrypi mopidy[524]: INFO     [SpotifyEventLoop] mopidy_spotify.backend Logged in to Spotify in online mode
Feb 28 11:51:44 raspberrypi mopidy[524]: INFO     [MpdSession-20] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:51016
Feb 28 11:51:45 raspberrypi mopidy[524]: INFO     [MpdSession-21] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:51020
Mar 01 22:15:01 raspberrypi mopidy[524]: INFO     [MpdSession-22] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:51024
Mar 01 22:15:01 raspberrypi mopidy[524]: INFO     [MpdSession-23] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:51028
Mar 01 22:16:03 raspberrypi mopidy[524]: INFO     [MpdSession-24] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:51032
Mar 01 22:16:03 raspberrypi mopidy[524]: INFO     [MpdSession-25] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:51036
Mar 01 22:17:04 raspberrypi mopidy[524]: INFO     [MpdSession-26] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:51040
Mar 01 22:17:04 raspberrypi mopidy[524]: INFO     [MpdSession-27] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:51044
Mar 01 22:18:06 raspberrypi mopidy[524]: INFO     [MpdSession-28] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:51048
Mar 01 22:18:06 raspberrypi mopidy[524]: INFO     [MpdSession-29] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:51052
Mar 01 22:19:07 raspberrypi mopidy[524]: INFO     [MpdSession-30] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:51056
Mar 01 22:19:07 raspberrypi mopidy[524]: INFO     [MpdSession-31] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:51060
Mar 01 22:20:08 raspberrypi mopidy[524]: INFO     [MpdSession-32] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:51064
Mar 01 22:20:08 raspberrypi mopidy[524]: INFO     [MpdSession-33] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:51068
[... cut ...]
Mar 01 22:24:12 raspberrypi mopidy[524]: INFO     [MpdSession-40] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:51096
Mar 01 22:24:12 raspberrypi mopidy[524]: INFO     [MpdSession-41] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:51100
Mar 01 22:25:14 raspberrypi mopidy[524]: INFO     [MpdSession-42] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:51104
Mar 01 22:25:14 raspberrypi mopidy[524]: INFO     [MpdSession-43] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:51108
~
lines 1-53/53 (END)

TimiRose, 2021-03-25:
Is that no help? Still having the same problem...

Black616Angel, 2021-09-28:
Since I am having the same issues, I dived a little into it.
The error comes from the empty "title.txt" files in the audio-folders.
I have yet to find out, where those files come from.
If someone like @s-martin could help with that, I will try to fix the issue.

Black616Angel, 2021-09-28:
Okay, I think, I've found the issue.
It is in the file "inc.viewFolderTree.php".
There are two errors to be precise, both have their own issues. (see  #1451)

They both come from the use of "file_get_contents" (lines 121 and 130) to grab webpages from Spotify.
This (at least on my machine) doesn't work.
That is why the output of the title as well as the cover do not work.

Now I am not that well versed in PHP to propose a working solution right now.
But maybe someone else can work with my analysis.

s-martin, 2021-09-29:
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/bb0361e41a48b5a36e564d0e92f1e2d551b1b879/htdocs/inc.viewFolderTree.php#L121

https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/bb0361e41a48b5a36e564d0e92f1e2d551b1b879/htdocs/inc.viewFolderTree.php#L130

Black616Angel, 2021-10-11:
So on my raspberry I upgraded to php7.4 today but the ph version seems to be hardcoded somwhere, because it still uses 7.3.

The error show is a 400 - bad request, but when I try it on another site like reqbin.com it works flawlessly.

The url I use is:
https://open.spotify.com/oembed/?url=spotify:playlist:76Y5e2vPhHsOQeOB96ghfL&format=json

I will try again another day.

olsins, 2021-11-28:
I still have the same problem - no cover ist loaded... 
I have the following configuration:

`
if(file_exists($subfolder."/podcast.txt")){
            $temp['type'] = "podcast";
        } elseif(file_exists($subfolder."/livestream.txt")){
            $temp['type'] = "livestream";
        } elseif(file_exists($subfolder."/spotify.txt")){
            $temp['type'] = "spotify";
            $titlefile = $subfolder."/title.txt";
            $coverfile = $subfolder."/cover.jpg";
                        
                        // this is a new and easier way for loading spotify informations!
                        $uri = file_get_contents($subfolder."/spotify.txt");
                        $url = "https://open.spotify.com/oembed/?url=".trim($uri)."&format=json";
                        
                        if (!file_exists($coverfile)) {
                                $str = file_get_contents($url);
                                $json  = json_decode($str, true);

                                $cover = $json['thumbnail_url'];
                                $coverdl = file_get_contents($cover);
                                file_put_contents($coverfile, $coverdl);
                        }
                        
                        if (!file_exists($titlefile)) {
                                $str = file_get_contents($url);
                                $json  = json_decode($str, true);

                                $title = $json['title'];
                                file_put_contents($titlefile, $title);
                        }
                        
        } else {
            $temp['type'] = "generic";
        }

`

Does anybody else having still the same issue?

olsins, 2021-11-28:
Little hint - it worked for me after I did:

sudo chown -R pi:www-data /home/pi/RPi-Jukebox-RFID/shared
sudo chmod -R 775 /home/pi/RPi-Jukebox-RFID/shared
sudo chown -R pi:www-data /home/pi/RPi-Jukebox-RFID/htdocs
sudo chmod -R 775 /home/pi/RPi-Jukebox-RFID/htdocs

usamer77, 2021-11-29:
@olsins : Glad that it worked! Did you do the changes @acidburn78 proposed as well?

Thx!

olsins, 2021-11-29:
yes - thats what I meant, the combination made it!

Oldie95, 2021-12-06:
Hey guys,
@acidburn78: Could you explain the error and tell me what the difference to the current code was? I installed the current stable version 2.x of Phoniebox and I just managed to do it your way with succes. However, I do not see the reason and would like to understand!? :-)

acidburn78, 2021-12-13:
> Hey guys, @acidburn78: Could you explain the error and tell me what the difference to the current code was? I installed the current stable version 2.x of Phoniebox and I just managed to do it your way with succes. However, I do not see the reason and would like to understand!? :-)

My Script uses a different Method (Curl) the API of Spotify didn't with the Old Method

s-martin, 2022-05-31:
Fixed in #1692

closed 2022-05-31 by s-martin