MiczFlor/RPi-Jukebox-RFID #1311: No Sound, Volume 0%

oneshow01, 2021-02-25:
Hi,
bisher hat die Hard- und Software hat unter „Stretch" perfekt funktioniert...
Wegen einer größeren Speicherkarte habe ich eine komplette Neuinstallation mit Buster und der Phoniebox über "one line script" ohne Spotify vorgenommen.
Der onboard sound über jack funktioniert direkt über Raspberry Pi einwandfrei.
In der Phoniebox habe ich keinen Ton. Die Lautstärke ist auch immer 0%. Wenn man sie auf 100% stellt passiert auch nichts.
Deshalb habe ich unter "Raspberry Pi Update May 2020 broke my audio“ versucht das Problem zu lösen. 

Ich komme aber leider nicht weiter, hier ein paar Fakten:
Raspberry Pi 3 Model B Rev 1.2, Raspbian GNU/Linux 10 (buster)

************************************
*** PHONIEBOX INFO
*** version: 2.2 - 305325d - master
*** edition: classic
*** Audio_iFace_Name: Master
*** Audio_Folders_Path: /home/pi/RPi-Jukebox-RFID/shared/audiofolders
*** Audio_Volume_Change_Step: 3
*** Max_Volume_Limit: 100
*** Idle_Time_Before_Shutdown: 0
*** Second_Swipe: RESTART
*** Playlists_Folders_Path: /home/pi/RPi-Jukebox-RFID/playlists
*** ShowCover: OFF
************************************
*** CONF FILES DEFAULT
 
*** /etc/samba/smb.conf
-rw-r--r-- 1 root root 8,2K Feb 24 21:19 /etc/samba/smb.conf
   path=/home/pi/RPi-Jukebox-RFID/shared
   path=/home/pi/RPi-Jukebox-RFID/shared/audiofolders
*** /etc/lighttpd/lighttpd.conf
-rw-r--r-- 1 root root 2,1K Feb 24 21:19 /etc/lighttpd/lighttpd.conf
*** /etc/lighttpd/conf-available/15-fastcgi-php.conf
-rw-r--r-- 1 root root 575 Feb 24 21:19 /etc/lighttpd/conf-available/15-fastcgi-php.conf
*** /etc/php/7.0/fpm/php.ini
ls: Zugriff auf '/etc/php/7.0/fpm/php.ini' nicht möglich: Datei oder Verzeichnis nicht gefunden
*** /etc/sudoers
-r--r----- 1 root root 784 Feb 24 21:19 /etc/sudoers
*** /etc/systemd/system/phoniebox*
-rw-r--r-- 1 root root 295 Feb 24 21:19 /etc/systemd/system/phoniebox-rfid-reader.service
ls: Zugriff auf '/etc/systemd/system/phoniebox-startup-sound.service' nicht möglich: Datei oder Verzeichnis nicht gefunden
ls: Zugriff auf '/etc/systemd/system/phoniebox-gpio-buttons.service' nicht möglich: Datei oder Verzeichnis nicht gefunden
-rw-r--r-- 1 root root 291 Feb 24 21:19 /etc/systemd/system/phoniebox-idle-watchdog.service
*** /etc/mpd.conf
-rw-r----- 1 mpd audio 14K Feb 25 18:35 /etc/mpd.conf
music_directory		"/home/pi/RPi-Jukebox-RFID/shared/audiofolders"
# music_directory are changed.
# symbolic links outside of the configured music_directory.
# symbolic links inside of the configured music_directory.
	mixer_control	"Master"		# optional
#	mixer_control	"PCM"		# optional
*** /etc/dhcpcd.conf
-rw-rw-r-- 1 root netdev 130 Feb 24 21:19 /etc/dhcpcd.conf
static ip_address=192.168.178.21/24
static routers=192.168.178.1
static domain_name_servers=8.8.8.8 192.168.178.1
*** /etc/wpa_supplicant/wpa_supplicant.conf
-rw-rw-r-- 1 root netdev 151 Feb 24 21:19 /etc/wpa_supplicant/wpa_supplicant.conf
country=DE
************************************
*** +Spotify Edition
 
*** /etc/locale.gen
-rw-r--r-- 1 root root 9,2K Jan 11 14:17 /etc/locale.gen
*** /etc/locale.gen
-rw-r--r-- 1 root root 9,2K Jan 11 14:17 /etc/locale.gen
*** /etc/mopidy/mopidy.conf
ls: Zugriff auf '/etc/mopidy/mopidy.conf' nicht möglich: Datei oder Verzeichnis nicht gefunden
cat: /etc/mopidy/mopidy.conf: Datei oder Verzeichnis nicht gefunden
*** ~/.config/mopidy/mopidy.conf
ls: Zugriff auf '/home/pi/.config/mopidy/mopidy.conf' nicht möglich: Datei oder Verzeichnis nicht gefunden
cat: /home/pi/.config/mopidy/mopidy.conf: Datei oder Verzeichnis nicht gefunden

Dank Euch für eure Hilfe

seagullki, 2021-02-27:
Hast du mal versucht ohne Spotify zu installieren?
Ich habe ebenfalls das Problem, dass ohne Spotify installiert alles funktioniert. Aber mit Spotify funktioniert auch alles andere Abspielen nicht.

oneshow01, 2021-02-27:
> Hast du mal versucht ohne Spotify zu installieren?
> Ich habe ebenfalls das Problem, dass ohne Spotify installiert alles funktioniert. Aber mit Spotify funktioniert auch alles andere Abspielen nicht.

Hallo,
ja ich habe mit spotify und ohne versucht, sogar die Entwickler Version. Immer exakt das selbe Problem...

s-martin, 2021-02-27:
Please check here:
https://github.com/MiczFlor/RPi-Jukebox-RFID/discussions/1300

oneshow01, 2021-02-28:
Danke, jetzt geht es wieder!

Ich habe zuerst eine Neuinstallation rein über ssh und vnc ohne Monitor gemacht. Dann hatte ich erstmal keinerlei Sound mehr. Aber dank der "Audio never worked" Anleitung
_
$ sudo nano /usr/share/alsa/alsa.conf
Replace the lines near the end of the file:
defaults.ctl.card 0
defaults.pcm.card 0
with:
defaults.ctl.card 1
defaults.pcm.card 1__

geht es jetzt wieder.

Ihr seid super!

closed 2021-02-28 by oneshow01