MiczFlor/RPi-Jukebox-RFID #1316: Abspielen nicht möglich bei Installation mit Spotify

seagullki, 2021-02-27:
## Bug

Abspielen von MP3s nicht möglich bei Installation mit Spotify.

### What I did
Wenn ich in der Webgui auf das "Play"-Symbol klicken, wird kurz eingeblendet das abgespielt wird. Aber es passiert nichts. Auch im Playerbereich taucht nichts auf und der Play-Button ändert sich nicht in den Pause-Button.
Audio selbst funktioniert problemlos auf der Cli.

### What happened
Installiere ich ohne Spotify, funktioniert alles problemlos. Mehrfach hin und her probiert, alles ist gleich nur entweder mit oder ohne Spotify.
mit Spotify = funktioniert nicht
ohne Spotify = funktioniert

### I expected this to happen

Titel (MP3) wird abgespielt.

### Further information that might help

/var/log/syslog
Feb 27 17:28:03 Phoniebox mopidy[4411]: INFO     [MpdSession-1927] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:53544
Feb 27 17:28:03 Phoniebox mopidy[4411]: INFO     [MpdSession-1928] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:53548
Feb 27 17:28:03 Phoniebox mopidy[4411]: INFO     [MpdSession-1929] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:53552
Feb 27 17:28:03 Phoniebox mopidy[4411]: ERROR    [IrisFrontend-13] mopidy.listener Triggering event failed: tracklist_changed()
Feb 27 17:28:03 Phoniebox mopidy[4411]: Traceback (most recent call last):
Feb 27 17:28:03 Phoniebox mopidy[4411]:   File "/usr/local/lib/python3.7/dist-packages/mopidy/listener.py", line 44, in on_event
Feb 27 17:28:03 Phoniebox mopidy[4411]:     getattr(self, event)(**kwargs)
Feb 27 17:28:03 Phoniebox mopidy[4411]:   File "/usr/local/lib/python3.7/dist-packages/mopidy_iris/frontend.py", line 29, in tracklist_changed
Feb 27 17:28:03 Phoniebox mopidy[4411]:     iris.ioloop.add_callback(functools.partial(iris.clean_queue_metadata))
Feb 27 17:28:03 Phoniebox mopidy[4411]: AttributeError: 'NoneType' object has no attribute 'add_callback'
Feb 27 17:28:03 Phoniebox mopidy[4411]: INFO     [MpdSession-1930] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:53556
Feb 27 17:28:03 Phoniebox lighttpd[4858]: error adding /home/pi/RPi-Jukebox-RFID/shared/audiofolders/Test/Test.mp3: directory or file not found

pi@Phoniebox:~ $ ls -l /home/pi/RPi-Jukebox-RFID/shared/audiofolders/Test/
insgesamt 876
-rwxrw-rw- 1 pi pi 895124 Dez 20  2016 Test.mp3

## Software

### Base image and version
2021-01-11-raspios-buster-armhf-lite.img

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

### Branch / Release
master

### Installscript
cd; rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/master/scripts/installscripts/buster-install-default.sh; chmod +x buster-install-default.sh; ./buster-install-default.sh

## Hardware

### RaspberryPi version
Pi3 B V1.2

### RFID Reader

### Soundcard

### Other notable hardware

derchan, 2021-02-27:
I noticed this on a RPi 3 A as well. It works on the classic installation and it does not work on the installation with Spotify.

s-martin, 2021-02-27:
please try:
`sudo pip3 install --upgrade --force-reinstall Mopidy-Iris` and reboot

derchan, 2021-02-28:
> sudo pip3 install --upgrade --force-reinstall Mopidy-Iris

Same issue still reproducible for me on Master and Develop branch. 
Any more information i can help you with?

s-martin, 2021-02-28:
Output of `journalctl -u mopidy`?

seagullki, 2021-02-28:
still the same at my installation too.

journal -u mopidy
-----------------------
-- Logs begin at Sun 2021-02-28 10:30:18 CET, end at Sun 2021-02-28 10:32:21 CET. --
Feb 28 10:30:36 Phoniebox systemd[1]: Starting Mopidy music server...
Feb 28 10:30:36 Phoniebox systemd[1]: Started Mopidy music server.
Feb 28 10:30:41 Phoniebox mopidy[5718]: INFO     [MainThread] mopidy.__main__ Starting Mopidy 3.1.1
Feb 28 10:30:41 Phoniebox mopidy[5718]: INFO     [MainThread] mopidy.config Loading config from builtin defaults
Feb 28 10:30:41 Phoniebox mopidy[5718]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy.conf
Feb 28 10:30:41 Phoniebox mopidy[5718]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy-local.conf
Feb 28 10:30:41 Phoniebox mopidy[5718]: INFO     [MainThread] mopidy.config Loading config from file:///etc/mopidy/mopidy.conf
Feb 28 10:30:41 Phoniebox mopidy[5718]: INFO     [MainThread] mopidy.config Loading config from command line options
Feb 28 10:30:41 Phoniebox mopidy[5718]: ERROR    [MainThread] mopidy.internal.log Loading logging config '/etc/mopidy/logging.conf' failed. 'formatters'
Feb 28 10:30:42 Phoniebox mopidy[5718]: INFO     [MainThread] mopidy.__main__ Enabled extensions: iris, spotify, mpd, local, http, stream, file, m3u, softwaremixer
Feb 28 10:30:42 Phoniebox mopidy[5718]: INFO     [MainThread] mopidy.__main__ Disabled extensions: none
Feb 28 10:30:44 Phoniebox mopidy[5718]: INFO     [MainThread] mopidy.commands Starting Mopidy mixer: SoftwareMixer
Feb 28 10:30:44 Phoniebox mopidy[5718]: INFO     [MainThread] mopidy.commands Mixer volume set to 30
Feb 28 10:30:44 Phoniebox mopidy[5718]: INFO     [MainThread] mopidy.commands Starting Mopidy audio
Feb 28 10:30:44 Phoniebox mopidy[5718]: INFO     [MainThread] mopidy.commands Starting Mopidy backends: FileBackend, M3UBackend, StreamBackend, SpotifyBackend, LocalBackend
Feb 28 10:30:44 Phoniebox mopidy[5718]: INFO     [Audio-2] mopidy.audio.actor Audio output set to "alsasink"
Feb 28 10:30:50 Phoniebox mopidy[5718]: ERROR    [SpotifyEventLoop] spotify.session Spotify login error: <ErrorType.BAD_USERNAME_OR_PASSWORD: 6>
Feb 28 10:31:17 Phoniebox mopidy[5718]: ERROR    [SpotifyBackend-6] mopidy_spotify.web OAuth token refresh failed: Unknown error.
Feb 28 10:31:17 Phoniebox mopidy[5718]: ERROR    [SpotifyBackend-6] mopidy_spotify.web Failed to load Spotify user profile
Feb 28 10:31:17 Phoniebox mopidy[5718]: INFO     [MainThread] mopidy.commands Starting Mopidy core
Feb 28 10:31:17 Phoniebox mopidy[5718]: INFO     [MainThread] mopidy.commands Starting Mopidy frontends: HttpFrontend, IrisFrontend, MpdFrontend
Feb 28 10:31:17 Phoniebox mopidy[5718]: INFO     [HttpFrontend-12] mopidy.http.actor HTTP server running at [::ffff:0.0.0.0]:6680
Feb 28 10:31:17 Phoniebox mopidy[5718]: INFO     [IrisFrontend-13] mopidy_iris.core Starting Iris 3.56.1
Feb 28 10:31:17 Phoniebox mopidy[5718]: INFO     [MainThread] mopidy_mpd.actor MPD server running at [::ffff:0.0.0.0]:6600
Feb 28 10:31:17 Phoniebox mopidy[5718]: INFO     [IrisFrontend-13] mopidy.internal.path Creating dir file:///var/lib/mopidy/iris
Feb 28 10:31:17 Phoniebox mopidy[5718]: INFO     [MainThread] mopidy.commands Starting GLib mainloop
Feb 28 10:31:17 Phoniebox mopidy[5718]: INFO     [MpdSession-15] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:50100
Feb 28 10:31:17 Phoniebox mopidy[5718]: INFO     [MpdSession-16] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:50104
Feb 28 10:31:17 Phoniebox mopidy[5718]: INFO     [MpdSession-17] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:50108
Feb 28 10:31:17 Phoniebox mopidy[5718]: INFO     [MpdSession-18] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:50112
Feb 28 10:31:22 Phoniebox mopidy[5718]: INFO     [MpdSession-19] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:50116
Feb 28 10:31:25 Phoniebox mopidy[5718]: INFO     [MpdSession-20] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:50120
Feb 28 10:31:25 Phoniebox mopidy[5718]: INFO     [MpdSession-21] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:50124
Feb 28 10:31:25 Phoniebox mopidy[5718]: INFO     [MpdSession-22] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:50128
Feb 28 10:31:25 Phoniebox mopidy[5718]: INFO     [MpdSession-23] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:50132
Feb 28 10:31:25 Phoniebox mopidy[5718]: INFO     [MpdSession-24] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:50136
Feb 28 10:31:25 Phoniebox mopidy[5718]: INFO     [MpdSession-25] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:50140
...
-> many more of this lines

The login error ist okay, becaus the logindata ist set to "bla" for testing.

s-martin, 2021-02-28:
`Feb 28 10:30:50 Phoniebox mopidy[5718]: ERROR [SpotifyEventLoop] spotify.session Spotify login error: <ErrorType.BAD_USERNAME_OR_PASSWORD: 6>`

You're not logged in correctly.

seagullki, 2021-02-28:
This is right, as i say. But i don't want to play from spotify. I want to play MP3 from SD Card.

derchan, 2021-02-28:
Ok I had in the master branch the same problem. I am right now in the delveop branch, because i am playing around with the USB encoder.  On a clean install it does not work. Spotifiy does work. 

I am using a RPI 3A and an USB sound card. The ondboard sound is deactivated in settings. 
After install folowing changes have been made but should not be important for this issue. 
Installed i2c interface for display. switch folder checkout for components to a previous version due to ticket #1318

The error should be probably this one. 

Feb 28 14:56:31 chanorama mopidy[3234]: ERROR    [LocalBackend-7] mopidy_local.library Lookup error for local:track:test/01%20-%20Baba%20Yetu%20%28feat.%20Malukah%29.mp3: no such table: tracks

and i assume this one here should not be a problem

Feb 28 01:23:40 chanorama mopidy[3234]: ERROR    [MainThread] mopidy.internal.log Loading logging config '/etc/mopidy/logging.conf' failed. 'formatters'

hope that helps

pi@chanorama:~/RPi-Jukebox-RFID/components/displays/HD44780-i2c $ journalctl -u mopidy
-- Logs begin at Sun 2021-02-28 01:23:13 GMT, end at Sun 2021-02-28 14:57:45 GMT. --
Feb 28 01:23:30 chanorama systemd[1]: Starting Mopidy music server...
Feb 28 01:23:30 chanorama systemd[1]: Started Mopidy music server.
Feb 28 01:23:40 chanorama mopidy[3234]: INFO     [MainThread] mopidy.__main__ Starting Mopidy 3.1.1
Feb 28 01:23:40 chanorama mopidy[3234]: INFO     [MainThread] mopidy.config Loading config from builtin defaults
Feb 28 01:23:40 chanorama mopidy[3234]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy.conf
Feb 28 01:23:40 chanorama mopidy[3234]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy-local.conf
Feb 28 01:23:40 chanorama mopidy[3234]: INFO     [MainThread] mopidy.config Loading config from file:///etc/mopidy/mopidy.conf
Feb 28 01:23:40 chanorama mopidy[3234]: INFO     [MainThread] mopidy.config Loading config from command line options
Feb 28 01:23:40 chanorama mopidy[3234]: ERROR    [MainThread] mopidy.internal.log Loading logging config '/etc/mopidy/logging.conf' failed. 'formatters'
Feb 28 01:23:41 chanorama mopidy[3234]: INFO     [MainThread] mopidy.__main__ Enabled extensions: local, softwaremixer, m3u, file, mpd, stream, spotify, iris, http
Feb 28 01:23:41 chanorama mopidy[3234]: INFO     [MainThread] mopidy.__main__ Disabled extensions: none
Feb 28 01:23:43 chanorama mopidy[3234]: INFO     [MainThread] mopidy.commands Starting Mopidy mixer: SoftwareMixer
Feb 28 01:23:43 chanorama mopidy[3234]: INFO     [MainThread] mopidy.commands Mixer volume set to 30
Feb 28 01:23:43 chanorama mopidy[3234]: INFO     [MainThread] mopidy.commands Starting Mopidy audio
Feb 28 01:23:43 chanorama mopidy[3234]: INFO     [MainThread] mopidy.commands Starting Mopidy backends: FileBackend, M3UBackend, StreamBackend, SpotifyBackend, LocalBackend
Feb 28 01:23:44 chanorama mopidy[3234]: INFO     [Audio-2] mopidy.audio.actor Audio output set to "alsasink"
Feb 28 01:23:45 chanorama mopidy[3234]: INFO     [SpotifyEventLoop] mopidy_spotify.backend Logged in to Spotify in online mode
Feb 28 01:23:45 chanorama mopidy[3234]: INFO     [SpotifyBackend-6] mopidy_spotify.web Logged into Spotify Web API as xxx
Feb 28 14:40:25 chanorama mopidy[3234]: INFO     [SpotifyBackend-6] mopidy_spotify.playlists Refreshed 8 Spotify playlists
Feb 28 14:40:25 chanorama mopidy[3234]: INFO     [MainThread] mopidy.commands Starting Mopidy core
Feb 28 14:40:25 chanorama mopidy[3234]: INFO     [MainThread] mopidy.commands Starting Mopidy frontends: HttpFrontend, IrisFrontend, MpdFrontend
Feb 28 14:40:25 chanorama mopidy[3234]: INFO     [HttpFrontend-12] mopidy.http.actor HTTP server running at [::ffff:0.0.0.0]:6680
Feb 28 14:40:25 chanorama mopidy[3234]: INFO     [IrisFrontend-13] mopidy_iris.core Starting Iris 3.56.1
Feb 28 14:40:25 chanorama mopidy[3234]: INFO     [MainThread] mopidy_mpd.actor MPD server running at [::ffff:0.0.0.0]:6600
Feb 28 14:40:25 chanorama mopidy[3234]: INFO     [MainThread] mopidy.commands Starting GLib mainloop
Feb 28 14:40:25 chanorama mopidy[3234]: INFO     [MpdSession-15] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:57978
Feb 28 14:40:25 chanorama mopidy[3234]: INFO     [MpdSession-16] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:57982
Feb 28 14:40:25 chanorama mopidy[3234]: INFO     [MpdSession-17] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:57986
Feb 28 14:40:25 chanorama mopidy[3234]: INFO     [MpdSession-18] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:57990
Feb 28 14:40:25 chanorama mopidy[3234]: INFO     [MpdSession-19] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:57994
Feb 28 14:40:26 chanorama mopidy[3234]: INFO     [MpdSession-20] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:57998
Feb 28 14:40:30 chanorama mopidy[3234]: INFO     [MpdSession-21] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:58002
Feb 28 14:56:19 chanorama mopidy[3234]: INFO     [MpdSession-23] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:58008
Feb 28 14:56:19 chanorama mopidy[3234]: INFO     [MpdSession-24] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:58012
Feb 28 14:56:22 chanorama mopidy[3234]: INFO     [MpdSession-25] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:58016
[... cut ...]
Feb 28 14:56:42 chanorama mopidy[3234]: INFO     [MpdSession-68] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:58170
Feb 28 14:56:42 chanorama mopidy[3234]: INFO     [MpdSession-69] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:58174
Feb 28 14:56:42 chanorama mopidy[3234]: INFO     [MpdSession-70] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:58178
Feb 28 14:56:42 chanorama mopidy[3234]: INFO     [MpdSession-71] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:58182

MiczFlor, 2021-02-28:
Habt ihr hier schon geschaut?
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ#spotify-wont-play-but-after-scan-of-library-it-works

Von dort kommt ihr auf einen Thread, in dem steht:

>I have the same issue, see #901 
>What helps is to restart the mopidy server manually (e.g. out of Phoniebox under Folder&Files, you can click on scan music library) and then mopidy is able to login to spotify without any errors and then you can play Spotify. I have no idea yet why this happens and why mopidy is not able to login directly after booting. 

Hilft das weiter? Restart und scan library?

derchan, 2021-02-28:
> Habt ihr hier schon geschaut?
> https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ#spotify-wont-play-but-after-scan-of-library-it-works
>
> Von dort kommt ihr auf einen Thread, in dem [...]

Face palm.... yup das hilft zumindste bei  **mir**.... hätte ich mir das ganze reinstallieren sparen können. Das blöde ist ich habe sogar genau die Stelle im Troubleshooting angeschaut, es aber irgendwie nicht damit in verbindung gebracht. 

D.H. eigentlich müssten wir da einen Hook nach dem Hochladen im php einführen der den DB refresh auslöst? 

Gibt es einen function call für ein delta update? 

Ich habe mich hier noch nicht wirklich in den Code eingelesen. Es fängt mich aber langsam an zu intressieren :)

seagullki, 2021-02-28:
Gerade probiert, ändert leider nichts.

MarcToth, 2021-03-12:
I had the same problem, fresh installation but no possibility to play local mp3 files. Finally I found out, that there is somehow a difference between scanning with mopidy or scanning with mopidy service.

So instead of

`mopidy local scan`

I did

`sudo mopidyctl local scan`

which finally did the trick and i was able to play the local mp3 files via web gui or by connecting the files to a rfid card.

closed 2022-05-29 by s-martin