MiczFlor/RPi-Jukebox-RFID #1318: setup-buttons-usb-encoder.sh broken after commit 3d7c7cf

derchan, 2021-02-27:
Hi, 

the setup of the buttons usb encoder broke last week in the development branch. I was able to pin it down to one commit 3d7c7cf
On previous commit the installer seems to work fine. 

`git diff 91dc192..3d7c7cf -- ./components`

`~/RPi-Jukebox-RFID $ git checkout 3d7c7cf ./components`
`~/RPi-Jukebox-RFID $ git clean -fd components/`
`sudo components/controls/buttons_usb_encoder/setup-buttons-usb-encoder.sh`

Than is no mapping possible following message appears:

`During the next step you can map your buttons to one of the following available functions:`
`[]`

`Ready to continue? (y/n)`

s-martin, 2021-02-27:
gpio_control was refactored to a class.
This needs to be considered here
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/5be26c528f7731cd4fd1d4ba23913e5f2622d8fa/components/controls/buttons_usb_encoder/map_buttons_usb_encoder.py#L9
and here
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/5be26c528f7731cd4fd1d4ba23913e5f2622d8fa/components/controls/buttons_usb_encoder/buttons_usb_encoder.py#L9

@arne123 could you have a look?

arne123, 2021-02-28:
I Just had a look, 
is my understanding correct, that either the USB Buttons or the GPIO Control are used?
The common point are just the function calls?

s-martin, 2021-02-28:
Yes and yes :)

s-martin, 2021-03-07:
PR #1323 is merged, issue should be fixed.

closed 2021-03-07 by s-martin