MiczFlor/RPi-Jukebox-RFID #1321: 🚀 | FEATURE SUMMARY

jorgemia, 2021-02-28:
## Feature Description

### What functionality would you like to see in your phoniebox?

i.e. `I would love to cook my breakfast eggs with the help of our phoniebox`

### How do you envision the feature to work from a users perspective?

i.e. `In the mornings when I stumble to the bathroom, I want the phoniebox to react on the RFID chip injected under the skin of my right hip. I envision it to walk to the kitchen, get the eggs and a pot out, and start the cooking of the eggs on the stove.`

### Further information that might help

i.e. `the stove is operated with cooking gas, so the implemenation has to make sure the stove is handled with the greatest possible care`

jorgemia, 2021-03-01:
I managed to do it by changing `playlistaddplay` to `playlistadd` here:

https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/9119d3cea67ba1e85fef377bbc4fe1f659d723e6/scripts/rfid_trigger_play.sh#L484

I then added `mpc clear` for it to work properly in in playout_controls.sh before `mpc load`. Otherwise it seems it wasn't loading the playlist if there was already one loaded.

https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/305325d5a9c094e4c47efe6f8ec6d5d7d0fd10d1/scripts/playout_controls.sh#L874-L879

s-martin, 2021-05-14:
Sounds like a useful enhancement. Do you want to write a PR, so this can be integrated in the code base?