MiczFlor/RPi-Jukebox-RFID #1333: on Develop can't import form components Directory in python

derchan, 2021-03-08:
Branch develop

unable to import components after the recent changes.
Since I am not that verse in Python I cant find where you initialize the components directory. 

If i am trying to import 
from components.gpio_control.function_calls import function_calls

I get 
import components.gpio_control.function_calls
ModuleNotFoundError: No module named 'components'
pi@chanorama:~/RPi-Jukebox-RFID/components/control

Can fix it with absolute path:
sys.path.append("/home/pi/RPi-Jukebox-RFID/")

But this does not feel like the proper solution. 
Please point me to the right config file to set path to the components directory for the whole project.

derchan, 2021-03-13:
Ok my idea would be just walking two directoies up 
will fix that and send pull request 

`dir_path = os.path.dirname(os.path.realpath(__file__))
sys.path.append(os.path.split(os.path.split(dir_path)[0])[0])`

derchan, 2021-03-18:
Ok the reason seems to be that in this setup the project path (~/RPi-Jukebox-RFID) is not part of the sys.path. I would like to fix it in the main.py and append that there to sys.path.  Which on whould be the correct py file?

s-martin, 2022-12-20:
Closing as there was no activity for a very long time.

Please check the current release 2.4 or our new implementation 3.2, if this issue still persists.

Feel free to open a new issue, if necessary.

closed 2022-12-20 by s-martin