MiczFlor/RPi-Jukebox-RFID #1335: No sound with internal sound card

etype-44, 2021-03-13:
Hello guys,
i have the following problem.
After a fresh installation (spotify one line installation), there is no sound from the internal sound card (headphone output).
Here is the behavior I have observed:
When the pi is connected to the TV via HDMI and I play music in the raspberry pi os, the sound comes out of the headphone output.
If I play a test mp3 via the card reader, the sound comes exclusively from the TV, not from the headphone output.

amixer scontrols is:
```
Simple mixer control 'Master',0
Simple mixer control 'Capture',0
```

So 'Master' should be the correct iface name:

amixer sget Master
```
Simple mixer control 'Master',0
  Capabilities: pvolume pswitch pswitch-joined
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 65536
  Mono:
  Front Left: Playback 65536 [100%] [on]
  Front Right: Playback 65536 [100%] [on]
```

sudo nano RPi-Jukebox-RFID/settings/Audio_iFace_Name
`Master`

sudo nano /etc/mpd.conf
`audio_output {
        type            "alsa"
        name            "My ALSA Device"
#       device          "hw:0,0"        # optional
#       mixer_type      "hardware"      # optional
#       mixer_device    "default"       # optional
        mixer_control   "Master"                # optional
#       mixer_index     "0"             # optional
`

As I understand it, this is how it should work.
Unfortunately, I am at my wit's end.
Does anyone have an idea what the error is?

derchan, 2021-03-13:
@etype-44 check device number 
[Troubleshooting-FAQ/audio-is-not-working](https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ#audio-is-not-working)

etype-44, 2021-03-13:
Thanks @derchan, that fixed my problems.
I read the FAQ, of course, and thought I tried it too... strange.

aplay -l
```
**** List of PLAYBACK Hardware Devices ****
card 0: b1 [bcm2835 HDMI 1], device 0: bcm2835 HDMI 1 [bcm2835 HDMI 1]
  Subdevices: 3/4
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
card 1: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 4/4
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
```

So i changed
 sudo nano /usr/share/alsa/alsa.conf
```
defaults.ctl.card 1
defaults.pcm.card 1
```
And now it works.
Thanks for the quick help (:

closed 2021-03-13 by etype-44

derchan, 2021-03-13:
@etype-44 no worries, had the same problem a couple of weeks ago just with an USB card.