MiczFlor/RPi-Jukebox-RFID #1339: Config file for Bluetooth Toggle LED pin

ChisSoc, 2021-03-14:
The feature 'Bluetooth Toggle' (#1192) has an optional LED pin to indicate current status. Currently the pin is configured directly in the script which is disadvantageous.  During the pull request (#1262) of the feature it has been suggested by @s-martin to integrate the led pin configuration into `gpio_settings.ini` to have all I/Os in one place.

Now I find the time to add that enhancement.

Proposal: 
- The bluetooth-toggle-script would read and parse `gpio_settings.ini` and take its configuration from there. 
- A specially named section `[BluetoothToggleLed]` will contain a single entry `led_pin` 
- The default value for `enabled` will be adhered
- This new section is of no meaning to the gpio_control.py script. Actually that script prints a warning as the section is unknown to it. I can certainly adapt the gpio_control.py to ignore this special section. 

What it means in the end: The gpio_control.ini file will be parsed by more than one script and changes to the config structure etc must be always be made in two places. But there are more different scripts to come if all GPIO config stuff should go into a single file in the future.

Let me know if this is what you had in mind.

s-martin, 2021-03-14:
I think that’s a good idea.

ChisSoc, 2021-03-22:
Closing this issues, as it is merged.

closed 2021-03-22 by ChisSoc