MiczFlor/RPi-Jukebox-RFID #1341: 🐛 | Subfolders do not play with RFID Tag

zaggul, 2021-03-15:
## Bug

A RFID Tag was assigned to a folder called 'Test' and it includes subfolders with Episodes of a story called '1 - Story 1', '2 - Story 2'. The folder itsself does only include subfolders, no audio in the root folder.
I would expect Phoenibox to play all subfolders if the RFID Tag is assigned to 'Test'. But nothing is played. 

### What I did

I assigned the RFID Tag to 'Test'.

### What happened

Nothing. If I use the webinterface to play 'Test' it perfectly works.

### I expected this to happen

Play all episodes which are in subfolders of 'Test'.

### Further information that might help

## Software

### Base image and version

### Branch / Release

### Installscript

## Hardware

### RaspberryPi version

Raspberry Pi 4

### RFID Reader

16c0:27db Van Ooijen Technische Informatica Keyboard

### Soundcard

hifiberry

s-martin, 2022-12-20:
Closing as there was no activity for a very long time.

Please check the current release 2.4 or our new implementation 3.2, if this issue still persists.

Feel free to open a new issue, if necessary.

alesrosina, 2024-12-22:
This is actually happening to me also. On 3.6.0-alpha.

Only flat folder structure works, subfolders never work.

pabera, 2024-12-23:
Can you do me a favor? Can you copy the card db entry for this particular request here please? You'll find it in the following file: `shared/settings/cards.yaml`

alesrosina, 2024-12-23:
> Can you do me a favor? Can you copy the card db entry for this particular request here please? You'll find it in the following file: `shared/settings/cards.yaml`

Sure:

```yml
'3002421778':
  alias: play_folder
  args:
  - Papa Moll
```

And folder structure is sth as follows

```
Papa Mol
 - CD1
   - File1.mp3
   - ...
 - CD2
   - SomethingElse1.mp3
```

pabera, 2024-12-27:
No worries.

In 3.6.0, `play_folder` is defined like this

```yaml
player.ctrl.play_folder(folder: str, recursive: bool = False) -> None: Playback a music folder.
```

`recursive` is `false` by default, so it's probably ok that it does play all sub-folders. In that version, it's also not possible to change this value in the Web App. 

But you should be able to update the `cards.yaml` like this.

```
'3002421778':
  alias: play_folder
  args:
  - Papa Moll
  - true
```

Give it a try and let me know, if it worked or not.

alesrosina, 2024-12-30:
> No worries.
> 
> In 3.6.0, `play_folder` is defined like this
> 
> ```python
> player.ctrl.play_folder(folder: str, recursive: bool = False) -> None
> ```
> 
> `recursive` is `false` by default, so it's probably ok that it does play all sub-folders. In that version, it's also not possible to change this value in the Web App.
> 
> But you should be able to update the `cards.yaml` like this.
> 
> ```yaml
> '3002421778':
>   alias: play_folder
>   args:
>   - Papa Moll
>   - true
> ```
> 
> Give it a try and let me know, if it worked or not.

Perfect, thank you. Will give it a try and will let you know, if it works. 
Is there an option, that this recursive is enabled by default?