MiczFlor/RPi-Jukebox-RFID #1347: Toggle Audio Sink: no switchback to disabled audio Output

greg4b, 2021-03-20:
## Bug
No switch back possible to other audio output - only one switch possible.

### What I did
Audio output hearable (Output 1)
click onto "Toggle Audio Sink"
Audio switches to BT Audio (Output 2)
click  onto "Toggle Audio Sink"
[image]

### What happened
No switch from Output 2 (hearable) to Output 1

### I expected this to happen
Audio from Output 1 hearable.

### Further information that might help

## Software
	
Version
2.3rc1 - 8d49dfa - develop
Edition
Classic edition (barebones)
MPD Server Status
active (running) since Fri 2021-03-19 21:41:13 GMT; 12h ago

### Base image and version
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

### Branch / Release
Refresh index: 100% (419/419), done.
On branch develop
Your branch is up to date with 'origin/develop'.

### Installscript

## Hardware
Output1: Audio Jack 
Output2: BT Speaker S10, Major Device Type Audio,  Minor Device Type Hands Free

### RaspberryPi version
Raspberry Pi 3 Model B Rev 1.2

### RFID Reader
  iManufacturer           1 HXGCoLtd

### Soundcard
None - internal Audio Jack

### Other notable hardware
None

/etc/asound.conf
```
pcm.btspeaker {
    type plug
    slave.pcm {
        type bluealsa
        service "org.bluealsa"
        device "59:43:46:a2:36:be"
        profile "a2dp"
        delay -20000
    }
    hint {
        show on
        description "S10"
    }
}
```

/etc/mpd.conf
``
audio_output {
        type            "alsa"
        name            "My ALSA Device"
        device          "hw:0,0"        # optional
#       mixer_type      "hardware"      # optional
#       mixer_device    "default"       # optional
        mixer_control   "Headphone"             # optional
#       mixer_index     "0"             # optional
}

# The second entry belongs to the bluetooth device
audio_output {
        type            "alsa"
        name            "S10" # This name will show up in the Web Interface Status
        device          "btspeaker"      # This is the pcm.btheadphone device from the asound.conf
        mixer_type      "software"
        auto_resample   "no"
        auto_format     "no"
        enabled         "no"               # Default is disabled, keepin the soundcard as primary output
}

ChisSoc, 2021-03-21:
Could you please try switching to/back directly from the console with the debug option and paste the output here.

~~~
$ cd /home/pi/RPi-Jukebox-RFID/components/bluetooth-sink-switch
$ ./bt-sink-switch.py toggle debug
~~~

Did you use the default install directory (i.e. /home/pi/RPi-Jukebox-RFID)?

Please also try to switch with `mpc enable only 1` and `mpc enable only 2`.

greg4b, 2021-03-21:
Hello ChisSoc,

Output of `$ ./bt-sink-switch.py toggle debug` 
when playing from **jack (standard audio output)**:
`21.03.2021 07:58:25 - bt-sink-switch.py - DEBUG: b'Output 1 (My ALSA Device) is enabled\nOutput 2 (S10) is disabled\n'
21.03.2021 07:58:26 - bt-sink-switch.py - DEBUG: b'Device 59:43:46:A2:36:BE (public)\n\tName: S10\n\tAlias: S10\n\tClass: 0x00240408\n\tIcon: audio-card\n\tPaired: yes\n\tTrusted: yes\n\tBlocked: no\n\tConnected: yes\n\tLegacyPairing: no\n\tUUID: Headset                   (00001108-0000-1000-8000-00805f9b34fb)\n\tUUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)\n\tUUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)\n\tUUID: Advanced Audio Distribu.. (0000110d-0000-1000-8000-00805f9b34fb)\n\tUUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)\n\tUUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)\n'
Switched audio sink to "Output 2"
21.03.2021 07:58:27 - bt-sink-switch.py - DEBUG: b'Output 1 (My ALSA Device) is enabled\nOutput 2 (S10) is enabled\nOutput 1 (My ALSA Device) is disabled\nOutput 2 (S10) is enabled\n'
21.03.2021 07:58:27 - bt-sink-switch.py - DEBUG: bt_check_mpc_err()
21.03.2021 07:58:27 - bt-sink-switch.py - DEBUG: b'LIED #1/23   0:45/3:50 (19%)\nvolume:100%   repeat: off   random: off   single: off   consume: off\n'
`
--> OK - Audio switches to BT

when playing **from BT**:
`
21.03.2021 08:01:50 - bt-sink-switch.py - DEBUG: b'Output 1 (My ALSA Device) is disabled\nOutput 2 (S10) is enabled\n'
21.03.2021 08:01:50 - bt-sink-switch.py - DEBUG: b'Device 59:43:46:A2:36:BE (public)\n\tName: S10\n\tAlias: S10\n\tClass: 0x00240408\n\tIcon: audio-card\n\tPaired: yes\n\tTrusted: yes\n\tBlocked: no\n\tConnected: yes\n\tLegacyPairing: no\n\tUUID: Headset                   (00001108-0000-1000-8000-00805f9b34fb)\n\tUUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)\n\tUUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)\n\tUUID: Advanced Audio Distribu.. (0000110d-0000-1000-8000-00805f9b34fb)\n\tUUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)\n\tUUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)\n'
Switched audio sink to "Output 2"
21.03.2021 08:01:50 - bt-sink-switch.py - DEBUG: b'Output 1 (My ALSA Device) is disabled\nOutput 2 (S10) is enabled\nOutput 1 (My ALSA Device) is disabled\nOutput 2 (S10) is enabled\n'
21.03.2021 08:01:50 - bt-sink-switch.py - DEBUG: bt_check_mpc_err()
21.03.2021 08:01:50 - bt-sink-switch.py - DEBUG: b'LIED #2/23   0:18/2:15 (13%)\nvolume:100%   repeat: off   random: off   single: off   consume: off\n'
`
--> nOK - Audio contiues on BT

Yes, default install directory.

Switching by `mpc enable only 1` / `mpc enable only 2` works. 
One minor observation here - after switching back and forth several times:
BT --> jack (2 --> 1): Audio contiues
jack --> BT (1 --> 2): player pauses (web interface jumps to pause - pressing play: audio plays )

Thanks,
Gregor

ChisSoc, 2021-03-21:
Hey, 

thanks for the fast reply. I have got the issue pinned down. The problem is a not-so-robust regexp in the script. I will make an update ASAP. 

Meanwhile, to quickly make it work on your end, rename your second device (bluetooth) **without** any '1'. For now best to avoid any numbers in the naming of both devices.

So, e.g., change your mpd.conf to
~~~
audio_output {
        type            "alsa"
        name            "Speakers"      # Instead of S10 <-- HERE IS THE CHANGE!
        device          "btspeaker"     
...
~~~

greg4b, 2021-03-21:
Hi,

thanks for looking into!

Workaround done - renamed to `Speakers` and it works like a charm!

Danke & VG Gregor

ChisSoc, 2021-03-21:
Thanks for the feedback. Enjoy it :-)

ChisSoc, 2021-03-22:
Closing this issue, as the fix is merged.

greg4b, 2021-03-27:
Issue closed - workaround https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/1347#issuecomment-803659401 fixes issue.

Merge https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/1354#issue-598342023 not tested.

Issue closed

greg4b, 2021-03-27:
Issue closed - workaround #1347 (comment) fixes issue.

Merge #1354 (comment) not tested.

Issue closed

closed 2021-03-27 by greg4b