MiczFlor/RPi-Jukebox-RFID #1358: 🚀 | Modularize RFID Readers

ChisSoc, 2021-03-22:
## Feature Description

The time has come: Modularize handling of different RFID readers and enable per-reader optional settings

### How do you envision the feature to work from a users perspective?

For users: 
- RFID reader can be selected and switched easily with copying files around. 
- Optional parameters give freedom to e.g. move reset pin of MFRC522 reader to a different pin
For developrs:
- Adding a new reader is as simple as creating a directory and placing the new reader's code inside. Everything else is done automatically - no need to register the new reader in existing Phoniebox code. For the new reader's module a handy template is provided.

### Further information that might help

Following the discussion #1329, I just had to give it some thought how to achieve a modularized RFID reader concept that is easy on developers for integrating new RFID readers. I have done some coding for proof of concept and like to propose that for the Phoniebox. @s-martin: You obviously also put some thought to this issue and I would value your opinion. I think this will be valuable even before Version 3.0.

 The proof-of-concept branch is here: 
[https://github.com/ChisSoc/RPi-Jukebox-RFID/tree/modreader](https://github.com/ChisSoc/RPi-Jukebox-RFID/tree/modreader)

Here is the idea:
- A new module handles the loading and configuration of the appropriate RFID handler (`readersupport.py`)
- This module searches all sub-directories in `rfid-readers` for possible readers
- Based on configuration the rfid reader is loaded **dynamically** which is very neat as this means, we do not need to have all the dependencies and we can avoid all kinds of long if-else structures!
- Very few conventions need to be followed for adding new readers. See `template_new_reader.py`
- Due to the generic approach and dynamic module loading, new readers can be added by simply adding a new subdirectory. No need to modify any existing code!

- A new `register_reader.py` is there for RFID reader selection and customization. This is now located in `rfid-readers`. 
- The settings are stored in `settings\rfid_reader.ini` replacing the `deviceName.txt` file

I have stayed as close to the original API of the `Reader.py` as possible. So far I have written and tested the code only for the default _Generic USB Reader_ and a _Template Reader_. I will add the other readers if this proposal meets approval :-)

Note: The now surplus original files are not yet deleted on the branch! daemon_rfid_reader.py is already adjusted to the new structure with only minimal changes necessary.

### Some experiences:

Selecting a reader. You can see which module and files are avaiable in this proof-of-concept. 
`generic_usb` is the RFID reader previously known as `Reader.py`
~~~
pi@exaberry:~/RPi-Jukebox-RFID/components/rfid-reader $ ./register_reader.py 

Choose Reader Module form list:

  0: Template RFID Reader Module              (Module: template_new_reader.template_new_reader in /home/pi/RPi-Jukebox-RFID/components/rfid-reader/template_new_reader/template_new_reader.py)
  1: Generic USB Reader                       (Module: generic_usb.generic_usb in /home/pi/RPi-Jukebox-RFID/components/rfid-reader/generic_usb/generic_usb.py)

Reader module number: 1

Choose USB device from list:
  0: HID 046a:0023
  1: HID 046a:0023

Device Number: 0
22.03.2021 22:40:55 - readersupport.py     - INFO    : Config file written to '/home/pi/RPi-Jukebox-RFID/components/rfid-reader/../../settings/rfid_reader.ini'.
~~~

And the config file looks like this 
~~~
$ cat /home/pi/RPi-Jukebox-RFID/components/rfid-reader/../../settings/rfid_reader.ini
[ReaderType]
reader_module = generic_usb

[generic_usb]
device_name = HID 046a:0023
~~~

The directory structure of rfid-reader:
~~~
+ rfid-reader
   + generic_usb
     + generic_usb.py
     + requirements.txt
   + other_spi
     + other_spi.py
     + requirements.txt
~~~

Naming scheme proposal for the directory / module structure:
- Note: We deviate slighlty from the python naming convention for modules here, as the module name is identical
 to directory name and that should be descriptive enough to know what is going on when looking at the file
 structure in git. Also it should specific enough to allow for later new module additions.
- Naming scheme for the module
  - <type_of_reader> _<io_bus> _<other_specials_like_special_lib>
  - e.g. generic_usb/generic_usb.py
  - e.g. pn532_spi/pn532_spi.py
  - ...
- DO NOT use '-' in the filename! Even if that is the suggested naming convention in the Phoniebox project.
  That is no a valid character in python for module names. It makes loading the module statically a pain.

s-martin, 2021-03-22:
I really like that approach, especially that this is kind of self-learning.
I think this could make maintenance much easier.

I think it's worth that you open a PR, so other people can have a look at it. You can mark a PR as Draft, so it can't be merged yet.

We have introduced a few naming conventions, please have a look and check, if this works for you:
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/develop/CONTRIBUTING.md#naming-conventions

Thanks for your contribution, I think this will greatly improve Phoniebox!

ChisSoc, 2021-03-23:
I double-checked out the naming convention. I think the naming convention fits very well already, except for one aspect:
- Naming scheme from module goes form general to specific. In this case, the most general part is the name of the reader or the name of the chip. This. in some cases, can be connected via different busses e.g. pn532-spi and pn532-uart and pn532-i2c. And even more differentiation could be added
- We need to diverge from the dash rule '-': If we want the filename to be a valid python symbol constituent, it must be free of '-' as that is not a word character but denotes the minus sign. i.e. is an operator. Since these are Python modules, we should respect that requirement from Python, to make these modules usable also in different scripts where they may be loaded statically.

s-martin, 2022-12-19:
Should be part of future3, so closing this issue.

closed 2022-12-19 by s-martin