MiczFlor/RPi-Jukebox-RFID #1369: 🐛 | RPi-Jukebox and Pirate Audio do not work together

kkmichi, 2021-04-06:
## Bug
Hello, I try to get a Jukebox for kids working with **Raspberry Zero WH**, **onoff-Shim** and **Pirate Audio 3W DAC**. After many attempts, including manual configuration of /etc/mpd.conf, /etc/asound.conf, etc. I do not get it working. The RPi-Jukebox/Phoniebox does not seem to work together with Pirate Audio DAC (audio, display and buttons). I tried many different things (always with new, clean install), but for the description below I tried to follow the "basic" attempt. What else can I do? (for information, I also open this issue at pirate audio https://github.com/pimoroni/pirate-audio/issues). I am a longtime Linux user with basic knowledge, but without deeper knowledge or even programming experience.

### What I did
1. fresh install of raspbian lite (Samsung SD)
2. edit of /boot/config.txt:
- config_hdmi_boost=4
- #dtparam=audio=on
- dtoverlay=hifiberry-dac
- dtparam=spi=on
- gpio=25=op,dh
3. create file /boot/wpa_supplicant.conf for Wlan-connection
4. create empty file /boot/ssh 
5. because of error messages: input `sudo dpkg-reconfigure locale` and `export LC_ALL=C` 
6. insert static IP configuration in /etc/dhcpcd.conf
7. `sudo apt update` and `sudo apt full-upgrade`
8. set up onoff Shim: `curl https://get.pimoroni.com/onoffshim | bash`
9. reboot
10. Set up of RPi-Jukebox `cd; rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/develop/scripts/installscripts/buster-install-default.sh; chmod +x buster-install-default.sh; GIT_BRANCH=develop bash ./buster-install-default.sh` with following selection: 
- WiFi: no
- iFace: No, Master
- Spotify: no
- MPD: yes
- Defalut location: yes
- GPIO-Control: yes
- last part of output: 
Configuring MPD...
Updating DB (#2) ...
volume: n/a   repeat: off   random: off   single: off   consume: off
Starting dhcpcd service...
Synchronizing state of dhcpcd.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable dhcpcd
Setting owner and permissions for directories...
amixer: Cannot find the given element from control default
- select RFID-Reader: yes (1, Neuftech USB)
11. reboot
12. Set up of Pirate Audio:
- `sudo raspi-config nonint do_spi 0`
- `git clone https://github.com/pimoroni/pirate-audio`
- `cd pirate-audio/mopidy`
- `sudo ./install.sh`
13. reboot

### What happened

- after I have proceeded as described above and reeboot, the pi started, including display. The display shows "Visit http://192.168.178.66:6680 to select content". The volume down button reduces the volume bar in the display, the other buttons give no feedback in display. Pushing the volume button has no impact on Volume status in phoniebox web interface and vice versa.
- upload of Song with Phoniebox and playing it (via web interface) gives no feedback on display. pushing buttons has no impact.

### I expected this to happen

I expected, that I can use RPi-Jukebox / phoniebox with RFID management and use the pirate audio for audio output, display and buttons. I expected it to work, as it is described in https://github.com/MiczFlor/Rpi-Jukebox-RFID/blob/develop/components/audio/PirateAudioHAT/README.md and https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/950.

### Further information that might help
notes:
https://github.com/MiczFlor/Rpi-Jukebox-RFID/blob/develop/components/audio/PirateAudioHAT/README.md
- /etc/asound.conf does not exist. I tried the procedure described above with the notes of https://github.com/MiczFlor/Rpi-Jukebox-RFID/blob/develop/components/audio/PirateAudioHAT/README.md and created asound.conf, added (not replaced) content to mpd.conf (had to comment out bcm24 = volume_up,active_low,250)

## Software

### Base image and version

`2021-03-04-raspios-buster-armhf-lite.img`

### Branch / Release

### Installscript

onoff shim: `curl https://get.pimoroni.com/onoffshim | bash`
Jukebox: `cd; rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/develop/scripts/installscripts/buster-install-default.sh; chmod +x buster-install-default.sh; GIT_BRANCH=develop bash ./buster-install-default.sh`
Pirate Audio:  `git clone https://github.com/pimoroni/pirate-audio
cd pirate-audio/mopidy
sudo ./install.sh`

## Hardware
Raspberry Pi Zero WH
on off Shim
Pirate Audio 3W DAC
Neuftech USB Reader

### RaspberryPi version

`Zero WH`

### RFID Reader

`Neuftech USB Reader`

### Soundcard

`Pirate Audio 3W DAC`

### Other notable hardware

s-martin, 2021-04-06:
Hi, although the Readme unfortunately doesn’t really state this, but at the moment the Pirate Audio HAT only works with the Spotify edition.

There is a PR #1112 which should implement this also for Classic edition, but unfortunately at the moment  I don’t really have the time to fix this.

s-martin, 2022-12-24:
Closing now as a duplicate of #1109

closed 2022-12-24 by s-martin