MiczFlor/RPi-Jukebox-RFID #1389: 🚀 | FEATURE SUMMARY

manajoe, 2021-04-23:
## Hack: Switchable Audio Interface with 2 different sound cards

### Prerequisites
- 2 different sound cards, e.g. onboard and additional USB sound card
- configuring both sound cards in various config files, in my setup :
#### -/etc/asound.conf
`pcm.usbaudio {
        type hw
        card 1
}
ctl.usbaudio {
        type hw
        card 1
}
pcm.headphone {
        type hw
        card 0
}
ctl.headphone {
        type hw
        card 0
}`
i.e. `I would love to cook my breakfast eggs with the help of our phoniebox`

### How do you envision the feature to work from a users perspective?

i.e. `In the mornings when I stumble to the bathroom, I want the phoniebox to react on the RFID chip injected under the skin of my right hip. I envision it to walk to the kitchen, get the eggs and a pot out, and start the cooking of the eggs on the stove.`

### Further information that might help

i.e. `the stove is operated with cooking gas, so the implemenation has to make sure the stove is handled with the greatest possible care`

s-martin, 2021-05-14:
Thanks!
I’m wondering, if we should add this to the wiki or if this could integrated in the code base with a PR?

Thilo-90, 2021-11-22:
I modified the script a bit. I cleared out the steps for saving the volume and named the devices to my likings.
And the toggle is working just fine.

I am currently trying to get the command `bash RPi-Jukebox-RFID/scripts/userscripts/switch_audio_output.sh`, which toggles the output devices, onto a button. (see [discussion-1638](https://github.com/MiczFlor/RPi-Jukebox-RFID/discussions/1638)

ddbkva, 2021-12-07:
how can activate this script ?  bash RPi-Jukebox-RFID/scripts/userscripts/switch_audio_output.sh is not working
my usb speaker are on Card 2 so i only change the settings at /etc/asound.conf ?

manajoe, 2021-12-08:
@ddbkva I'm not sure, in my setup, exactly this command works
Additionally, i did a chmod +x `RPi-Jukebox-RFID/scripts/userscripts/switch_audio_output.sh`.
Did you also read https://github.com/MiczFlor/RPi-Jukebox-RFID/discussions/1638, maybe this helps?