MiczFlor/RPi-Jukebox-RFID #1395: 🐛 | BUG SUMMARY

Miko1985, 2021-04-27:
## Bug

### What I did

Ich habe auf einen Pi4 das aktuellste Raspbian installiert und die One-line Spotify Version aufgespielt.

Es werden sämtlich Alben über Spotify gestreamt. Allerdings kann ich keine Playlisten abspielen.
Hier kommt in Mopidy auch immer die Fehlermeldung das die Titel nicht hinzugefügt werden können.

Gibt es dafür eine Lösung, oder können keine Playlisten in der aktuellen Version abgespielt werden.

Ollimoller, 2021-05-01:
Hat keiner eine Idee wo das Problem liegen könnte?

s-martin, 2021-05-01:
What's the output of `journalctl -u mopidy`?

Miko1985, 2021-05-02:
Hello,

the Output is:
May 02 08:23:59 raspberrypi mopidy[5606]: INFO     [SpotifyBackend-6] mopidy_spotify.lookup Failed to lookup 'spotify:playlist:5I3SybkxkyE1VCPgRDm77p': Playlist Web API lookup failed
May 02 08:23:59 raspberrypi mopidy[5606]: INFO     [MpdSession-421] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:59552

pboos, 2021-05-05:
Same here. Just wondered why only albums worked (e.g. `spotify:album:5EeUZitOFX7Nad5yp2CFud`) and not playlists (e.g. `spotify:playlist:4GyC60kHGrESOcZmj3cD2N`). I also get the same error as above:
```
May 05 18:16:44 raspberrypi mopidy[21234]: INFO     [SpotifyBackend-6] mopidy_spotify.lookup Failed to lookup 'spotify:playlist:4GyC60kHGrESOcZmj3cD2N': Playlist Web API lookup failed
```

Just set up my box today. So was wondering if it is me.

Miko1985, 2021-05-07:
Have you found a solution to the problem?

pboos, 2021-05-13:
Sadly for me that didn't solve the issue 😢 . But maybe my issue is different. Playlists work fine when I use the web interface. But it won't work using RFID tags.

s-martin, 2021-05-14:
> Sadly for me that didn't solve the issue 😢 . But maybe my issue is different. Playlists work fine when I use the web interface. But it won't work using RFID tags.

What’s the output of `journalctl -u mopidy `?

pboos, 2021-05-24:
> > Sadly for me that didn't solve the issue 😢 . But maybe my issue is different. Playlists work fine when I use the web interface. But it won't work using RFID tags.
> 
> What’s the output of `journalctl -u mopidy `?

Output is (from start until scanning of RFID tag with playlist). Same error as others above.:
```
May 24 20:42:29 raspberrypi systemd[1]: Started Mopidy music server.
May 24 20:42:45 raspberrypi mopidy[7315]: INFO     [MainThread] mopidy.__main__ Starting Mopidy 3.1.1
May 24 20:42:45 raspberrypi mopidy[7315]: INFO     [MainThread] mopidy.config Loading config from builtin defaults
May 24 20:42:45 raspberrypi mopidy[7315]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy.conf
May 24 20:42:45 raspberrypi mopidy[7315]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy-local.conf
May 24 20:42:45 raspberrypi mopidy[7315]: INFO     [MainThread] mopidy.config Loading config from file:///etc/mopidy/mopidy.conf
May 24 20:42:45 raspberrypi mopidy[7315]: INFO     [MainThread] mopidy.config Loading config from command line options
May 24 20:42:45 raspberrypi mopidy[7315]: ERROR    [MainThread] mopidy.internal.log Loading logging config '/etc/mopidy/logging.conf' failed. 'formatters'
May 24 20:42:50 raspberrypi mopidy[7315]: INFO     [MainThread] mopidy.__main__ Enabled extensions: m3u, local, http, file, iris, spotify, mpd, softwaremixer, stream
May 24 20:42:50 raspberrypi mopidy[7315]: INFO     [MainThread] mopidy.__main__ Disabled extensions: none
May 24 20:42:54 raspberrypi mopidy[7315]: INFO     [MainThread] mopidy.commands Starting Mopidy mixer: SoftwareMixer
May 24 20:42:54 raspberrypi mopidy[7315]: INFO     [MainThread] mopidy.commands Mixer volume set to 30
May 24 20:42:54 raspberrypi mopidy[7315]: INFO     [MainThread] mopidy.commands Starting Mopidy audio
May 24 20:42:54 raspberrypi mopidy[7315]: INFO     [MainThread] mopidy.commands Starting Mopidy backends: FileBackend, M3UBackend, StreamBackend, SpotifyBackend, LocalBackend
May 24 20:42:54 raspberrypi mopidy[7315]: INFO     [Audio-2] mopidy.audio.actor Audio output set to "alsasink"
May 24 20:42:57 raspberrypi mopidy[7315]: ERROR    [SpotifyBackend-6] mopidy_spotify.web Authorization failed, not attempting Spotify API request. Please get new credentials from https://www.mopidy.com/authenticate and/or restart Mopidy to resolve this problem.
May 24 20:42:57 raspberrypi mopidy[7315]: ERROR    [SpotifyBackend-6] mopidy_spotify.web OAuth token refresh failed: invalid_client Client not known.
May 24 20:42:57 raspberrypi mopidy[7315]: ERROR    [SpotifyBackend-6] mopidy_spotify.web Failed to load Spotify user profile
May 24 20:42:57 raspberrypi mopidy[7315]: INFO     [MainThread] mopidy.commands Starting Mopidy core
May 24 20:42:57 raspberrypi mopidy[7315]: INFO     [MainThread] mopidy.commands Starting Mopidy frontends: HttpFrontend, IrisFrontend, MpdFrontend
May 24 20:42:57 raspberrypi mopidy[7315]: INFO     [HttpFrontend-12] mopidy.http.actor HTTP server running at [::ffff:0.0.0.0]:6680
May 24 20:42:57 raspberrypi mopidy[7315]: INFO     [IrisFrontend-13] mopidy_iris.core Starting Iris 3.58.0
May 24 20:42:57 raspberrypi mopidy[7315]: INFO     [MainThread] mopidy_mpd.actor MPD server running at [::ffff:0.0.0.0]:6600
May 24 20:42:57 raspberrypi mopidy[7315]: INFO     [MainThread] mopidy.commands Starting GLib mainloop
May 24 20:43:10 raspberrypi mopidy[7315]: INFO     [SpotifyEventLoop] mopidy_spotify.backend Logged in to Spotify in online mode
May 24 20:43:12 raspberrypi mopidy[7315]: INFO     [MpdSession-15] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52360
May 24 20:43:12 raspberrypi mopidy[7315]: INFO     [MpdSession-16] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52364
May 24 20:43:12 raspberrypi mopidy[7315]: INFO     [MpdSession-17] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52368
May 24 20:43:12 raspberrypi mopidy[7315]: INFO     [MpdSession-18] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52372
May 24 20:43:15 raspberrypi mopidy[7315]: INFO     [MpdSession-19] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52376
May 24 20:43:15 raspberrypi mopidy[7315]: INFO     [MpdSession-20] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52380
May 24 20:43:15 raspberrypi mopidy[7315]: INFO     [MpdSession-21] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52384
May 24 20:43:16 raspberrypi mopidy[7315]: INFO     [MpdSession-22] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52388
May 24 20:43:16 raspberrypi mopidy[7315]: INFO     [MpdSession-23] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52392
May 24 20:43:16 raspberrypi mopidy[7315]: INFO     [MpdSession-24] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52396
May 24 20:43:16 raspberrypi mopidy[7315]: INFO     [SpotifyBackend-6] mopidy_spotify.lookup Failed to lookup 'spotify:playlist:4GyC60kHGrESOcZmj3cD2N': Playlist Web API lookup failed
May 24 20:43:17 raspberrypi mopidy[7315]: INFO     [MpdSession-25] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52400
May 24 20:43:17 raspberrypi mopidy[7315]: INFO     [MpdSession-26] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52404
May 24 20:43:17 raspberrypi mopidy[7315]: INFO     [MpdSession-27] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52408
May 24 20:43:17 raspberrypi mopidy[7315]: INFO     [MpdSession-28] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52412
May 24 20:43:18 raspberrypi mopidy[7315]: INFO     [MpdSession-29] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52416
```

I just ran the commands above again (this time it upgraded mopidy-iris (`Successfully installed mopidy-Iris-3.58.0`). Output above is after update & restart.

rmk-ch, 2021-05-26:
Hi
When reading logfiles you should also scan for other errors. These errors seem very relevant:
```
May 24 20:42:57 raspberrypi mopidy[7315]: ERROR    [SpotifyBackend-6] mopidy_spotify.web Authorization failed, not attempting Spotify API request. Please get new credentials from https://www.mopidy.com/authenticate and/or restart Mopidy to resolve this problem.
May 24 20:42:57 raspberrypi mopidy[7315]: ERROR    [SpotifyBackend-6] mopidy_spotify.web OAuth token refresh failed: invalid_client Client not known.
May 24 20:42:57 raspberrypi mopidy[7315]: ERROR    [SpotifyBackend-6] mopidy_spotify.web Failed to load Spotify user profile

```
So please get new credentials from https://www.mopidy.com/authenticate and/or restart Mopidy to resolve this problem.

pboos, 2021-05-27:
@rmk-ch I actually didn't even notice 🤦 . But now I checked and saw the config was there. I had entered the data during setup and those where in the mopidy.conf. But tried as the log said and after authenticating anew and replacing those configs and restarting, it still shows the same error. I wonder if the token refresh is the problem 🤔 . Also strange is, that albums work fine, only playlists don't work. And on the web frontend everything works. It is only when using the rfid cards that playlists don't work.

I wonder why it says `invalid_client` 🤔 .
```
invalid_client Client not known.
```

Also somehow login worked still a few lines later?
```
May 24 20:43:10 raspberrypi mopidy[7315]: INFO     [SpotifyEventLoop] mopidy_spotify.backend Logged in to Spotify in online mode
```

Not 100% sure where the problem lies. Kind of out ideas on what to try. Also tried rescanning the library (as in some other places suggested).

s-martin, 2024-04-15:
#2315 fixed Spotify and brought various improvements on `develop`.

Closing. Feel free to reopen, if your issue still persists.

closed 2024-04-15 by s-martin