MiczFlor/RPi-Jukebox-RFID #1400: 🐛 | BUG SUMMARY

Duky87, 2021-05-03:
## Bug

### What I did

I installed RPI Jukebox with Spotify via the on-line installer on the latest Raspbian OS lite. 

### What happened

During the installation the following line is shown:

pykka 3.0.0 has requirement importlib_metadata<4.0,>=1.6; python_version < "3.8", but you'll have importlib-metadata 4.0.1 which is incompatible.

I have not installed anything additional, just the plain Raspian OS lite with the latest updates. 
e.g.
sudo apt-get update
sudo apt-get upgrade

Therefore the Jukebox is not working it starts, and I can connect via the web-interface. I also can upload files and can connect them to a RFID card, but if i click play via the web-interface or via the RFID it doesnt work. Nothing happens.

E.g. if I click on volume up in the web-interface, following message appears:
An error occorured: Execution failed Command: /home/pi/RPi-Jukebox-RFID/scripts/playout_controls.sh -c=volumeup Output: RC: .1

### I expected this to happen

It should play the music i uploaded.

### Further information that might help

journalctl -u mopidy
-- Logs begin at Mon 2021-05-03 23:26:57 BST, end at Mon 2021-05-03 23:28:21 BST. --
May 03 23:27:09 raspberrypi systemd[1]: Starting Mopidy music server...
May 03 23:27:09 raspberrypi systemd[1]: Started Mopidy music server.
May 03 23:27:14 raspberrypi mopidy[3080]: Traceback (most recent call last):
May 03 23:27:14 raspberrypi mopidy[3080]:   File "/usr/local/lib/python3.7/dist-packages/pkg_resources/__init__.py", line 568, in _build_master
May 03 23:27:14 raspberrypi mopidy[3080]:     ws.require(__requires__)
May 03 23:27:14 raspberrypi mopidy[3080]:   File "/usr/local/lib/python3.7/dist-packages/pkg_resources/__init__.py", line 886, in require
May 03 23:27:14 raspberrypi mopidy[3080]:     needed = self.resolve(parse_requirements(requirements))
May 03 23:27:14 raspberrypi mopidy[3080]:   File "/usr/local/lib/python3.7/dist-packages/pkg_resources/__init__.py", line 777, in resolve
May 03 23:27:14 raspberrypi mopidy[3080]:     raise VersionConflict(dist, req).with_context(dependent_req)
May 03 23:27:14 raspberrypi mopidy[3080]: pkg_resources.ContextualVersionConflict: (importlib-metadata 4.0.1 (/usr/local/lib/python3.7/dist-packages), Requirement.parse('importlib_metadata<4.0,>=1.6; python_version < "3.8"'), {'Pykka'})
May 03 23:27:14 raspberrypi mopidy[3080]: During handling of the above exception, another exception occurred:
May 03 23:27:14 raspberrypi mopidy[3080]: Traceback (most recent call last):
May 03 23:27:14 raspberrypi mopidy[3080]:   File "/usr/bin/mopidy", line 6, in <module>
May 03 23:27:14 raspberrypi mopidy[3080]:     from pkg_resources import load_entry_point
May 03 23:27:14 raspberrypi mopidy[3080]:   File "/usr/local/lib/python3.7/dist-packages/pkg_resources/__init__.py", line 3242, in <module>
May 03 23:27:14 raspberrypi mopidy[3080]:     @_call_aside
May 03 23:27:14 raspberrypi mopidy[3080]:   File "/usr/local/lib/python3.7/dist-packages/pkg_resources/__init__.py", line 3226, in _call_aside
May 03 23:27:14 raspberrypi mopidy[3080]:     f(*args, **kwargs)
May 03 23:27:14 raspberrypi mopidy[3080]:   File "/usr/local/lib/python3.7/dist-packages/pkg_resources/__init__.py", line 3255, in _initialize_master_working_set
May 03 23:27:14 raspberrypi mopidy[3080]:     working_set = WorkingSet._build_master()
May 03 23:27:14 raspberrypi mopidy[3080]:   File "/usr/local/lib/python3.7/dist-packages/pkg_resources/__init__.py", line 570, in _build_master
May 03 23:27:14 raspberrypi mopidy[3080]:     return cls._build_from_requirements(__requires__)
May 03 23:27:14 raspberrypi mopidy[3080]:   File "/usr/local/lib/python3.7/dist-packages/pkg_resources/__init__.py", line 583, in _build_from_requirements
May 03 23:27:14 raspberrypi mopidy[3080]:     dists = ws.resolve(reqs, Environment())
May 03 23:27:14 raspberrypi mopidy[3080]:   File "/usr/local/lib/python3.7/dist-packages/pkg_resources/__init__.py", line 772, in resolve
May 03 23:27:14 raspberrypi mopidy[3080]:     raise DistributionNotFound(req, requirers)
May 03 23:27:14 raspberrypi mopidy[3080]: pkg_resources.DistributionNotFound: The 'importlib_metadata<4.0,>=1.6; python_version < "3.8"' distribution was not found and is required by Pykka
May 03 23:27:14 raspberrypi systemd[1]: mopidy.service: Main process exited, code=exited, status=1/FAILURE
May 03 23:27:14 raspberrypi systemd[1]: mopidy.service: Failed with result 'exit-code'.

My Pythin Version is:
pi@raspberrypi:~ $ python --version
Python 3.7.3

## Software

### Base image and version
cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

### Branch / Release

 `master`

-->

### Installscript

cd; rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/master/scripts/installscripts/buster-install-default.sh; chmod +x buster-install-default.sh; ./buster-install-default.sh
[image]

## Hardware

### RaspberryPi version

3 B+ Rev 1.3

### RFID Reader

HXGCoLtd

### Soundcard

Hifiberry miniamp

### Other notable hardware

N/A

selmaohneh, 2021-05-04:
Same here. I tried a fix by executing `sudo pip3 install --upgrade --force-reinstall Mopidy-Iris`.
After a reboot I am able to start Spotify from the web app and even register cards with spotify uris.
The web app then shows that it is playing the track (current seconds, play/pause button working as expected), but I do not get any sound at all.

Duky87, 2021-05-04:
Pykka integrated the fix in 3.0.1 see https://github.com/jodal/pykka/issues/131
Now it is working.

closed 2021-05-04 by Duky87

MiczFlor, 2021-05-04:
Hi @Duky87 @s-martin 
Thanks for finding and fixing this. Would this need to go into the install script?
https://github.com/jodal/pykka/issues/131

Duky87, 2021-05-05:
@MiczFlor No not required. I installed it fresh after the fix was published, now it works perfectly fine.

selmaohneh, 2021-05-05:
new installation did not fix this for me - still no sound.