MiczFlor/RPi-Jukebox-RFID #1410: Future1: Flake8 etc not compatible with develop

ChisSoc, 2021-05-13:
## Issue in Future1 Branch

I'd like to get the actions flake8 and pytest working on the future1. Unfortunately the paths to certain files are not compatible between future1 and develop.

One way to go would be to configure pythonpackage.yml to run on all branches except `future1`. e.g.:

```
on: 
  push:
    branches-ignore [future1]
```

and setup a new workflow for python dedicated to the future1 branch with 

```
on: 
  push:
    branches [future1]
```

@s-martin: Could you advise? You have set up the GitHub actions workflow if I am seeing it correctly.

@arne123: PhonieboxRfidReader.py fails Flake8 in any case. Is this file actively necessary in future1 at the moment?

arne123, 2021-05-13:
Well, I am using the FakeReader for testing a lot, but just for convenience, it could be replaced using the pbc tool for example.

But I think it is also easy to control flake8 specifically for this branch by modifying the .flake8 file in the project root.
If I e.g. add
`exclude = src/jukebox/components/` 
all files under this path will be ignored by flake8 (at least locally on my machine)

ChisSoc, 2021-05-14:
On other thing: The above solution will only work for the exact branch of future1. But there will be other branches from future1 that need the future1 workflow. So, we could use a naming convention for future1-derived branches. (I don't see any other way  with those two very different development paths)

```
on: 
  push:
    branches-ignore:
      - future1
      - future1/*
```

The recommendation would be to name all branches from future1 with future1/branchname. That would be very clear and is also the usually taken approch in sub-categorizing branches and implicitly supported by many tools.

s-martin, 2021-05-14:
@ChisSoc from my point of view excluding certain branches with a wildcard and using a naming convention for the new development would be the most straightforward way.

arne123, 2021-05-14:
In general good idea, but it seems that / in branch names are not working, because future1 is already a branch *1

so we could try if e.g. _ would work

*1 https://stackoverflow.com/questions/33652953/how-to-create-a-sub-branch

ChisSoc, 2021-05-14:
Oh, nein! Na klar funktioniert das nicht mehr!

Wollen wir auf einen anderen Branchnamen wechseln, damit wir vernünftige hierarchische Branches anlegen können? Zum Beispiel

future3/develop  <-- der neue "main" branch von future ...
future3/newfeature
future3/otherfeature
future3/somethingElse

future3, weil wir die 3. Version entwickeln? Macht irgendwie auch mehr Sinn als future1 ;-)

Noch könnten wir das tun, da die Anzahl der Entwickler der Zukunft sehr begrenzt ist. 
@MiczFlor, @pabera, @arne123, @s-martin: Eure Meinung?

s-martin, 2021-05-14:
Klingt gut :)

arne123, 2021-05-14:
I finde den Vorschlag gut. 
Es wird features geben an denen mehrere Personen über längere Zeit unabhängig vom future1 entwickeln, und dann das Ergebniss mergen, spätestens dann werden wir mit dem aktuellen workflow an die grenzen laufen. 
Was nebenbei noch mit herausspringt ist evtl. der Kompromiss der es uns ermöglicht in diesem Repository zu bleiben, und trotzdem die Entwicklung gut zu strukturieren.

pabera, 2021-05-14:
Finde ich auch gut!

ChisSoc, 2021-05-14:
Cool. Ich erledige das. Dann gleich mit den passenden Github Workflows. Aber nicht mehr heute ;-)

ChisSoc, 2021-05-15:
The branch is there. Time to switch to future3/develop :-)

closed 2021-05-16 by ChisSoc