MiczFlor/RPi-Jukebox-RFID #1420: Install Script not Parsing Spotify Password into mopidy.conf correctly

Jdiesel87, 2021-05-18:
I have confirmed this with both the master and develop releases.

When you enter your Spotify credentials during the install script it results in a mopify login error. When viewing the mopidy.conf the spotify password field has incorrectly inserted "spotify_password" into the center of the correct password text. It must be edited in the config manually to authenticate with Spotify.

s-martin, 2021-05-19:
this should happen here:

https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/be6bc67c0fbcd4d5b209f80aaeb8d2b77f4035e7/scripts/installscripts/buster-install-default.sh#L923-L948

by just looking at it I can’t see an obvious problem.

UbiGIT89, 2021-05-31:
Same issue with my config.

For me it seems that the trigger is a dollar sign ($) in the password.
The "$" and the next character are missing in the mopidy.conf and have to be added manually.

It doesn't matter if I ran automated or interactiv installation.

Jdiesel87, 2021-05-31:
There is know $ in my password but I do have a & if that helps with troubleshooting. I haven't had a chance to rerun the installer script to see exactly what is happening yet.

AlvinSchiller, 2024-04-10:
This will be fixes with https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/2315

kingosticks, 2024-04-10:
haha, you have a fix now. I shouldn't have gone for lunch before submitting that!

closed 2024-04-14 by AlvinSchiller