MiczFlor/RPi-Jukebox-RFID #1425: RDM6300 output format

wolfthom, 2021-05-27:
I am currently playing around with RDM6300 card reader. 

In order to enable RDM6300 support, I replaced the Reader.py by the Reader.py.experimental file (fresh copy from repository today at 8pm) . After registering the RDM6300 using the RegisterReader script, all went fine - the reader is working and new cards are recognized by Phoniebox software.

However, the output format seems a little bit strange to me as it contains also chars. Let me give you two examples:

**10 digit card id**: 0009351929
**Phoniebox output**: 61008EB2F9A4

**10 digits card id**: 0009466639
**Phoniebox output**: 380090730FD4

This might work as long as the same card always returns the same code. However, I'd prefer  the 10 digits card id being returned from the reader script as it makes things a lot easier when id used by phoniebox internally is the same than what's printed on the card. Furthermore, I have several phonieboxes with the classic USB card reader in use which sync their audio and shortcut folder. So I'd like the reader scripts of each phoniebox to return the same ID for a card.

I tried getting the 10 digits number from the reader with a very easy script using [rdm6300 package](https://pypi.org/project/rdm6300/) - just as suggested by the documentation:

```
import rdm6300

reader = rdm6300.Reader('/dev/ttyS0')
print("Please insert an RFID card")
while True:
    card = reader.read()
    if card:
        print(f"[{card.value}] read card {card}")
```
This gives my the usual 10 digit representation. So I guess it is a matter of interpreting parity bits or stuff like that. 

Does anybody else have the same issue?

tcorbat, 2021-12-04:
I had exactly the same issue when I made first attempts with a Seeed Grove RFID reader with an Arduino. I analyzed the output format and figured out how to convert it. I described it in an issue on Seeed Studios github repo for the corresponding library ([here](https://github.com/Seeed-Studio/RFID_Library/issues/10#issuecomment-775803932)).

As you said correctly the data is transmitted including a preceeding version number and also contains the parity at the end. When cutting that away the remaining characters are ascii encoded hex digits. They represent the value of the card. The value printed on the card is its decimal value with leading 0.

I plugged the RFID reader mentioned above to the UART of my phoniebox and with the RDM6300 device got exactly the same result as you did. I'll give the library a try as well, as I would also like to have the format printed on the card.

I guess I should have looked to this issue here first, before implementing the whole transformation myself....

s-martin, 2023-12-05:
RDM6000 works with future3, see https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/future3/develop/documentation/developers/rfid/rdm6300.md

closing this as fixed

closed 2023-12-05 by s-martin