MiczFlor/RPi-Jukebox-RFID #1426: smallest phonie?

ingkebil, 2021-05-27:
First off: apologies for posting here, please refer me to the right place if possible.

What is the smallest phoniebox ever build?

I am looking to get one into an old Nintendo DS shell, but i might be pushing my luck there. The plan is to use a pi0, onoffshim, a salvaged USB Soundcard, rf522 rfid card reader, and the DS speakers. Hopefully I can reuse the DS battery, otherwise something that fits the form factor.

Anyone else tried something super small and portable?

reptilex, 2021-07-16:
Well I put mine into a bigger shell. But the pi zero w and rfc522 work like a charm. I'm using an usb soundcard which has a bit of noise but since this is for my small children they manage without complaints. [Here is](https://github.com/MiczFlor/RPi-Jukebox-RFID/discussions/1331#discussioncomment-1010263) what I built with it, you can see the usb card the pi and the reader. This could be a 4th of the size easily. The only thing you have to consider is the speakers. I seriously doubt you can power the one's from the ds but with headphones it should be no problem. I don't know about the ds battery but there are a lot of batteries that would fit the size you are aiming leaving space for the pi0. Maybe you will have to arrange for the rfid card reader to be on the top? Good luck.

iwishitwassummer, 2021-09-25:
Here’s mine 
https://github.com/iwishitwassummer/zerobox/wiki/Photos