MiczFlor/RPi-Jukebox-RFID #1435: No Audio Output through PhonieBox, VLC is working

Roman7891, 2021-06-12:
No Audio Output through PhonieBox, VLC is working

### What I did

Installed a fresh RasPi OS and went through every troubleshooting I could find.

Found out, that iFace-name ist "Master" and changed it in any Config-File I found.

Audio output via VLC is working, vie PhonieBox it's not. Even the StartUp and ShutDown Sounds are not working.

Do you have any idea how to find out where the problem is?

Hardware:
     -RPi 4 8GB
     -Neuftech RFID (which is working)
     -Internal Audio Out
     -3W Hama computer speakers

Regards
Roman

PS: If we get this thing to work, it will get a BIG beer via PayPal ;)

Roman7891, 2021-06-12:
I just installed everything again and wrote everything I did down. Hopefully someone finds a mistake I did.

- Raspberry Pi Imager v1.6.2.    —    Raspberry Pi OS Full (32-bit) 2021-05-07
- „ssh“ - txt file to „boot“ 
- Via SSH
	
	- sudo raspi-config

		- changed audio output
		- changed PW
		- enabled boot to login
		- changed screen resolution
		- enabled VNC
	

-Via VNC

	- Went through Welcome-Wizzard
		
		- Set Location
		- Connected to WLAN
		- updated system

	- via terminal

		- sudo apt-get update
		- sudo apt-get upgrade
		- cd; rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/master/scripts/installscripts/buster-install-default.sh; chmod +x buster-install-default.sh; ./buster-install-default.sh
		
- uploaded a MP3 file via PhonieBox web interface, no sound with PhonieBox and VLC
-  after reboot, audio output has been changed back to HDMI, so I entered raps-config and changed it to headphones back again. 
- VLC plays MP3 file and sound is working
- PhonieBox does not play any sound

- via VNC

	- Terminal

		- sudo nano /etc/mpd.conf
		- checked mixer_control is set correctly to „Master“

		- pi@raspberrypi:~ $ amixer scontrols
		
			Simple mixer control 'Master',0
			Simple mixer control 'Capture',0

		- pi@raspberrypi:~ $ amixer sget Master
		
			Simple mixer control 'Master',0
  			Capabilities: pvolume pswitch pswitch-joined
  			Playback channels: Front Left - Front Right
  			Limits: Playback 0 - 65536
  			Mono:
  			Front Left: Playback 65536 [100%] [on]
  			Front Right: Playback 65536 [100%] [on]

Sina-Sche, 2021-06-29:
I have the exact same problem, have you been able to figure anything out yet? I also tried all troubleshooting solutions but so far, no luck. Would be great to know if you figure out something!

s-martin, 2021-07-04:
Are you using a monitor for setting up the Pi and unplug it afterwards?

then the index of the output changes. This annoying behavior was introduced in Pi last year, please check https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ#raspberry-pi-update-may-2020-broke-my-audio

Roman7891, 2021-07-05:
@sina-sche 

Unfortunately I have not been able to solve the problem yet.

@s-martin

I have not connected a monitor to the RPI. Installed it via ssh and later vie vnc.

KaAnor86, 2021-08-05:
Hallo 
@Sina-Sche 
@Roman7891

I had the same problems but the Troubelshooting sektion helpt me and now i have sound with the phoniebox on headphonejack 

 
$ sudo nano /usr/share/alsa/alsa.conf
replace

defaults.ctl.card 0
defaults.pcm.card 0

with:

defaults.ctl.card 1    
defaults.pcm.card 1 

and i also change
 /etc/asound.conf.

with
$ sudo nano /etc/asound.conf

Strg +C 

##################
pcm.!default {
    type hw
    card 1
}
ctl.!default {
    type hw
    card 1
}
####################
shift + strg + V

strg+ o 
enter 
Strg +x

sudo reboot 

and the sound works

s-martin, 2022-12-24:
Closing for now as there was no activity for a long time.
Feel free to reopen a new issue, if the issue persists with current version 2.4 (or beta 3.2).

closed 2022-12-24 by s-martin