MiczFlor/RPi-Jukebox-RFID #1440: hold time on shutdown button is ignored

phprehl, 2021-06-20:
### What I did

installed the latest pi os on a raspi 3. 
Installed phoniebox with the script on github. 
Installed a pushbutton between GND and GPIO3 without any resistor.
changed the gpio_settings.ini to
 [Shutdown]
enabled: True
Type:  Button
Pin: 3
pull_up: True
hold_time: 2
functionCall: functionCallShutdown

### What happened

the hold time is ignored when, so when I push the button the raspi shutsdown immediately.

### I expected this to happen

After 2 seconds the raspi shutsdown

veloxidSchweiz, 2021-06-25:
I am not sure which version you are using.
In case you use the develop branch the section should be [ShutdownButton] (see https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/develop/components/gpio_control/gpio_control.py#L76).
In the master branch the shutdown Button is not enabled yet 

I am wondering why the shutdown is occuring at all. Could it be that there is a configuration in the /boot​/config.txt?
see also https://blog.gc2.at/post/pi-herunterfahren/

T0bi79, 2021-06-25:
Hi @phprehl ,

have a look here. I reported the same bug 2 months ago: https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/1377

Afterwards, I fixed it and reworked large parts of the GPIO module.
However, to my knowledge this fix is not yet in master branch, only in develop branch. But it is possible to just replace the component files of a master installation, that is what I finally did on my boxes.

Before my fixes, the hold_time parameter only worked when using the hold&repeat function (holding a button causes instant action and repetition after each hold_time interval).
Obviously, the prior feature of postponing an action until hold_time is reached was lost when the old GPIO code was reworked into the GPIO component.

You might want to read the new extended documentation about how the postponed mode can be used now (at least once the develop stuff will be merged into master for next release):
https://github.com/MiczFlor/RPi-Jukebox-RFID/tree/develop/components/gpio_control

s-martin, 2021-08-10:
Duplicates #1377

s-martin, 2022-12-20:
Closing as there was no activity for a very long time. #1377 was fixed, so this should also be fixed.

Please check the current release 2.4 or our new implementation 3.2, if this issue still persists.

Feel free to open a new issue, if necessary.

closed 2022-12-20 by s-martin