MiczFlor/RPi-Jukebox-RFID #1448: 🐛 | hold_time for GPIO buttons does not work

Kullid, 2021-07-14:
Hello,

I'm just set up a new phoniebox with the latest buster image and version 2.2 using the one line install script and activated gpio support.

Everything works fine, except the hold time of the shutdown button. I put those lines in gpio_settings.ini:

[Shutdown]
enabled: True
Type:  Button
Pin: 3
pull_up: True
hold_mode: Postpone
hold_time: 2
functionCall: functionCallShutdown

That should be correct, if I want that you have to press the shutdown button for 2 seconds before any action is taken. At least if I understand https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/develop/components/gpio_control/README.md correctly, do I?

How to fix?

Thanks
Robert

Here my config
`************************************
*** PHONIEBOX INFO
*** version: 2.2 - 305325d - master
*** edition: classic
*** Audio_iFace_Name: Headphone
*** Audio_Folders_Path: /home/pi/RPi-Jukebox-RFID/shared/audiofolders
*** Audio_Volume_Change_Step: 10
*** Max_Volume_Limit: 65
*** Idle_Time_Before_Shutdown: 10
*** Second_Swipe: NOAUDIOPLAY
*** Playlists_Folders_Path: /home/pi/RPi-Jukebox-RFID/playlists
*** ShowCover: ON
************************************
*** CONF FILES DEFAULT
 
*** /etc/samba/smb.conf
-rw-r--r-- 1 root root 8,2K Jul  9 21:41 /etc/samba/smb.conf
   path=/home/pi/RPi-Jukebox-RFID/shared
   path=/home/pi/RPi-Jukebox-RFID/shared/audiofolders
*** /etc/lighttpd/lighttpd.conf
-rw-r--r-- 1 root root 2,1K Jul  9 21:41 /etc/lighttpd/lighttpd.conf
*** /etc/lighttpd/conf-available/15-fastcgi-php.conf
-rw-r--r-- 1 root root 575 Jul  9 21:41 /etc/lighttpd/conf-available/15-fastcgi-php.conf
*** /etc/php/7.0/fpm/php.ini
ls: Zugriff auf '/etc/php/7.0/fpm/php.ini' nicht möglich: Datei oder Verzeichnis nicht gefunden
*** /etc/sudoers
-r--r----- 1 root root 784 Jul  9 21:41 /etc/sudoers
*** /etc/systemd/system/phoniebox*
-rw-r--r-- 1 root root 295 Jul  9 21:41 /etc/systemd/system/phoniebox-rfid-reader.service
ls: Zugriff auf '/etc/systemd/system/phoniebox-startup-sound.service' nicht möglich: Datei oder Verzeichnis nicht gefunden
ls: Zugriff auf '/etc/systemd/system/phoniebox-gpio-buttons.service' nicht möglich: Datei oder Verzeichnis nicht gefunden
-rw-r--r-- 1 root root 291 Jul  9 21:41 /etc/systemd/system/phoniebox-idle-watchdog.service
*** /etc/mpd.conf
-rw-r----- 1 mpd audio 14K Jul  9 21:42 /etc/mpd.conf
music_directory		"/home/pi/RPi-Jukebox-RFID/shared/audiofolders"
# music_directory are changed.
# symbolic links outside of the configured music_directory.
# symbolic links inside of the configured music_directory.
	mixer_control	"Headphone"		# optional
#	mixer_control	"PCM"		# optional
*** /etc/dhcpcd.conf
-rw-rw-r-- 1 root netdev 130 Jul  9 21:42 /etc/dhcpcd.conf
static ip_address=192.168.205.85/24
static routers=192.168.205.1
static domain_name_servers=8.8.8.8 192.168.205.1
*** /etc/wpa_supplicant/wpa_supplicant.conf
-rw-rw-r-- 1 root netdev 163 Jul  9 21:42 /etc/wpa_supplicant/wpa_supplicant.conf
country=DE
************************************
*** +Spotify Edition
 
*** /etc/locale.gen
-rw-r--r-- 1 root root 9,2K Mai  7 17:13 /etc/locale.gen
*** /etc/locale.gen
-rw-r--r-- 1 root root 9,2K Mai  7 17:13 /etc/locale.gen
*** /etc/mopidy/mopidy.conf
ls: Zugriff auf '/etc/mopidy/mopidy.conf' nicht möglich: Datei oder Verzeichnis nicht gefunden
cat: /etc/mopidy/mopidy.conf: Datei oder Verzeichnis nicht gefunden
*** ~/.config/mopidy/mopidy.conf
ls: Zugriff auf '/home/pi/.config/mopidy/mopidy.conf' nicht möglich: Datei oder Verzeichnis nicht gefunden
cat: /home/pi/.config/mopidy/mopidy.conf: Datei oder Verzeichnis nicht gefunden`

T0bi79, 2021-07-14:
Hi!

It seems you installed the PhonieBox from the master branch. The GPIO component there does not support this behavior yet.

I fixed it a few weeks ago, but the code is not yet in the master branch but only in develop.
Your link and your ini syntax seems fine for me and it should work if (!) you use the GPIO component from the develop branch, not from the master branch,

If you do not want to re-install from the develop branch or to wait for the next master release:  I did also setup my current (productive) boxes using the master branch. To use the postpone feature on master, it is sufficient to only replace the components/gpio_control folder with the reworked one from the develop branch.

Kullid, 2021-07-17:
Thanks Tobi your suggestion works.

closed 2021-07-17 by Kullid