MiczFlor/RPi-Jukebox-RFID #1451: No cover displayed from Spotify in web GUI🐛

usamer77, 2021-07-15:
## Bug

### What I did

After several installations with the installscript and Spotify authentication, no covers get displayed in the web GUI.
I am on this buster image: 5.10.17-v7l+ #1421 SMP on a PI4 B Rev. 1.1

I believe it could be a rights issue. I tried to pull up the user groups:

_pi adm dialout cdrom sudo audio video plugdev games users input netdev gpio i2c spi_

Which one would be the correct one?

Many thank in advance!

BR

usamer77, 2021-07-19:
I just refreshed the user rights like this:
```
sudo chown -R pi:www-data /home/pi/RPi-Jukebox-RFID/shared
sudo chmod -R 775 /home/pi/RPi-Jukebox-RFID/shared
sudo chown -R pi:www-data /home/pi/RPi-Jukebox-RFID/htdocs
sudo chmod -R 775 /home/pi/RPi-Jukebox-RFID/htdocs
```
But I still get this broken pic instead of the cover.jpg  displayed... I tried to copy a "cover.jpg" manually, but also no success...

Browser console info:

[image]

I can't see the user "www-data", could that be a problem?

BR
U

r0k5t4r, 2021-09-12:
Same problem here. Just did a complete reinstall raspbian (buster) and latest master

usamer77, 2021-09-17:
Still trying to get it solved, without success so far... If somebody could help?!?! Many thanks in advance!

passuff, 2021-09-17:
I have exactly the same issue even though the old topic was closed after fixing more than a year [ago.](https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/884) I just installed Version "2.3 - 2425890 - master". I think it's the most recent version. Cover.jpeg and title.txt are created but empty. Any help appreciated...

closed 2022-05-31 by s-martin