MiczFlor/RPi-Jukebox-RFID #1464: MF522 based SPI reader does not work

mippen, 2021-08-03:
Trying my best to get this great software to work but have huge issues getting the MF522 reader to work.
My steps so far:

- Installed the MF522 according to https://pimylifeup.com/raspberry-pi-rfid-rc522/
- - Works great reading tags with the example code on this page.
- Taking a blank raspos.. only configure wifi + ssh
- running the default install script. No Spotify, No GPIO, With the 522 hardware selected.
- With lsmod | grep spi the reader shows up
- 
pi@raspberrypi:~ $ lsmod | grep spi
spidev                 20480  2
spi_bcm2835            20480  0

-- No working reading of card in the gui

- Replacing the Reader.py with Reader.py.experimental
- - No working reader
- Running the ./setup_rc522.sh
-- No working reader

Any input would be appreciated.

mippen, 2021-08-03:
Running this small test software reads cards.. Hardware seems to be working fine.
Have not managed to do it with the Reader.py.experimental file.

#!/usr/bin/env python

import RPi.GPIO as GPIO
from mfrc522 import SimpleMFRC522

reader = SimpleMFRC522()

try:
        id, text = reader.read()
        print(id)
        print(text)
finally:
        GPIO.cleanup()

Dalada89, 2021-08-07:
Hey mippen,

did you connect the IRQ PIN from the card reader with the GPIO?
In the link to [pimylifeup](https://pimylifeup.com/raspberry-pi-rfid-rc522/) it isn't required, but the software of the jukebox needs this connection. It is also mentioned in the [Wiki](https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Wiring_for_RC522_card_reader).
That was my mistake a few weeks ago. The card reader worked in a simple test, but not with the jukebox software. I just connected the IRQ PIN and it worked properly.

Best regards,
Dalada