MiczFlor/RPi-Jukebox-RFID #1480: Spotify Edition - Single File Playback Working / Folder Playback not working

aradhir, 2021-08-12:
Dear all,

I just installed the Spotify Edition on a Raspberry Pi 2 B. Installation happened via the automatic, guided installer from the current repository. After configuring audio output, I am struggling with folder playbacks.

**The following things are working**
- Youtube-integration --> clicking the "Play" button in the section "Player -> Available Audio" works.
- Single file playback --> clicking the "Play" button for the single file in the section "Player -> Available Audio -> Folder -> File" works.
- RFID swipes Youtube

**What is not working?**
When trying to start playback of a whole folder (both manually and via RFID), the player shows "XXX is playing", but nothing happens, and also the playlist shown in the web inteface is not refreshed.

**Logs**
I simulated the scenario with basically the same file, once played via the folder play button, once via the file play button:

_Folder playback:_
`########## SCRIPT rfid_trigger_play.sh (2021-08-13.00:13:27) ##
# Attempting to play: /home/pi/Music/Meister Eder und sein Pumuckl/Die Gummiente
# Type of play $VALUE: 
$FOLDER set, not empty and dir exists: /home/pi/Music/Meister Eder und sein Pumuckl/Die Gummiente
  Var $LASTFOLDER: Meister Eder und sein Pumuckl/Die abergläubische Putzfrau
  Var $LASTPLAYLIST: Meister Eder und sein Pumuckl % Die abergläubische Putzfrau
Checking 'recursive' list? VAR $VALUE: 
recursive? NO
/home/pi/RPi-Jukebox-RFID/scripts/playlist_recursive_by_folder.php --folder "Meister Eder und sein Pumuckl/Die Gummiente" > "/home/pi/RPi-Jukebox-RFID/playlists/Meister Eder und sein Pumuckl % Die Gummiente.m3u"
  Var $SECONDSWIPE: RESTART
  Var $PLAYLISTNAME: Meister Eder und sein Pumuckl % Die Gummiente
  Var $LASTPLAYLIST: Meister Eder und sein Pumuckl % Die abergläubische Putzfrau
We must play the playlist no matter what: $PLAYPLAYLIST == yes
  VAR FOLDER: Meister Eder und sein Pumuckl/Die Gummiente
  VAR PLAYLISTPATH: /home/pi/RPi-Jukebox-RFID/playlists/Meister Eder und sein Pumuckl % Die Gummiente.m3u
########### SCRIPT playout_controls.sh (2021-08-13.00:13:28) ##
VAR COMMAND: playerstop
VAR VALUE: 
current file: /home/pi/Music/file:///home/pi/Music/Meister-Eder-und-sein-Pumuckl/Der-verstauchte-Daumen.mp3
chapters file: /home/pi/Music/file:///home/pi/Music/Meister-Eder-und-sein-Pumuckl/Der-verstauchte-Daumen.chapters.json
chapters for extension enabled: 0
chapters for duration enabled: 0
   playerstop
#START##### SCRIPT resume_play.sh (2021-08-13.00:13:28) ##
VAR AUDIOFOLDERSPATH: /home/pi/Music
VAR COMMAND: savepos
VAR VALUE: 
VAR FOLDER: 
  content of /home/pi/Music//folder.conf
CURRENTFILENAME="filename"
ELAPSED="0"
PLAYSTATUS="Stopped"
RESUME="OFF"
SHUFFLE="OFF"
LOOP="OFF"
SINGLE="OFF"
  Now doing what COMMAND wants: savepos
VAR FOLDER from settings/Latest_Folder_Played: Meister Eder und sein Pumuckl/Die abergläubische Putzfrau
  savepos FOLDER: Meister Eder und sein Pumuckl/Die abergläubische Putzfrau
CURRENTFILENAME="filename"
ELAPSED="0"
PLAYSTATUS="Stopped"
RESUME="OFF"
SHUFFLE="OFF"
LOOP="OFF"
SINGLE="OFF"
#END####### SCRIPT resume_play.sh (2021-08-13.00:13:28) ##
remove playing.txt
########### SCRIPT playout_controls.sh (2021-08-13.00:13:28) ##
VAR COMMAND: playlistaddplay
VAR VALUE: Meister Eder und sein Pumuckl % Die Gummiente
current file: /home/pi/Music/file:///home/pi/Music/Meister-Eder-und-sein-Pumuckl/Der-verstauchte-Daumen.mp3
chapters file: /home/pi/Music/file:///home/pi/Music/Meister-Eder-und-sein-Pumuckl/Der-verstauchte-Daumen.chapters.json
chapters for extension enabled: 0
chapters for duration enabled: 0
   playlistaddplay playlist name VALUE: Meister Eder und sein Pumuckl % Die Gummiente
   playlistaddplay FOLDER: Meister Eder und sein Pumuckl/Die Gummiente
mpc load Meister Eder und sein Pumuckl % Die Gummiente
## SCRIPT single_play.sh (2021-08-13.00:13:29) ##
VAR AUDIOFOLDERSPATH: /home/pi/Music
VAR COMMAND: single_check
VAR VALUE: 
VAR FOLDER: Meister Eder und sein Pumuckl/Die Gummiente
  content of /home/pi/Music/Meister Eder und sein Pumuckl/Die Gummiente/folder.conf
CURRENTFILENAME="filename"
ELAPSED="0"
PLAYSTATUS="Stopped"
RESUME="OFF"
SHUFFLE="OFF"
LOOP="OFF"
SINGLE="OFF"
  Now doing what COMMAND wants: single_check
  entering: single_check with value OFF
## SCRIPT shuffle_play.sh (2021-08-13.00:13:29) ##
VAR AUDIOFOLDERSPATH: /home/pi/Music
VAR COMMAND: shuffle_check
VAR VALUE: 
VAR FOLDER: Meister Eder und sein Pumuckl/Die Gummiente
# found content in folder.conf:
CURRENTFILENAME="filename"
ELAPSED="0"
PLAYSTATUS="Stopped"
RESUME="OFF"
SHUFFLE="OFF"
LOOP="OFF"
SINGLE="OFF"
  Now doing what COMMAND wants: shuffle_check
  entering: shuffle_check with value OFF
# new content in folder.conf:
CURRENTFILENAME="filename"
ELAPSED="0"
PLAYSTATUS="Stopped"
RESUME="OFF"
SHUFFLE="OFF"
LOOP="OFF"
SINGLE="OFF"
mpc load Meister Eder und sein Pumuckl % Die Gummiente && /home/pi/RPi-Jukebox-RFID/scripts/resume_play.sh -c=resume -d=Meister Eder und sein Pumuckl/Die Gummiente
#START##### SCRIPT resume_play.sh (2021-08-13.00:13:29) ##
VAR AUDIOFOLDERSPATH: /home/pi/Music
VAR COMMAND: resume
VAR VALUE: 
VAR FOLDER: Meister Eder und sein Pumuckl/Die Gummiente
  content of /home/pi/Music/Meister Eder und sein Pumuckl/Die Gummiente/folder.conf
CURRENTFILENAME="filename"
ELAPSED="0"
PLAYSTATUS="Stopped"
RESUME="OFF"
SHUFFLE="OFF"
LOOP="OFF"
SINGLE="OFF"
  Now doing what COMMAND wants: resume
  savepos FOLDER: Meister Eder und sein Pumuckl/Die Gummiente
  entering: resume with value OFF
  entering: single with value OFF
#END####### SCRIPT resume_play.sh (2021-08-13.00:13:29) ##
  echo Meister Eder und sein Pumuckl/Die Gummiente > /home/pi/RPi-Jukebox-RFID/scripts/../settings/Latest_Folder_Played
  VAR Latest_Folder_Played: Meister Eder und sein Pumuckl/Die Gummiente
  # end playout_controls.sh playlistaddplay
  Command: /home/pi/RPi-Jukebox-RFID/scripts/playout_controls.sh -c=playlistaddplay -v="Meister Eder und sein Pumuckl % Die Gummiente" -d="Meister Eder und sein Pumuckl/Die Gummiente"`

_Single file playback:_
`########### SCRIPT playout_controls.sh (2021-08-13.00:13:37) ##
VAR COMMAND: playsinglefile
VAR VALUE: /home/pi/Music/Meister Eder und sein Pumuckl/Die Gummiente/Die-Gummiente.mp3
current file: /home/pi/Music/
chapters file: /home/pi/Music.chapters.json
chapters for extension enabled: 0`

I tried to browse the issues for my topic, but I think it is not related to any of the other issues.

Best
Aradhir

simontiffert, 2021-08-13:
I am new to the project and it looks like I am having the folder issue as well. I thought it is a sound issue, but after opening a folder in the web ui and clicking on the test mp3, it played without any issues. 

Spotify Edition installed today with one-line-installer.

s-martin, 2021-09-02:
What’s the output of `journalctl -u mopidy`?

aradhir, 2021-09-11:
Sorry, just saw, that I tried with a different file/folder, but it shouldn't matter for the debugging. I recorded the follow logs:

1) directly after startup and the "startup sound" before starting any playback
2) after trying to start the playback via the "play folder" button
3) after starting the playback via the "play single file" button

Note: I know, that my Spotify credentials are wrong. Currently I do not have a subscription, because I wanted to do the testing first. I assumed so far, that playing local files with the Spotify Edition (even without subscription) should be possible at any time, or am I wrong?

Log No.1
[mopidy_1.txt](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/7148510/mopidy_1.txt)

Log No.2
[mopidy_2.txt](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/7148511/mopidy_2.txt)

Log No.3
[mopidy_3.txt](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/7148512/mopidy_3.txt)

robinjoerke, 2021-09-16:
I experiienced the same issue (Folder did not play, web UI & RFID, but clicking the singe file worked). 
Try the following. open web interface and go to `Folders & Files.` At the very Bottom of the screen click the `Scan Music Library` Button. 
I could playback the folder afterwards (both card and RFID ) even though the label on the button indicates this is only for Track info in web UI

aradhir, 2021-09-16:
> I experienced the same issue (Folder did not play, web UI & RFID, but clicking the singe file worked).
> Try the following. open web interface and go to `Folders & Files.` At the very Bottom of the screen click the `Scan Music Library` Button.
> I could playback the folder afterwards (both card and RFID ) even though the label on the button indicates this is only for Track info in web UI
Thanks for the hint, but it did not work for me.

bwa-, 2021-09-27:
I'd just like to chime in and say that I have the same problem as OP. Scanning the library didn't help me either. I'm running the very same version on a Pi 4B. Spotify playback with RFID works, everything else is the same as described above.

bwa-, 2021-09-27:
So I've made some progress on this:

After taking a look at some of the Mopidy logs and trying to figure out what's wrong, I decided to try and play these files with `mpc` directly. After messing around with the syntax for a bit, I found out that, while the playlist _existed, it did not contain any files_. I found this out by running `mpc playlist $FOLDER_NAME_IN_WEBINTERFACE`. You can also show existing playlists with `mpc lsplaylists`.

So much for the straightforward part, now it gets weird: After figuring this out, I created a new directory in the shared/audiofolders dir with a simple name and moved one mp3 over to make sure the filenames didn't have anything to do with this. To make Mopidy pick this dir up, I then hit 'Scan Music Library'. After that. for some reason, _the files that had been missing were suddenly part of the playlist that was empty before._

Now I've also found out that the playlists apparently are created when you first hit that 'Play Folder' button for the first time. So if you're wondering where your playlist is, this may be why it's missing.
Also, you can browse the files Mopidy knows about with 'mpc ls'. I found all my files after that restart with `mpc ls 'Local media/Tracks'`

Hope this helps somebody - ever since restarting after messing around in the local audio directory my problems have vanished for some reason.

simontiffert, 2021-10-18:
@bwa- Thanks, your hints helped to fix the mp3 playback of folders. Not sure about the order but I tried different things.
Overall working with mpc on command line helps to identify and fix the issues.

As the Spotify playback didn't work as well, I looked into the log files and Mopidy seems to start before the network was available. Changed the settings (https://discourse.mopidy.com/t/spotify-authentication-works-only-after-server-restart/3617/6) and now it is working (with a slower start).

Lippsson, 2021-11-23:
> > Try the following. open web interface and go to `Folders & Files.` At the very Bottom of the screen click the `Scan Music 
Thank you - That help me when i forced this issue

s-martin, 2024-11-10:
Spotify implementation has been undergone various improvements.
Please check release [2.8.0](https://github.com/MiczFlor/RPi-Jukebox-RFID/releases/tag/v2.8.0).

Closing now, please reopen, if the issue persists.

closed 2024-11-10 by s-martin