MiczFlor/RPi-Jukebox-RFID #1498: Swipe or place a card Functionality not working

arvidortwig, 2021-09-01:
Hi,

first of all thank you for this wonderful project. I got the phoniebox working so far. RFID (Neuftech) scanner works fine but i cant save the "swipe or place" option properly. After some tries (random number) the option to place stays but when i place the card it just starts the audiofile to play for half a second and then stops. Swiping works flawlessly. Any idea ?

	
Version
2.3 - 2425890 - master
Edition
Classic edition (barebones)

Thanks,
Apunkt

s-martin, 2021-09-02:
What’s the output of `tail -n 200 /var/log/syslog`?

ChisSoc, 2021-09-03:
Place does not work with Neuftech readers. They only support Swipe.

Reason: Neuftech only gives a single event when a new card is placed.

You will need to change the reader to RDM6300. 

Or MFRC522 - but that uses a different NFC standard.

arvidortwig, 2021-09-04:
hi, guys. thank you for answering. I will stick to the neuftech as the RDM6300 needs some kind of voltage altering through an additional board. usb is so comfortable :-)

closed 2021-09-04 by arvidortwig