MiczFlor/RPi-Jukebox-RFID #1509: 🐛 | hold time on shutdown button is ignored

lenlennart, 2021-09-08:
## Bug

### What I did

install fresh phoniebox
connect push button to GPIO 3 and GRD
activate (enabled : True) shutdown button in config
leave other parameters on default (in particular "hold_time : 2")
press shutdown button only briefly (0.2 sec)

### What happened

phoniebox goes in shutdown

### I expected this to happen

phoniebox stay on, as hold time not reached

### Further information that might help

tried setting holdtime to "2.0"

## Software

### Base image and version

2.3 - 2425890 - master

## Hardware

### RaspberryPi version

3 B

T0bi79, 2021-09-09:
Can you post the entire section of this button in settings/gpio_settings.ini?

lenlennart, 2021-09-09:
after digging thorugh the documentation, it seems the default gpio_settings.ini is missing "hold_mode" (-> Postpone) for the shutdown button.
Works like a charm once added.

closed 2021-09-09 by lenlennart

T0bi79, 2021-09-09:
Ah, you're right, that's the problem. It's the same on my test installation.

However, in \components\gpio_control\example_configs\gpio_settings.ini the shutdown button example is correct. It seems another file is copied to 'settings' during installation. I have a look on this later...