MiczFlor/RPi-Jukebox-RFID #1513: 🐛 | BUG SUMMARY RFID Reader Neuftech not working

Baggerfahrer, 2021-09-12:
## Bug

### What I did

Used the one-line install script to install Phoeniebox.
The USB Reader from Neuftech does not work in the web-app and the Service is not activate in the web-app.
Everytime i swap a new chip over the reader, no new id is recognized.
See the following picture.

[image]

Moreiver, i also avticated the GPIO Service in the installation script, but this is also not activated? Side-issue!

### What happened

During installation script, the following error occurs.

./RegisterDevice.py
Choose the reader from list
0 HXGCoLtd HIDKeys Keyboard
Device Number: 0
Exception ignored in: <function InputDevice.__del__ at 0x76167fa8>
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/evdev/device.py", line 160, in __del__
  File "/usr/local/lib/python3.7/dist-packages/evdev/device.py", line 305, in close
  File "/usr/local/lib/python3.7/dist-packages/evdev/eventio_async.py", line 55, in close
  File "/usr/lib/python3.7/asyncio/events.py", line 640, in get_event_loop
  File "/usr/lib/python3.7/asyncio/events.py", line 660, in new_event_loop
  File "/usr/lib/python3.7/asyncio/unix_events.py", line 51, in __init__
  File "/usr/lib/python3.7/asyncio/selector_events.py", line 49, in __init__
  File "/usr/lib/python3.7/asyncio/base_events.py", line 373, in __init__
  File "/usr/lib/python3.7/asyncio/coroutines.py", line 28, in _is_debug_mode
AttributeError: 'NoneType' object has no attribute 'dev_mode'
Exception ignored in: <function BaseEventLoop.__del__ at 0x761a8618>
Traceback (most recent call last):
  File "/usr/lib/python3.7/asyncio/base_events.py", line 622, in __del__
  File "/usr/lib/python3.7/asyncio/base_events.py", line 390, in __repr__
  File "/usr/lib/python3.7/asyncio/base_events.py", line 1794, in get_debug
AttributeError: '_UnixSelectorEventLoop' object has no attribute '_debug'

### I expected this to happen

Clear recognition of the USB Reader.
The USB reader is working, i checked it at another PC, and it clearly finds the chip id in a e.g. notepad.

### Further information that might help

uname -a
Linux raspberrypi-lulu 5.10.60-v7+ #1449 SMP Wed Aug 25 15:00:01 BST 2021 armv7l GNU/Linux

## Software

### Base image and version

cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

### Branch / Release

cd /home/pi/RPi-Jukebox-RFID/ && git status | head -2
Refresh index: 100% (427/427), done.
On branch master
Your branch is up to date with 'origin/master'.

## Hardware

### RaspberryPi version
sudo cat /sys/firmware/devicetree/base/model
Raspberry Pi 3 Model B Rev 1.2

### RFID Reader

  idVendor           0x16c0 Van Ooijen Technische Informatica
  idProduct          0x27db Keyboard
  bcdDevice            0.01
  iManufacturer           1 HXGCoLtd
  iProduct                2 HIDKeys

### Soundcard

### Other notable hardware

Baggerfahrer, 2021-09-16:
issue can be closed.
After a newly set-up of the rpi with an sudo apt-get update afterwards, everything went well....

closed 2021-09-16 by Baggerfahrer

timoweber1985, 2021-11-01:
All, I have the same issue with a fresh installation of RPI and Jukebox. Any help or ideas?