MiczFlor/RPi-Jukebox-RFID #1516: Can't open Web App via IP

JayJayJayJayYES, 2021-09-15:
## Bug
If I type the IP of the phonie box into browser it is not going anywhere: no connection
### What I did
i Installed via the one line install script. After installation and reboot I lost connection to the pi via VNC viewer. I connected the pi then to monitor and keyboard. There is still the w-lan connection, but when I enter the IP into the browser it does not find anything. After disconnecting the w-lan and typing in the password for the w-lan in again, I can connect via VNC. But still can't get connection via browser and IP to the web app. I checked if the IP changed but it did not! 

### What happened
nothing after entering the IP into the browser

### I expected this to happen
entering the IP and the Web App opens up

### Further information that might help

## Software

### Base image and version

### Branch / Release

### Installscript

## Hardware

### RaspberryPi version

### RFID Reader

### Soundcard

### Other notable hardware

lenlennart, 2021-09-23:
not reproducible for me, works as expected on my pi3.
Did you check for error messages during installation? Also, did you try re-installing?

s-martin, 2022-12-20:
Closing as there was no activity for a very long time.

Please check the current release 2.4 or our new implementation 3.2, if this issue still persists.

Feel free to open a new issue, if necessary.

closed 2022-12-20 by s-martin