MiczFlor/RPi-Jukebox-RFID #1551: NAS with cifs, nfs an symbolic links will not be played

ASchmidt84, 2021-10-08:
Hi,
I set all up. And if i upload a file i can play it without any problems.

For my son I want add the music and stories from our wd my cloud nas server.

I linked them in fstab. The files are visible.

First I mounted only an directory with 12 subdirs. If i click play i get the green box (webpage) with is playing but their is no play.
Another change I create a symbolic link (found in one issue here). Also the same.

If i upload a file, i can play.

Where is my problem?

Thank you

janisHD, 2021-11-08:
Hi,

I just created following issue: #1621
After creation I saw that you *maybe* have a similar problem.
My playback was at least working after 2nd swipe (waiting for 2s in between) and it worked as expected after changing from read-only to read/write permission on the remote (NFS) folder.

gizmo21, 2022-03-21:
Perhaps some those old comments here help: https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/199

spiderbit, 2023-04-02:
Have the same problem, don't get what's wrong, I have write access, the UI finds the files but it doesn't start playing in no sound comes.

I am confused how that is even possible it has write access but can't play it? the user,group and other has read access of the files...

Also I think if there is no easy config option to include samba shares or nfs, it should be described in the wiki, it's pretty obvious who wants to put their whole mp3 collection on a sdcard? Sure you could use a external harddisk but even then the problem would be similar.

AlvinSchiller, 2023-04-02:
Hi, 
how did you mount your network drive and created your links?
Which usergroup have you assigned?

Please try `mpc listall` in a terminal and see if your songs are listed.
If not execute `mpc update` (a reboot should also perform the update) and try again.

spiderbit, 2023-04-03:
@AlvinSchiller thanks this actually fixed it. Sorry for my flamatory tone yesterday, was a bit frustrated.

pabera, 2023-04-03:
You might have to rerun the command every time you add songs to your NAS.