MiczFlor/RPi-Jukebox-RFID #1552: [feature3] I want to contribute

Groovylein, 2021-10-08:
Hey guys,

I appreciate your work on future3. I did not look into everything, nor did a test installation yet, but I want to help! 

I have knowledge in Linux, python and pytest but want also to extend my knowledge. I'm also interested in the webapp 😊

I saw your status list of futures, but I don't know which ones are WIP. Is there some kind of communication so that you don't work on the same topics?

Regards, 
Groovylein

pabera, 2021-10-09:
There is plenty to work on :-) If you want to get your hands dirty, I suggest to look at the [features that are already implemented](https://rpi-jukebox-rfid.readthedocs.io/en/latest/featurelist.html) but miss details - like `[Volume](https://rpi-jukebox-rfid.readthedocs.io/en/latest/featurelist.html#volume)`. We still need `volume_max` (not sure if `volume_min` is required) which requires a few things like storing in config and read/write via RPC (e.g. through UI). This way, you could get familiar with the code and it's small enough to see a quick win (especially if you know Python well). I can help with the Webapp part or you try to handle that as well.

Similar options are

* [Idle timer](https://rpi-jukebox-rfid.readthedocs.io/en/latest/featurelist.html#timer) - requires also storing in config, RPC read/write and the actual execution -> shutting down the box after the period of time
* The entire [Playback](https://rpi-jukebox-rfid.readthedocs.io/en/latest/featurelist.html#playback) sections looks interesting and has not yet been taken care of.
* Or [Others](https://rpi-jukebox-rfid.readthedocs.io/en/latest/featurelist.html#others)
* GPIO is currently in the making, but I am unsure about the status @arne123

pabera, 2021-10-10:
Sounds good.. [RPC](https://rpi-jukebox-rfid.readthedocs.io/en/latest/concepts.html#remote-procedure-call-server-rpc) ist just a server concept (written in Python), nothing too complicated ;-)

arne123, 2021-10-10:
>     * GPIO is currently in the making, but I am unsure about the status @arne123
Yes, I already started porting GPIO to the new structure, so it's in progress. 
But of course every help is welcome, also here.
I think within the next few days I will push first suggestions to a branch.

ChisSoc, 2021-10-10:
Volume would be a good starting point. 

Playback is currently tackled by myself. Idle timer is difficult as it means very deep integration into the RPC and MPD. That's the reason it is not there yet. As a second step, you can also look at the non-ported modules and see if there is one that interests you most. Please check back with us first - there is lots of stuff already in our heads and we can give you a few starting hints :-)

I would also ask the following: When starting to find your way around and you think: "Here is a vital point missing to the documentation", please add it. Documentation is also at the very beginning. It would help others no end, if we keep improving it along the code-development.

Groovylein, 2021-10-13:
What are your typical tests before creating a PR or merging? (except flake8)

pabera, 2021-10-13:
> What are your typical tests before creating a PR or merging? (except flake8)

At this very moment, not much more than that :-)

Groovylein, 2021-10-13:
> > What are your typical tests before creating a PR or merging? (except flake8)
> 
> At this very moment, not much more than that :-) 

Well @ChisSoc added some pre-commits which causes some problems (see #1567)